ghdl/ghdl #1362: Support for --work between files

ozbenh, 2020-06-09:
When using ghdl -c <bunch of vhdl files> we might need to be able to put the various files
into different libraries.

For example, in microwatt, I need some in unisim (I have models mimmic'ing some xilinx stuff), some in fmf (support files for some chip models) etc...

Ideally, we want to be able to do

ghdl -c <files in work> --work foo <files in library foo> --work bar <files in library bar>

eine, 2020-06-09:
Hi @ozbenh! As commented in https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#analyze-and-elaborate-c, `ghdl -c` is a shortcut to analyze and elaborate in a single step, which puts all the units into library `work`. In order to build units into different libs, the suggested approach is to use `ghdl -a` and `ghdl -e`. 

I'm not sure it is possible for `ghdl -c` or `ghdl -i` to support having different libs. I feel it somehow defeats the purpose of them being simpler than using `-a` and `-e`. Anyway, let's see what Tristan thinks.

Apart from that, the proposed syntax feels wrong. On the one hand, the prototype is `-c <[options] file... -<e|r> primary_unit [secondary_unit]>`. So, you are missing filename(s) and either the libs or the primary unit. What do you mean with `foo` and `bar`? OTOH, I think that passing `--work` twice is not supported. According to https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-work, you might be misunderstanding its purpose with https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-workdir or https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-p-directory.

Hence, I think that you might want something such as the following:

```
# Pre-build unisim libs
mkdir -p unisim
ghdl -i --work=unisim --workdir=unisim path/to/unisim/sources/*.vhd

ghdl -c -Punisim file... -e foo
```

Note that I'm not completely sure about combining `-i` and `-m`. However, I hope you get the idea about pre-building unisim, and using `-P`.

ozbenh, 2020-06-09:
So this was more/less something Tristan suggested :-) he told me to file an issue.

The libraries in .cf files (as done by the classic ghdl -a) are somewhat annoying as they effectively are 'side effect' files note directly referenced on the command line with little control or where they are etc.. It makes things a bit difficult to deal with in Makefiles, especially when trying to keep things in subdirectories, or using separate build directories etc..

For that reason (among others), Anton changed Microwatt Makefile to always use ghdl -c with all the files listed for each test bench. It's slower but keeps things simpler and less likely to mess up dependencies.

That said, if the above doesn't work, we can try re-writing the Makefile again to better control where the .o and .cf files are created and try to get the dependencies automatically generated. Sadly none of us is a make artist..

ozbenh, 2020-06-09:
As for pre-building, I will experiment, thanks

eine, 2020-06-09:
> So this was more/less something Tristan suggested :-) he told me to file an issue.

😄 Did he propose that specific syntax or was it your interpretation? Was it in some public issue/chat that I can peek?

> The libraries in .cf files (as done by the classic ghdl -a) are somewhat annoying as they effectively are 'side effect' files note directly referenced on the command line with little control or [...]

I absolutely agree with this. In fact, during the last months/years I have opened several related issues:

- #1204 is mostly cosmetic, but can help with clarifying the similarities/differences between `-a`, `-e`, `--elab-run`, `-i`, `-c` and `-m`.
- #1060 is almost done, but makefiles/build scripts are not adapted yet.
- #980 will hopefully simplify how `*.cf`, `*.o` and `~*.o` are handled with regard to command line controls. This was recently commented by Tristan in the chat; hence it might be the next big thing after (or along with) synthesis.
- #781 is about some *niche* features which don't have support for telling the location of intermediate artifacts/objects. These are mostly `--bind` and `--list-link`, which might be used in microwatt for co-simulation with VHPIDIRECT. A workaround to simplify it is to use `ghdl -e -Wl,-Wl,...`, instead of `bind` and `list-link`. This is what might be driving you mad if you want to have everything in subdirectories.

> For that reason (among others), Anton changed Microwatt Makefile to always use ghdl -c with all the files listed for each test bench. It's slower but keeps things simpler and less likely to mess up dependencies.

> As for pre-building, I will experiment, thanks

The point here is that having some files compiled to specific libraries and then using them from a different location is a basic feature in almost any non-toy VHDL design. Hence, it is very common to pre-build some sets of libs, to avoid doing it again and again. In fact, I believe that pre-building unisim can be done with the vendor scripts contributed by @Paebbels: https://github.com/ghdl/ghdl/tree/master/libraries/vendors.

Hence, I would expect the feature you suggest to be useful if `foo` and `bar` are different libs in your own sources, which you want to compile at the same time but to different locations. E.g.:

```
ghdl -c \
  -Ppath/to/unisim \
  --work=mylib path/to/mylib/sources/*.vhd \
  --work=myotherlib path/to/myotherlib/sources/*.vhd \
  --work=work ./ \
  -e primary_unit`
```

> Sadly none of us is a make artist..

Did you consider using a tool such as [VUnit](https://github.com/VUnit/vunit) instead? Say:

```py
from pathlib import Path
from vunit import VUnit

VU = VUnit.from_argv()

SRC_PATH = Path(__file__).parent / "src"

VU.add_library("unisim").add_source_files(...)
VU.add_library("foo").add_source_files(...)
VU.add_library("bar").add_source_files(...)

VU.main()
```

VUnit handles dependencies and incremental re-compilation for you, and it allows to only generate the executable binary for simulation, without actually running it. However, it adds a little complexity to an external CLI call.

tgingold, 2020-06-09:
@eine, this issue is the place to discuss this feature.  It is very similar to what is allowed by `--synth`.

ozbenh, 2020-06-09:
Thanks, I'll investigate vunit as well. But I think -i might work for me.

As for your above example of what the syntax could be, yes mostly the libraries are in my own source (for example, unisim isn't the Xilinx one, I use it to hold a sim variant of BSCANE2 I wrote which communicates over a network socket :)

aspdigital, 2020-06-09:
@ozbenh brings up an interesting point. I have my own variants on the unisim library, as well as other FPGA vendor's simulation primitive libraries. Why? Because unisim and the others all conform to VHDL'93 (if we're lucky) and I write everything in VHDL 2008. And we can't mix those two standards in ghdl. 

Is there a particular reason why we can't? That would make everyone's lives a lot easier, especially since there's nothing to indicate that the FPGA vendors even care that their models use an obsolete version of the language.

eine, 2020-06-09:
> As for your above example of what the syntax could be, yes mostly the libraries are in my own source

All makes sense now 😃 I was confused with a "regular" unisim. Anyway, the approach of pre-building and using `-P` might be useful for you in other contexts.

> (for example, unisim isn't the Xilinx one, I use it to hold a sim variant of BSCANE2 I wrote which communicates over a network socket :)

You definitely have interesting ideas (#1331, #1335)... Did you see [ghdl/ghdl-cosim](https://github.com/ghdl/ghdl-cosim) (ghdl/ghdl-cosim#1, ghdl/ghdl-cosim#3) and specially [netpp and ghdlex](https://ghdl.github.io/ghdl-cosim/vhpidirect/examples/other.html#ghdlex-and-netpp)?

eine, 2020-06-09:
> Is there a particular reason why we can't? That would make everyone's lives a lot easier, especially since there's nothing to indicate that the FPGA vendors even care that their models use an obsolete version of the language.

@aspdigital, see:

> #980 will hopefully simplify how `*.cf`, `*.o` and `~*.o` are handled with regard to command line controls. This was recently commented by Tristan in the chat; hence it might be the next big thing after (or along with) synthesis.

It is about GHDL supporting mixing standard versions. The reference in the chat is [May 27, 2020 5:51 PM](https://gitter.im/ghdl1/Lobby?at=5ece8c9e3ffa6106f1d9e877).

closed 2020-06-10 by tgingold

tgingold, 2020-06-10:
I am not sure to fully understand you.  If your own design is using VHDL 2008, you can still use unisim.  Just analyze unisim using VHDL 2008 (--std=08).

> Is there a particular reason why we can't? That would make everyone's lives a lot easier, especially since there's nothing to indicate that the FPGA vendors even care that their models use an obsolete version of the language.

The reason you cannot easily mix vhdl93 and vhdl08 (with ghdl) is the ieee packages.  They have different (and incompatible) std_logic_1164 packages.

If the vhdl93 part of a design is using the vhdl93 version of ieee and the vhdl08 part the vhdl08 version of ieee, then you won't be able to mix them because the std_logic types (and the others) won't be the same in both parts.  You could then argue that ghdl should know that this is the same type, but 1) that would require a special circuitry, and 2) this is not so obvious.

If the vhdl93 part was using the vhdl08 ieee packages, some mistakes (like mixing std_logic_vector and std_ulogic_vector) won't be detected.

A possible solution I plan to implement is to analyze as today and then mix using the highest version of ieee packages at the end. So that mistakes would be detected during analysis.  The problem is that this is not really compatible with gcc/llvm backend models, where code is generated during analysis.

commit 47547a13 (2020-06-10): vhdl: allows --work between files for -c.  Fix #1362