ghdl/ghdl #1363: GHDL v. 0.37 for MacOS with LLVM backend

Dotto92, 2020-06-09:
I'm using GHDL v. 0.36 for MacOS with LLVM backend obtained from the pre-built packages page and it work fine.
I wish to try GHDL v. 0.37 with LLVM backend but the pre-built package for MacOS is missing in the repository.
I'm not a software developer and build from source it's very difficult for me (I tried to build it by installing GNAT, but make command returns with error).
I ask if there is a way to obtain a pre-built GHDL for MacOS with LLVM backend.

Thank you in advance

eine, 2020-06-09:
Hi @Dotto92!

We use CI services to build packages with mcode or LLVM backend for Ubuntu and/or Windows. We use those to build macOs packages with mcode backend too. Unfortunately, there is some issue which is preventing us from building all the features. As commented in #867, users have reported successfully building GHDL with LLVM backend on their own machines. Hence, we know it does work, but we are unable to reproduce in CI...

Unfortunately, I don't have a mac myself, so it's hard to debug. @diogratia, who is the user contributing macOs packages available in GitHub Releases, has explained that he cannot debug issues with Catalina (see #1115 and #401).

Should you want to try building it yourself, #959 is the last reference I remember where a user tried to do it.

I'm closing this as a duplicate of #744, but feel free to ask about or post the specific errors you get when trying to build GHDL.

closed 2020-06-09 by eine

Dotto92, 2020-06-09:
Thank you for your answer.