ghdl/ghdl #1364: Parenthesis dependent error : 'others' choice not allowed for an aggregate in this context

ndesgranges, 2020-06-09:
Hello,

**Description**
I've seen close issues, but not exactly this one.
GHDL would analyse without any issue a design with `(others=>(others => '0'))`, but won't analyse one with `((others=>(others => '0')))` (yes the only differences are the extra parenthesis). 

This syntax is useful to initialize an array of buses. The extra parenthesis should not create an issue. (I mean, even if the syntax itself is wrong, parenthesis shouldn't change anything in my understanding of VHDL).

**Expected behaviour**
- What's expected :
`variable <= ((others=>(others => '0')))` should assign 0 to each bit of each bus 

- What happens
`variable <= ((others=>(others => '0')))` won't compile

**How to reproduce?**
Just declare an array of buses and use the syntax `((others=>(others => '0')))` and then issue 
``` bash
ghdl -a -g  [filename.vhd]
```

```vhd :file: mwe.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity mwe is
	port(
		n_rst_i		: in std_logic;
		clk_i     	: in std_logic	
	);
end entity mwe;

architecture arch of mwe is

-- I declare a new type which is an array of buses 
type my_new_type is array(natural range <>) of std_logic_vector(31 downto 0);

-- Then I declare a constant of that new type 
constant constant_of_my_new_type : my_new_type (0 to 7) := ((others=>(others => '0')));

signal signal_of_my_new_type 		: my_new_type (0 to 7);

begin

process(clk_i)
begin
	
	-- if Reset low then signal_of_my_new_type = constant_of_my_new_type
	-- else, signal_of_my_new_type is filled with one at the next clock rising edge

	if n_rst_i = '0' then
		signal_of_my_new_type <= constant_of_my_new_type;
	elsif rising_edge(clk_i) then
		signal_of_my_new_type <= ((others=>(others => '1')));
	end if;
		
end process;

end architecture;

```

> Please note that an easy fix is to delete the extra parenthesis, `((others=>(others => '1')))` would become `(others=>(others => '1'))`, and everything would work correclty

> I'm writing this because it's a bug, not because I can't get around it. However, this is a simple syntax which breaks basic feature around since VHDL 93

> Other simulator won't complain about it.

**Context**
Please, provide the following information:

- OS: Ubuntu 18.04.4 LTS
- Origin:
  - [ ] ~~Package manager~~
  - [ ] ~~Released binaries~~
  - [x] Built from sources: master, commit `dcc6dc4eccea56104bbea43e4407ce8b82dbdab2`

Couldn't find how to get the GHDL bug report. don't hesitate to ask more information.
 
The error is the following :
``` bash
ghdl -a -g  mwe.vhd
```
``` bash
mwe.vhd:18:61:error: 'others' choice not allowed for an aggregate in this context
mwe.vhd:33:42:error: 'others' choice not allowed for an aggregate in this context
ghdl:error: compilation error
```

tgingold, 2020-06-09:
Sorry but ghdl is correct here.

There are only a few places where an aggregate with an 'others' choice can appear.  They are listed in LRM93 7.3.2.2 'Array aggregate', and within parentheses is not in that list.

This is a similar to the reason you can write `case x is` but not `case (x) is` when x is a one-dimensional array.

tgingold, 2020-06-10:
But feel free to argue if you don't agree.

ndesgranges, 2020-06-10:
In IEEE Std 1076-1993, 7.3.2.2 Array aggregates :
> The subtype of an array aggregate that has an others choice must be determinable from the context.  That is, anarray aggregate with an others choice may only appear 

> d) As  a  value  expression  in  an  assignment  statement,  where  the  target  is  a  declared  object,  and  the subtype of the target is a constrained array subtype (or subelement of such a declared object)

> e) As  the  expression  defining  the  initial  value  of  a  constant  or  variable  object,  where  that  object  isdeclared to be of a constrained array subtype

e) suits with the first use in the given mwe and d) suits with the second one. 

Even if I couldn't find anything about parenthesis in that kind of situation, I get your point with the parenthesis, but I've never seen such a behavior in VHDL where parenthesis break a statement. I've always seen `(x)` interpreted as `x` as it's only wrapping it.

About your example. I tried `case (x) is` and `case x is` and both work, I don't get what you mean. See the mwe with both  `case (x) is` and `case x is` : 

``` VHDL
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity mwe is
	port(
		n_rst_i		: in std_logic;
		clk_i     	: in std_logic	
	);
end entity mwe;

architecture arch of mwe is

signal counter 					: unsigned(2 DOWNTO 0);

signal weird_clock 				: std_logic;
signal weird_clock_parenthesis 	: std_logic;

begin

process(clk_i)
begin
	
	-- if Reset low then weird_clock = 0
	-- else, weird_clock is set to one for the counter = 1 or 3 or 6

	if n_rst_i = '0' then
		counter <= (others => '0');
		weird_clock <= '0';
		weird_clock_parenthesis <= '0';
	elsif rising_edge(clk_i) then
		counter <= counter+1;
		case (counter) is
			when "001"  => weird_clock <= '1';
			when "011"  => weird_clock <= '1';
			when "110"  => weird_clock <= '1';
			when others => weird_clock <= '0';
		end case;

		case counter is
			when "010"  => weird_clock_parenthesis <= '1';
			when "101"  => weird_clock_parenthesis <= '1';
			when others => weird_clock_parenthesis <= '0';
		end case;
		
	end if;
		
end process;
```

tgingold, 2020-06-11:
Maybe it should be accepted at least with -frelaxed.

tgingold, 2020-06-11:
> e) suits with the first use in the given mwe and d) suits with the second one.

Not literally.  When expressions within parentheses are allowed, that's mentioned in the LRM.  See for example LRM 7.4.1 j).

ndesgranges, 2020-06-11:
> Not literally. When expressions within parentheses are allowed, that's mentioned in the LRM. See for example LRM 7.4.1 j).

Alright, that something I wasn't sure about, thank you for clarifying. It makes sense then.

> Maybe it should be accepted at least with -frelaxed.

I think this is a good idea : 
I initially opened this issue because the buggy code is synthesizable, it actually works on a xilinx card (didn't try with the mwe above though) and I could simulate it with another tool (which might be too permissive according to the LRM, I agree).

tgingold, 2020-06-13:
Now fixed and part of `-frelaxed`.  Thank you for the issue.

closed 2020-06-13 by tgingold