ghdl/ghdl #1368: Cannot build v. 0.37 with LLVM in macOS 10.15

Dotto92, 2020-06-11:
I'm trying to build GHDL v. 0.37 from source with LLVM backend in macOS 10.15 (Catalina) because is missing in the pre-built packages.
I created a build folder inside the sources folder:

mkdir build
cd build

I configured with the following command:

`./configure --with-llvm-config=/usr/local/Cellar/llvm/10.0.0_3/bin/llvm-config --disable-werror`

and the following message returns:

```
Use full IEEE library
Build machine is: x86_64-apple-darwin17.7.0
Unhandled version llvm 10.0.0
```

Following the indications on #949, I made two changes in the generated makefile:

from CC=gcc to CC=clang

and from this command:

```
brew info llvm

....

For compilers to find llvm you may need to set:
  export LDFLAGS="-L/usr/local/opt/llvm/lib"
  export CPPFLAGS="-I/usr/local/opt/llvm/include"
....
```

LDFLAGS= -L/usr/local/opt/llvm/lib,-Wl,-no_compact_unwind

Saved the file and

`make`

After few minutes the following error returns:

```
....
gcc -c -I./ -gnat12 -I../src -I../src/vhdl -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/mcode -I../src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A -I- /Users/Calimero/Desktop/ghdl-0.37/src/grt/grt-shadow_ieee.adb
gcc -c -I./ -gnat12 -I../src -I../src/vhdl -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/mcode -I../src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A -I- /Users/Calimero/Desktop/ghdl-0.37/src/grt/grt-stats.adb
gcc -c -I./ -gnat12 -I../src -I../src/vhdl -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/mcode -I../src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A -I- /Users/Calimero/Desktop/ghdl-0.37/src/grt/grt-disp_rti.adb
grt-disp_rti.adb:895:40: warning: writable actual for "Obj_Loc" overlaps with actual for "Addr"
gnatmake: "/Users/Calimero/Desktop/ghdl-0.37/src/grt/grt-disp_rti.adb" compilation error
make: *** [ghdl_mcode] Error 4
```

eine, 2020-06-11:
Hi @Dotto92! Overall, you are doing it good. These are typical details that need to be sorted.

> Unhandled version llvm 10.0.0

This is because you are trying to build `v0.37`. LLVM 10 was not supported when that release whan published/tagged back in february (see #1213). I suggest you to clone/download branch `master` instead. GHDL is a "rolling" project, so it is not more risky building master than v0.37.

> Following the indications on #949, I made two changes in the generated makefile:
> 
> from CC=gcc to CC=clang

I think you can achieve the same result without changing the makefile. Just define the envvars right before calling make: `CC=gcc to CC=clang make`.

> LDFLAGS= -L/usr/local/opt/llvm/lib,-Wl,-no_compact_unwind

I think that this should not be required, but it should not be harmful anyway.

> After few minutes the following error returns

That's in fact not an error, but a warning. It fails because, by default, compilation fails on warnings. You can disable it by configuring with `-Wno-error` (see https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html?highlight=warning#warnings and https://github.com/ghdl/ghdl/issues/1339#issuecomment-633001993). Nevertheless, that specific warning should be gone in `master`.

Dotto92, 2020-06-11:
Thank you for your answer.
I tried to build from the branch master and the message

`Unhandled version llvm 10.0.0`

Desappeared. 
I used the configure command:

`../configure --with-llvm-config=/usr/local/Cellar/llvm/10.0.0_3/bin/llvm-config --disable-werror`
 
because configuring with -Wno-error option returns:

`../configure: unknown option -Wno-error; try ../configure --help`

and that option seems to be unavailable:

```
Usage: configure [options]

Options [defaults in brackets]:
  --prefix=PREFIX             install in PREFIX [/usr/local]
  --srcdir=SRCDIR             source code path [..]
  --with-gcc=DIR              use gcc backend from DIR
  --with-llvm-config[=LLVM-CONFIG]  use llvm backend [llvm-config]
  --with-backtrace-lib=LIB.a  link with libbacktrace LIB.a to display a
                              backtrace on errors (only for llvm).
  --disable-werror            warnings don't stop build
  --disable-checks            disable internal checks
  --disable-libghdl           do not build libghdl shared library
  --enable-openieee           use gpl-compatible sources for ieee library
  --enable-libghdl            also build the libghdl shared library
  --disable-synth             do not build with synthesis feature
```

Now, at the make command, another error returns:

```
....

gcc -c -I./ -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm4-nodebug -I../src/synth -I../src/ghdldrv -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -I- /Users/Calimero/Desktop/ghdl-master/src/synth/synth-ieee-std_logic_1164.adb
gnatbind -aI../src -aI../src/vhdl -aI../src/synth -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm4-nodebug -aI../src/synth -aI../src/ghdldrv -E -x ghdl_llvm.ali
gnatlink ghdl_llvm.ali -g -Wl,-no_compact_unwind grt-cstdio.o
ld: library not found for -lSystem
collect2: error: ld returned 1 exit status
gnatlink: error when calling /Users/Calimero/opt/GNAT/2020/bin/gcc
gnatmake: *** link failed.
make: *** [ghdl_llvm] Error 4
```

tgingold, 2020-06-11:
> gcc -c -I./ -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm4-nodebug -I../src/synth -I../src/ghdldrv [...]

Not too bad.  The error message is surprizing as libSystem is the system library.

Can you try:
gnatlink -v -v ghdl_llvm.ali -g -Wl,-no_compact_unwind grt-cstdio.o

Dotto92, 2020-06-11:
I ran the command you suggested; it returns:

```
GNATLINK Community 2020 (20200429-84)
Copyright (C) 1995-2020, Free Software Foundation, Inc.
gcc -c -g -gnatA -gnatWb -gnatiw -v -g -gnatws /Users/Calimero/Desktop/ghdl-master/build/b~ghdl_llvm.adb
Using built-in specs.
COLLECT_GCC=/Users/Calimero/opt/GNAT/2020/bin/gcc
Target: x86_64-apple-darwin17.7.0
Configured with: ../src/configure --enable-languages=ada,c,c++,objc --enable-checking=release --enable-lto --with-stage1-ldflags=-static-libstdc++ --with-boot-ldflags=-static-libstdc++ --disable-libcilkrts --with-build-sysroot=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk --with-sysroot= --with-bugurl=URL:mailto:report@adacore.com --disable-nls --without-libiconv-prefix --disable-libstdcxx-pch --disable-libada --disable-multilib --with-specs='%{!sysroot=*:--sysroot=%:xcode-path(/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk)}' --with-mpfr=/gnatmail/sandbox/a2c2/x86_64-darwin/mpfr_stable-c/install --with-gmp=/gnatmail/sandbox/a2c2/x86_64-darwin/gmp_stable-c/install --with-mpc=/gnatmail/sandbox/a2c2/x86_64-darwin/mpc_stable-c/install --with-build-time-tools=/gnatmail/sandbox/a2c2/x86_64-darwin/gcc-c/build/buildtools/bin --prefix=/gnatmail/sandbox/a2c2/x86_64-darwin/gcc-c/pkg --build=x86_64-apple-darwin17.7.0
Thread model: posix
gcc version 8.4.1 20200430 (for GNAT Community 2020 20200429) (GCC) 
COLLECT_GCC_OPTIONS='-c' '-g' '-gnatA' '-gnatWb' '-gnatiw' '-v' '-g' '-gnatws' '-mtune=core2' '-mmacosx-version-min=10.15.0'
 /Users/Calimero/opt/GNAT/2020/bin/../libexec/gcc/x86_64-apple-darwin17.7.0/8.4.1/gnat1 -quiet -dumpbase b~ghdl_llvm.adb -auxbase b~ghdl_llvm -g -gnatA -gnatWb -gnatiw -g -gnatws -mtune=core2 -mmacosx-version-min=10.15.0 -fPIC -feliminate-unused-debug-symbols /Users/Calimero/Desktop/ghdl-master/build/b~ghdl_llvm.adb -o /var/folders/s0/ww4dv8g91t75yxl2p287g7tr0000gn/T//ccJrsGjR.s
COLLECT_GCC_OPTIONS='-c' '-g' '-gnatA' '-gnatWb' '-gnatiw' '-v' '-g' '-gnatws' '-mtune=core2' '-mmacosx-version-min=10.15.0'
 as -arch x86_64 -v -force_cpusubtype_ALL -mmacosx-version-min=10.15 -o b~ghdl_llvm.o /var/folders/s0/ww4dv8g91t75yxl2p287g7tr0000gn/T//ccJrsGjR.s
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
 "/Library/Developer/CommandLineTools/usr/bin/clang" -cc1as -triple x86_64-apple-macosx10.15.0 -filetype obj -main-file-name ccJrsGjR.s -target-cpu penryn -fdebug-compilation-dir /Users/Calimero/Desktop/ghdl-master/build -dwarf-debug-producer "Apple clang version 11.0.3 (clang-1103.0.32.62)" -dwarf-version=4 -mrelocation-model pic -o b~ghdl_llvm.o /var/folders/s0/ww4dv8g91t75yxl2p287g7tr0000gn/T//ccJrsGjR.s
COMPILER_PATH=/Users/Calimero/opt/GNAT/2020/bin/../libexec/gcc/x86_64-apple-darwin17.7.0/8.4.1/:/Users/Calimero/opt/GNAT/2020/bin/../libexec/gcc/
LIBRARY_PATH=/Users/Calimero/opt/GNAT/2020/bin/../lib/gcc/x86_64-apple-darwin17.7.0/8.4.1/:/Users/Calimero/opt/GNAT/2020/bin/../lib/gcc/:/Users/Calimero/opt/GNAT/2020/bin/../lib/gcc/x86_64-apple-darwin17.7.0/8.4.1/../../../
COLLECT_GCC_OPTIONS='-c' '-g' '-gnatA' '-gnatWb' '-gnatiw' '-v' '-g' '-gnatws' '-mtune=core2' '-mmacosx-version-min=10.15.0'
/Users/Calimero/opt/GNAT/2020/bin/gcc b~ghdl_llvm.o grt-cstdio.o ./grt.o ./grt-c.o ./grt-fcvt.o ./grt-severity.o ./grt-stdio.o ./grt-types.o ./grt-vstrings.o ./psl.o ./synth.o ./synth-ieee.o ./types.o ./vhdl.o ./vhdl-nodes_priv.o ./vhdl-tokens.o ./vhdl-types.o ./areapools.o ./default_paths.o ./dyn_tables.o ./flags.o ./grt-algos.o ./grt-table.o ./grt-files_operations.o ./grt-to_strings.o ./hash.o ./dyn_maps.o ./dyn_interning.o ./interning.o ./mutils.o ./psl-nodes_priv.o ./psl-priorities.o ./psl-types.o ./simple_io.o ./logging.o ./synth-flags.o ./tables.o ./flists.o ./lists.o ./name_table.o ./std_names.o ./str_table.o ./files_map.o ./errorout.o ./errorout-console.o ./psl-errors.o ./psl-hash.o ./psl-nodes_meta.o ./psl-nodes.o ./psl-dump_tree.o ./psl-nfas.o ./psl-nfas-utils.o ./psl-prints.o ./psl-cse.o ./psl-disp_nfas.o ./psl-optimize.o ./psl-qm.o ./psl-build.o ./psl-rewrites.o ./psl-subsets.o ./synth-errors.o ./types_utils.o ./netlists-gates.o ./netlists-utils.o ./netlists.o ./netlists-builders.o ./netlists-cleanup.o ./netlists-gates_ports.o ./netlists-internings.o ./netlists-iterators.o ./netlists-locations.o ./netlists-butils.o ./netlists-errors.o ./netlists-folds.o ./netlists-concats.o ./utils_io.o ./netlists-disp_vhdl.o ./netlists-dump.o ./netlists-disp_dot.o ./version.o ./bug.o ./vhdl-flists.o ./vhdl-lists.o ./vhdl-nodes_meta.o ./vhdl-nodes.o ./synth-objtypes.o ./synth-ieee-std_logic_1164.o ./vhdl-back_end.o ./vhdl-disp_tree.o ./vhdl-elocations_meta.o ./vhdl-elocations.o ./vhdl-nodes_utils.o ./vhdl-ieee-std_logic_1164.o ./vhdl-utils.o ./vhdl-errors.o ./vhdl-ieee.o ./vhdl-sem_utils.o ./vhdl-std_package.o ./synth-source.o ./netlists-memories.o ./netlists-inference.o ./netlists-expands.o ./synth-ieee-numeric_std.o ./vhdl-annotations.o ./vhdl-ieee-math_real.o ./vhdl-ieee-numeric.o ./vhdl-ieee-std_logic_arith.o ./vhdl-ieee-std_logic_misc.o ./vhdl-ieee-std_logic_unsigned.o ./vhdl-nodes_walk.o ./vhdl-prints.o ./vhdl-scanner.o ./vhdl-evaluation.o ./vhdl-formatters.o ./vhdl-sem_inst.o ./vhdl-sem_scopes.o ./vhdl-xrefs.o ./vhdl-parse_psl.o ./vhdl-parse.o ./options.o ./libraries.o ./ghdlmain.o ./vhdl-nodes_gc.o ./vhdl-canon_psl.o ./vhdl-sem_lib.o ./vhdl-sem_specs.o ./vhdl-canon.o ./vhdl-ieee-vital_timing.o  [... cut ...]
configured to support archs: armv6 armv7 armv7s arm64 arm64e arm64_32 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em
Library search paths:
	./
	/Users/Calimero/opt/GNAT/2020/lib/gcc/x86_64-apple-darwin17.7.0/8.4.1/adalib/
	/Users/Calimero/opt/GNAT/2020/lib/gcc/x86_64-apple-darwin17.7.0/8.4.1
	/Users/Calimero/opt/GNAT/2020/lib/gcc
	/Users/Calimero/opt/GNAT/2020/lib
Framework search paths:
ld: library not found for -lSystem
collect2: error: ld returned 1 exit status
gnatlink: error when calling /Users/Calimero/opt/GNAT/2020/bin/gcc
```

tgingold, 2020-06-12:
Two possible solutions:

1) Add `-L/usr/lib`
2) use `gnatlink --LINK=clang ...`

Dotto92, 2020-06-12:
Both solutions works!
I needed to use the command two times:
```
....

gnatbind -aI../src -aI../src/vhdl -aI../src/synth -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm4-nodebug -aI../src/synth -aI../src/ghdldrv -E -x ghdl_llvm.ali
gnatlink ghdl_llvm.ali -g -Wl,-no_compact_unwind grt-cstdio.o
ld: library not found for -lSystem
collect2: error: ld returned 1 exit status
gnatlink: error when calling /Users/Calimero/opt/GNAT/2020/bin/gcc
gnatmake: *** link failed.
make: *** [ghdl_llvm] Error 4

gnatlink --LINK=clang ghdl_llvm.ali -g -Wl,-no_compact_unwind grt-cstdio.o

make

....

gnatbind -aI. -aO/Users/Calimero/Desktop/ghdl-master/build/pic -aI../src -aI../src/vhdl -aI../src/synth -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/synth -aI../src/vhdl/libghdl -aI../src/ghdldrv -shared -Llibghdl_ -z -x /Users/Calimero/Desktop/ghdl-master/build/pic/libghdl.ali
gnatlink /Users/Calimero/Desktop/ghdl-master/build/pic/libghdl.ali -shared-libgcc -o libghdl-1_0_dev.dylib -g -g -shared -Wl,-install_name,@rpath/libghdl-1_0_dev.dylib pic/grt-cstdio.o
ld: library not found for -lSystem
collect2: error: ld returned 1 exit status
gnatlink: error when calling /Users/Calimero/opt/GNAT/2020/bin/gcc
gnatmake: *** link failed.
make: *** [libghdl-1_0_dev.dylib] Error 4

gnatlink --LINK=clang /Users/Calimero/Desktop/ghdl-master/build/pic/libghdl.ali -shared-libgcc -o libghdl-1_0_dev.dylib -g -g -shared -Wl,-install_name,@rpath/libghdl-1_0_dev.dylib pic/grt-cstdio.o

make

....

make install
```

It complete the operation without any other errors and it seems work:

```
./ghdl -v
GHDL 1.0-dev () [Dunoon edition]
 Compiled with GNAT Version: Community 2020 (20200429-84)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I also tried to simulate a simple project in VHDL and it works well.

I really appreciated your help.

Many thanks and regards.

tgingold, 2020-06-13:
Thank you for the status update.  I close this issue.

closed 2020-06-13 by tgingold

focussed, 2020-09-13:
On Mojave 10.15.6, the build fails looking for lz  (zlib) and later -lSystem
By adding 
LDFLAGS=-L/usr/lib
In the main Makefile removes the first issue.

eine, 2020-09-14:
Ref #1115.