ghdl/ghdl #1369: FTBFS on POWER systems

madscientist159, 2020-06-11:
**Description**
GHDL fails to build on POWER systems.  When I attempt to compile on my Blackbird desktop (POWER9), I receive the following error:

```
ghdl$ make -j8
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o jumps.o /redacted/ghdl/./src/grt/config/jumps.c
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o times.o /redacted/ghdl/./src/grt/config/times.c
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o grt-cstdio.o /redacted/ghdl/./src/grt/grt-cstdio.c
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o grt-cgnatrts.o /redacted/ghdl/./src/grt/grt-cgnatrts.c
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o grt-cvpi.o /redacted/ghdl/./src/grt/grt-cvpi.c
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o grt-cdynload.o /redacted/ghdl/./src/grt/grt-cdynload.c
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o fstapi.o /redacted/ghdl/./src/grt/fst/fstapi.c -I/redacted/ghdl/./src/grt/fst
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o lz4.o /redacted/ghdl/./src/grt/fst/lz4.c
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o fastlz.o /redacted/ghdl/./src/grt/fst/fastlz.c
echo "with Grt.Backtraces.Jit;" > grt-backtraces-impl.ads
echo "package Grt.Backtraces.Impl renames Grt.Backtraces.Jit;" >> grt-backtraces-impl.ads
gcc -c -g -o memsegs_c.o src/ortho/mcode/memsegs_c.c
gcc -c -g -fPIC -DWITH_GNAT_RUN_TIME -o chkstk.o /redacted/ghdl/./src/grt/config/chkstk.S
chkstk.S: Assembler messages:
chkstk.S:41: Error: unrecognized opcode: `testl'
chkstk.S:42: Error: unrecognized opcode: `je'
chkstk.S:43: Error: unrecognized opcode: `subl'
chkstk.S:44: Error: unrecognized opcode: `subl'
chkstk.S:45: Error: unrecognized opcode: `jmp'
chkstk.S:47: Error: unrecognized opcode: `ret'
make: *** [src/grt/Makefile.inc:228: chkstk.o] Error 1
```

**Expected behaviour**
GHDL compiles without erroring out.

**How to reproduce?**
Clone the latest repository from GIT on a POWER system.  Run `./configure --prefix=/usr` and `make`.

**Context**
Please, provide the following information:

- OS: Debian Buster
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `aa67933c715cc0de09131c9b5852aaffb629c3aa`

madscientist159, 2020-06-11:
Note that Debian offers a package of ghdl that is built for POWER -- and I can confirm it builds from source there, despite the offending chkstk.S file being present.  Fixing this should be as simple as determining what flags Debian is passing to non-x86 systems and making sure `./configure` automatically uses those flags if it detects it is being run on a non-x86 platform.

tgingold, 2020-06-12:
Sorry, but I don't have a POWER system to test any change.

The default backend, mcode, doesn't support POWER.  So you need to configure ghdl either for llvm or for gcc.

The error you get is very strange, as chkstk.S is not compiled anymore on any system.

madscientist159, 2020-06-12:
@tgingold No problem, I'm happy to debug if you can point me in the right direction, which I think you did with the backend information.  Passing `--with-llvm-config` is at least allowing it to build so far; I'll report back once it's done.

What would the correct way be to change the default backend in the source files?  I can glue in architecture detection if I know where to change the defaults.

Thanks!

tgingold, 2020-06-12:
That's done in `configure` which is a manually written file (and thus understandable).
I don't think you can easily change the default backend, because both llvm and gcc needs additional info.  Well, maybe with llvm you can use the installed library.

But your report is strange: normally non x86 platforms are rejected as not supported when mcode is selected.

What is the output of `gcc -dumpmachine` ?

madscientist159, 2020-06-12:
Yeah, it worked right out of the box with no argument on the LLVM flag.

```
gcc -dumpmachine
powerpc64le-linux-gnu
```

tgingold, 2020-06-12:
Great, so I close this issue.

closed 2020-06-12 by tgingold

umarcor, 2020-06-12:
> Sorry, but I don't have a POWER system to test any change.

In hub.docker.com, there are some "*other architectures built by official images: (but not officially supported by Docker, Inc.)"* (https://github.com/docker-library/official-images#architectures-other-than-amd64). By using QEMU, it is possible to run ppc64 containers on x86_64 CI services, as shown in [dbhi/qus](https://github.com/dbhi/qus).

dbhi/qus is already used in [dbhi/docker](https://github.com/dbhi/docker) to produce multiarch manifests of images containing GHDL, Octave and other tools. These are for x86_64, arm32v7 and arm64v8 only, for now, because I don't have any use case for other platforms.

The point is that using QEMU introduces a x8-x10 time penalty. E.g., "main" jobs in https://github.com/dbhi/docker/runs/744178632?check_suite_focus=true take 2 min on x86_64, but 17-20 min on (emulated) ARM. Nonetheless, it would be possible to extend GHDL CI tests to evaluate platforms other than x86_64.