ghdl/ghdl #1373: Commit e0ca8bf0b0d8049c7e9f7fe53f7ed942fac61453 breaks some vendor library builds

AppMathIsRealMath, 2020-06-16:
**Description**
I rebuild the xilinx, osvvm, and uvvm vendor libraries after each set of ghdl commits I update. Using gnat-2020 CE I execute a personal script containing

./compile-osvvm.sh -a
./compile-uvvm.sh -a
./compile-xilinx-vivado.sh -a --vhdl2008

This procedure successfully builds these libraries (used by me in emacs running flycheck for VHDL linting).

After commit e0ca8bf0b0d8049c7e9f7fe53f7ed942fac61453 

osvvm builds successfully but with a set of different warnings
uvvm fails with tons of warnings
xilinx-vivado fails with a bunch of new warnings similar to
../../src/vital2000/timing_p.vhdl:152:14: result subtype of a pure function cannot have access subelements
../../src/vital2000/timing_p.vhdl:152:14:note: (you can use -frelaxed to turn this error into a warning)
/home/sam/mnt/tools/xilinx/Vivado/2020.1/data/vhdl/src/unisims/unisim_VPKG.vhd:583:12:warning: declaration of "s" hides constant "s" [-Whide]

As a workaround I've created a branch at the last working commit 8eb418a695000718920c2ef5294c185a2ef40cca

Do the vendor lib build scripts need to be updated to account for commit e0ca8bf0b0d8049c7e9f7fe53f7ed942fac61453?

tgingold, 2020-06-16:
Should be fixed with the last commit.

AppMathIsRealMath, 2020-06-16:
Fix confirmed for osvvm, uvvm and xilinx-vivado (I don't build the others). Thanks!

tgingold, 2020-06-17:
Thank you for the check.

closed 2020-06-17 by tgingold