ghdl/ghdl #1374: BUG report raised  vhdl-sem_scopes.adb:469

eduardomdalmaso, 2020-06-17:
**Description**
Got a bug when try to use ghdl -s my_program.vhd

**Expected behaviour**
What you expected to happen, and what is happening instead.

**How to reproduce?**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (tarball) [Dunoon edition]
Compiled with GNAT Version: 10.1.0
Target: x86_64-pc-linux-gnu
/home/eduardomd/UVV/eletronica_digital_ll/
Command line:
ghdl -s maq_est00.vhdl
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-sem_scopes.adb:469
Call stack traceback locations:
0x58bf32 0x4ccad0 0x4fcbf6 0x500f32 0x4eec25 0x557a1c 0x557b18 0x4ecd7f 0x407169 0x7f43bdbb2000 0x4071dc 0xfffffffffffffffe
******************************************************************

```vhd :file: ent.vhd
library.ieee;
use sdt_logic_1164.all;

entity maq_est00 is
	port(
	    	CLK, RST, BT 	: in	std_logic;
		Q		: out 	std_logic
	    );
end maq_est00;

architecture box of maq_est00 is

type tipo_estado is (s0,s1,s2,s3);
signal estado_atual 	:	tipo_estado := s0;
signal proximo_estado 	:	tipo_estado := s0;

begin

	process(CLK,RST) --estado inicial
	begin
		if(RST='0') then 
			estado_atual 	<= s0;
			proximo_estado 	<= s0;
		elsif(CLK'event and CLK='1') then
			estado_atual <= proximo_estado;
		end if;
	end process;

	process(Q,estado_atual) --transição de estados
	begin
		case(estado_atual) is
			when s0 =>	if (BT='1') then
						proximo_estado <= s1;
					else
						proximo_estado <= s2;
					end if;
				
			when s1 =>	if (BT='1') then
						proximo_estado <= s1;
					else
						proximo_estado <= s2;
					end if;
			when s2 =>	if (BT='1') then
						proximo_estado <= s3;
					else
						proximo_estado <= s1;
					end if;
			when s3 => 	if (BT='1') then
						proximo_estado <= RST;
					else
						proximo_estado <= s0;
					end if;
			when others =>		proximo_estado <= s0;  
		end case;
	end process;

end box;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: 5.7.2-arch1-1
- Origin:
  - [ X] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tmeissner, 2020-06-17:
GHDL should not crash, but `proximo_estado <= RST; ` seems like an assignment of incompatible types.

tgingold, 2020-06-17:
Sorry, but I cannot reproduce the crash.  I get:

ent.vhdl:1:8:error: an identifier is expected instead of '.'
ent.vhdl:1:8:error: missing ";" at end of library clause
ent.vhdl:1:8:error: missing entity, architecture, package or configuration

eduardomdalmaso, 2020-06-17:
Ok i will check, thanks.

Enviada do Outlook Mobile<https://aka.ms/blhgte>

eine, 2020-06-18:
@eduardomdalmaso, I'd say that `proximo_estado 	<= s0;` should be removed from the first process. `proximo_estado` (*next state* for non-latin speakers), is assigned in the second process, and it is not a register.

eduardomdalmaso, 2020-06-18:
I already fix, im really sorry for the trouble, it was just syntax error.

Thanks,

Enviada do Outlook Mobile<https://aka.ms/blhgte>

eduardomdalmaso, 2020-06-18:
Fixed.

closed 2020-06-18 by eduardomdalmaso