ghdl/ghdl #1375: Crash when compiling UVVM/bitvis_vip_hvvc_to_vvc_bridge/src/support_pkg.vhd

geddy11, 2020-06-19:
**Description**
GHDL crashes with "GHDL Bug occurred".

**Expected behaviour**
No crashing.

**How to reproduce?**
Trying to compile the testbench in UVVM/bitvis_vip_ethernet/tb

**Context**
Please, provide the following information:

- OS: Debian GNU/Linux 10 (buster) in Docker on Win10  
Docker image: ghdl/ext:1.0

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev () [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
/home/ghdl-flow/script/
Command line:
/usr/local/bin/ghdl1-llvm -frelaxed-rules --std=08 -Wno-hide -Wno-shared --workdir=bitvis_vip_hvvc_to_vvc_bridge --work=bitvis_vip_hvvc_to_vvc_bridge -Puvvm_util/ -Pwork/ -Puvvm_vvc_framework/ -Pbitvis_vip_hvvc_to_vvc_bridge/ -Pbitvis_vip_ethernet/ -Pbitvis_vip_scoreboard/ -Pbitvis_vip_gmii/ -Pbitvis_vip_sbi/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o bitvis_vip_hvvc_to_vvc_bridge/support_pkg.o ../UVVM/bitvis_vip_hvvc_to_vvc_bridge/src/support_pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:48
Call stack traceback locations:
0x62981a 0x62dd3f 0x62e063 0x62e4b5 0x63b498 0x6225ec 0x623b94 0x626113 0x5b4e66 0x5b5bf6 0x5b971c 0x5b9887 0x641d07 0x5e0aa4 0x642f16 0x58f04d 0x6523a0 0x653cb4 0x6555ad 0x409e35 0x7f4810f33099 0x409118 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-06-28:
I confirm the issue.

tgingold, 2020-08-04:
Now fixed.  Thank you for the issue.

closed 2020-08-04 by tgingold