ghdl/ghdl #1376: Inconsistency in Generation of SLV Windows due to Unbounded SLV Array Input

RocketRoss, 2020-06-19:
**Description**
A bit difficult to sum up in the title, but I have tried to be as accurate as possible.

In creating an entity that takes in an array of SLVs and outputs windows of the input (as in convolution), the output's values **sometimes** do not match the corresponding input's value.

The entity is purely combinatorial, with only a set of generate loops.

The testbench verifies that the input signals have not changed after the delay, ensuring that the issue is in the processing. The check uses the exact same code as the entity's generation, just with an assert instead of signal assignment.

This issue does not occur when the input slv_array is fixed (both array and slv), nor does it occur when the entity operates on an integer array.

The issue does occur when the entity's input SLV definition is unbounded. ```i_in : in slv_array_t;```

It does not occur when the entity's input SLV definition is bounded. (The input type is unconstrained, but generics bound the dimensions of the type.) ```i_in : in slv_array_t(0 to INPUT_SIZE-1)(BIT_WIDTH-1 downto 0);```

The MWE below has difference between the above 2 scenarios commented out.

When the issue occurs, the SLV that is wrong is consistent through multiple repeated assignments of the same input.

**Expected behaviour**
I know that unconstrained arrays/records aren't really supported yet, but it is curious that they were almost flawless in this case. With the output port I had to bound the definition to get any assignments to work correctly.

I expected that the lack of bounds would produce an error if they were not fully supported, or that the unbounded definitions would work correctly.

I did not expect that the lack of bounds would work **most** of the time.

**How to reproduce?**

**pkg.vhd**

```vhd :file: pkg.vhd
package int_arr is
	type int_array_t is array(natural range <>) of integer;
end package;

library ieee;
use ieee.std_logic_1164.all;

package slv_arr is
	type slv_array_t is array(natural range <>) of std_logic_vector;
	type slv_arr_array_t is array(natural range <>) of slv_array_t;
end package ;
```

**window_splitter_1D.vhd**

```vhd :file: window_splitter_1D.vhd
library ieee;
use ieee.std_logic_1164.all;

use work.int_arr.all;
use work.slv_arr.all;

entity window_splitter_1D is
	generic (
		-- INPUT_SIZE : integer;
		WINDOW_COUNT : integer;
		WINDOW_SIZE : integer;
		BIT_WIDTH : integer;
		WINDOW_STRIDE : integer
	);
	port (
		i_in 	: in slv_array_t;--(0 to INPUT_SIZE-1)(BIT_WIDTH-1 downto 0);
		o_windows : out slv_array_t(0 to WINDOW_COUNT*WINDOW_SIZE-1)(BIT_WIDTH-1 downto 0)
	);

	constant IN_LEN1 : integer := i_in'length;
	constant WINDOW_COUNT_1 : integer := 1 + (IN_LEN1 - WINDOW_SIZE)/WINDOW_STRIDE;
end entity;

architecture comb of window_splitter_1D is
begin
	gen_w1 : for c1 in 0 to WINDOW_COUNT_1-1 generate
		gen_wj : for j in 0 to WINDOW_SIZE-1 generate
			o_windows(c1*WINDOW_SIZE + j) <= i_in(c1*WINDOW_STRIDE+j);
		end generate gen_wj;
	end generate gen_w1;
end architecture;
-----------------------------------------------------------
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.int_arr.all;
use work.slv_arr.all;

entity window_splitter_1D_tb is
	generic (
		WINDOW_SIZE 	: integer := 3;
		WINDOW_STRIDE : integer := 1;
		IN_SIZE 	: integer := 6; 
		BIT_WIDTH : positive := 16
	);

	constant WINDOW_COUNT : integer := 1 + (IN_SIZE - WINDOW_SIZE)/WINDOW_STRIDE;

	function rand return integer is
	begin assert false report "VHPIDIRECT rand" severity failure; end;
	attribute foreign of rand : function is "VHPIDIRECT rand";

end entity;

architecture behav of window_splitter_1D_tb is
	signal s_in : slv_array_t(0 to IN_SIZE-1)(BIT_WIDTH-1 downto 0) := (others => (others => '1'));
	signal s_windows : slv_array_t(0 to WINDOW_COUNT*WINDOW_SIZE-1)(BIT_WIDTH-1 downto 0) := (others => (others => '0'));
begin
	s : entity work.window_splitter_1D
		generic map (	
			-- INPUT_SIZE => IN_SIZE,
			WINDOW_COUNT => WINDOW_COUNT,
			BIT_WIDTH => BIT_WIDTH,
			WINDOW_SIZE => WINDOW_SIZE,
			WINDOW_STRIDE => WINDOW_STRIDE
		)
		port map (
			i_in => s_in,
			o_windows => s_windows
		);

	process
		variable v_in : int_array_t(0 to IN_SIZE-1) := (others => 0);
		variable v_windows : int_array_t(0 to WINDOW_COUNT*WINDOW_SIZE-1) := (others => 0);
		variable v_in_val : integer := 0;
	begin

		for t in 1 to 100 loop

			for i in 0 to IN_SIZE-1 loop
				v_in(i) := rand mod 256 - 128;
				s_in(i) <= std_logic_vector(to_signed(v_in(i), BIT_WIDTH));
			end loop;
			
			wait for 10 ns;

			for i in 0 to IN_SIZE-1 loop
				v_in_val := to_integer(signed(s_in(i)));
				assert v_in_val = v_in(i) report "[" & integer'image(i) & "] = " & integer'image(v_in_val) & " != " & integer'image(v_in(i));
			end loop;

			for c1 in 0 to WINDOW_COUNT-1 loop
				for j in 0 to WINDOW_SIZE-1 loop
					v_in_val := to_integer(signed(s_windows(c1*WINDOW_SIZE + j)));
					assert s_windows(c1*WINDOW_SIZE + j) = s_in(c1*WINDOW_STRIDE+j) report "Test " & integer'image(t) & " 'window_splitter_1D': incorrect SLV on output."; 
					-- assert v_in_val = v_in(c1*WINDOW_STRIDE+j) 
					-- 	report "Test " & integer'image(t) & " 'window_splitter_1D': Window(" & integer'image(c1) & ")(" & integer'image(j) & ") = " 
					-- 									& integer'image(v_in_val) & " != " & integer'image(v_in(c1*WINDOW_STRIDE)) & " In(" & integer'image(c1*WINDOW_STRIDE+j) & ")." severity note;
				end loop;
			end loop;

		end loop;

		report "Test 'window_splitter_1D': Success.";
		wait;
	end process;

end architecture;
```

**run.sh**

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a pkg.vhd
ghdl -a window_splitter_1D.vhd
ghdl -r window_splitter_1D_tb
```

- OS: Ubuntu WSL
- Origin:
GHDL 1.0-dev (v0.37.0-760-g1666a4c7-dirty) [Dunoon edition]
 Compiled with GNAT Version: 7.5.0
 llvm code generator

tgingold, 2020-06-27:
I can reproduce the issue and I am investigating.

closed 2020-06-27 by tgingold

RocketRoss, 2020-06-28:
Further confirmation of the solution in my original 2D case.

Also the [little fix](e54549c2265afce5456046e1a98f4d93be80235e) enables one to leave the ```out``` port unbounded.

Many thanks!

commit e54549c2 (2020-06-27): trans-chap3: propagate object kind to bounds mnode.  Fix #1376