ghdl/ghdl #1378: PSL: endpoints only working inside PSL comments, not in inline mode

tmeissner, 2020-06-21:
**Description**
PSL endpoints seem only work inside PSL comments with the `-fpsl` option. But they should also work with inline PSL supported since VHDL-08.

**Expected behaviour**
PSL endpoints should also work with inline PSL supported since VHDL-08.

**How to reproduce?**
Get this file: https://github.com/tmeissner/psl_with_ghdl/blob/master/issues/issue_134x.vhd

The error occurs in simulation. I will test synthesis later.

compile the file:

```
ghdl -a --std=08 -fpsl issue_1345.vhd
ghdl -e --std=08 -fpsl issue
```

works, but when you comment in ENDPOINT_0_e

```vhdl
-- Doesn't work with inline PSL
endpoint ENDPOINT_0_e is {a; b[*2]; c};
```

you get an error like this:

```
$ ghdl -a --std=08 -fpsl issue_137x.vhd         
issue_137x.vhd:92:12:error: '<=' is expected instead of "endpoint_0_e"
issue_137x.vhd:92:24:error: ';' expected at end of signal assignment
issue_137x.vhd:92:24:error: (found: 'is')
issue_137x.vhd:92:25:error: unexpected token 'is' in a concurrent statement list
issue_137x.vhd:92:34:error: ']' is expected instead of '*'
issue_137x.vhd:92:35:error: signature name not expected here
issue_137x.vhd:92:35:error: '<=' is expected instead of '<integer>'
issue_137x.vhd:92:36:error: ';' expected at end of signal assignment
issue_137x.vhd:92:36:error: (found: ']')
issue_137x.vhd:92:36:error: unexpected token ']' in a concurrent statement list
issue_137x.vhd:92:40:error: '<=' is expected instead of '}'
issue_137x.vhd:92:40:error: unexpected token '}' in a primary
issue_137x.vhd:92:40:error: ';' expected at end of signal assignment
issue_137x.vhd:92:40:error: (found: '}')
issue_137x.vhd:92:40:error: unexpected token '}' in a concurrent statement list
ghdl:error: compilation error
```

**Context**

- OS: Debian Buster
- Origin:
  - [x] Docker (Image: ghdl/synth:formal)

tgingold, 2020-06-23:
But endpoint is not part of vhdl-2008 (nor of ieee1850-2005).

tmeissner, 2020-06-24:
Regarding the IEEE-standards, you're right. `endpoint` was included in the Accellera PSL 1.1 LRM, but in the later IEEE standards replaced by the `ended()` function, I think.  However, most tools I know support `endpoint` because of backward compatibility.

Another way could be the implementation of the `ended()` function. And let `endpoint` as it is implemented now in PSL comments. The implementation of the `ended()` function should similar to endpoint, shouldn't it?

tgingold, 2020-06-27:
I think supporting `ended()` is the way to go.  Supporting `endpoint` in vhdl-2008 means making `endpoint` a reserved word, which is not.

tmeissner, 2020-06-27:
Yes, implementing `ended()` would be the best standard-conforming way. Please ask, if I could help (test cases for example).

Some time ago I wanted to implement the three "easy" sounding PSL function `countones`, `onehot` and  `onehot0`, but I've stuck at these tables with assignments of ids to functions. Maybe I try again and ask you when found things that I don't really understand.

tmeissner, 2020-07-06:
Okay, I will close that bug because support of `ended()` is mentioned in ghdl/ghdl#662.

closed 2020-07-06 by tmeissner