ghdl/ghdl #1380: Simulation time is zero

Werni2A, 2020-06-23:
**Description**
Simulating the design below results in a simulation time of zero. This means GtkWave does not show any traces.

I'm kind of a novice in VHDL so I hope it's not just a design mistake.

**Expected behaviour**
I would expect to see the simulation of my circuit. It works in Vivado 2018.3.

**How to reproduce?**
Design:
```vhd :file: gcd_algo.vhd

library ieee;
use ieee.std_logic_1164.all;

entity gcd_algo is
    port(UIn  : in  std_logic_vector(31 downto 0);
         ZOut : out std_logic_vector(31 downto 0);
         CLK  : in  std_logic);
end gcd_algo;

architecture Behavioral of gcd_algo is

begin

    process(CLK)
    begin
        if rising_edge(CLK) then
            ZOut <= UIn;
        end if;
    end process;

end architecture;
```
Testbench:
```vhd :file: tb_gcd_algo.vhd

library ieee;
use ieee.std_logic_1164.all;

entity tb_gcd_algo is
end tb_gcd_algo;

architecture Behavioral of tb_gcd_algo is

    component gcd_algo is
        port(UIn  : in  std_logic_vector(31 downto 0);
             ZOut : out std_logic_vector(31 downto 0);
             CLK  : in  std_logic);
    end component;

    signal sU, sZ : std_logic_vector(31 downto 0) := (others => '0');
    signal sCLK   : std_logic := '0';

begin

    uut: gcd_algo port map(UIn  => sU,
                           ZOut => sZ,
                           CLK  => sCLK);

    clk_gen: process
        variable ctr : integer := 0;
    begin
        sCLK <= not sCLK;
        ctr := ctr + 1;

        if (ctr < 15) then
            wait for 0.5 ms;
        else
            assert false report "Simulation finished." severity note;
            wait;
        end if;
    end process;

    stimuli: process
    begin

        sU <= (others => '0');
        wait for 0.5 ms;

        sU <= (others => '1');
        wait for 2 ms;

        sU <= (others => '0');
        wait for 0.5 ms;

        sU <= (others => '1');
        wait for 2 ms;

        assert false report "Simulation finished." severity note;
        wait;

    end process;

end Behavioral;
```
Shell script that I used to run the simulation:
```sh :image: ghdl/ghdl:buster-mcode
echo "Analyse..."
ghdl -a gcd_algo.vhd
ghdl -a tb_gcd_algo.vhd
echo "Elaborate..."
ghdl -e tb_gcd_algo
echo "Run testbench..."
ghdl -r gcd_algo --vcd=gcd_algo.vcd

gtkwave gcd_algo.vcd
```

**Context**

- OS: Ubuntu 20.04 x64
- GTKWave Analyzer v3.3.103
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `commit SHA: 18a71a430a7cfc460e9b013b37465ba7a9e32b1e` I retrieved the SHA with `git log -1`, hope this is the right way to do.

Werni2A, 2020-06-23:
The according VCD output from GHDL. Is it on purpose that the GHDL version is always "v0"?

$date
  Tue Jun 23 14:30:35 2020
$end
$version
  GHDL v0
$end
$timescale
  1 fs
$end
$scope module standard $end
$upscope $end
$scope module std_logic_1164 $end
$upscope $end
$scope module gcd_algo $end
$var reg 32 ! uin[31:0] $end
$var reg 32 " zout[31:0] $end
$var reg 1 # clk $end
$upscope $end
$enddefinitions $end
#0
bUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU !
bUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU "
U#

tgingold, 2020-06-23:
You are simulating gcd_algo and not the testbench.  Nothing happens in the design when it is not driven by the testbench.

Werni2A, 2020-06-24:
You're right, sorry for the inconvenience.

closed 2020-06-24 by Werni2A