ghdl/ghdl #1382: Xilinx bug when instantiating primitives for simulation

GlenNicholls, 2020-06-25:
When instancing Xilinx primitives in `unisim.VCOMPONENTS`, GHDL does not allow both the following library declaration and adding a component declaration to the architecture:

```
library unisim;
use unisim.VCOMPONENTS.all;
```

This is unexpected since this module needs to be used for synthesis and simulation. In synth, Xilinx requires the component declaration, but GHDL throws the following warning:

```
D:\dev\fw_dev\caci_anets_fw\source\common\serial_io\firmware\src\ClockGenNco.vhd:187:5:warning: instance "ncoaccumulator_inst" of component "dsp48e2" is not bound [-Wbinding]
D:\dev\fw_dev\caci_anets_fw\source\common\serial_io\firmware\src\ClockGenNco.vhd:69:14:warning: (in default configuration of clockgennco(rtl))
```

Then during simulation with the below code, the ports are undefined because GHDL didn't bind the DSP48E2 component to the primitive defined in `vcomponents`. For reference, the exact same code in the first code block below compiled and simulated perfectly in ModelSim and activeHDL. Per my questions in Gitter, identical behavior has been seen for other primitives as well, so it appears GHDL doesn't deal with the first style well.

```vhd :file: ClockGenNco.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

library unisim;
use unisim.VCOMPONENTS.all;

entity ClockGenNco is
    generic (
        NcoWidth : integer range 1 to 48 := 32
    );
    port (
        clk   : in std_logic;
        reset : in std_logic;

        i_en           : in std_logic := '1';
        i_freqCtrlWord : in std_logic_vector(NcoWidth-1 downto 0);

        o_ncoClk : out std_logic := '0';
        o_phase  : out std_logic_vector(NcoWidth-1 downto 0) := (others => '0')
    );
end entity ClockGenNco;

architecture rtl of ClockGenNco is
    component DSP48E2 is
    generic (
        ACASCREG                  : integer := 1;
        ADREG                     : integer := 1;
        ALUMODEREG                : integer := 1;
        AMULTSEL                  : string := "A";
        AREG                      : integer := 1;
        AUTORESET_PATDET          : string := "NO_RESET";
        AUTORESET_PRIORITY        : string := "RESET";
        A_INPUT                   : string := "DIRECT";
        BCASCREG                  : integer := 1;
        BMULTSEL                  : string := "B";
        BREG                      : integer := 1;
        B_INPUT                   : string := "DIRECT";
        CARRYINREG                : integer := 1;
        CARRYINSELREG             : integer := 1;
        CREG                      : integer := 1;
        DREG                      : integer := 1;
        INMODEREG                 : integer := 1;
        IS_ALUMODE_INVERTED       : std_logic_vector(3 downto 0) := "0000";
        IS_CARRYIN_INVERTED       : bit := '0';
        IS_CLK_INVERTED           : bit := '0';
        IS_INMODE_INVERTED        : std_logic_vector(4 downto 0) := "00000";
        IS_OPMODE_INVERTED        : std_logic_vector(8 downto 0) := "000000000";
        IS_RSTALLCARRYIN_INVERTED : bit := '0';
        IS_RSTALUMODE_INVERTED    : bit := '0';
        IS_RSTA_INVERTED          : bit := '0';
        IS_RSTB_INVERTED          : bit := '0';
[... 333 lines cut ...]
        RSTCTRL        => '0',
        INMODE         => "00000", -- dynamic pipelining for regs in A/B/AD logic, not used
        CEINMODE       => '0',
        RSTINMODE      => '0',
        CEM            => '0',
        RSTM           => '0',
        OVERFLOW       => open,
        UNDERFLOW      => open,
        PATTERNBDETECT => open,
        PATTERNDETECT  => open,
        XOROUT         => open,
        MULTSIGNIN     => '0',
        MULTSIGNOUT    => open
    );

end architecture;
```

I'm not providing the analysis/elab commands since the Xilinx libs need to be compiled and I don't know the order since I used the compile scripts

- OS: Windows
- Origin:
  - [x] Built from sources: `1666a4c7`

Compiled using VHDL2008, compiled the vendor libraries with `./compile-xilinx-vivado.sh --all --vhdl2008`, and used components in Vivado v2019.1

GlenNicholls, 2020-06-26:
@eine I think that this is a bug, why was it marked as an enhancement?

tgingold, 2020-06-26:
Rules that define default binding are in LRM08 7.3.3, and in your case there is no default binding.

Why do you need to redeclare the `DSP48E2` component ?  Isn't it already declared in `unisim.vcomponents` ?

tgingold, 2020-06-26:
You can also use `--syn-binding` which search for any entity with the same name as the component.

tgingold, 2020-06-26:
But my general advice is: do not use components, use direct instantiation.
Components are almost useless.

GlenNicholls, 2020-06-26:
>Why do you need to redeclare the DSP48E2 component ? Isn't it already declared in unisim.vcomponents ?

That's just a style we have in a lot of our code for instantiating vendor primitives. I understand what you're saying so I'll close this out.

closed 2020-06-26 by GlenNicholls

eine, 2020-06-26:
> @eine I think that this is a bug, why was it marked as an enhancement?

Tristan doesn't make any difference because of the labels, he has his own priorities. Hence, I use:

- Bug: this was supposed to work, or it's a regression.
- Enhancement/FeaReq: this was never supported but it should.

I do this to avoid all not completely supported language features from being considered bugs in the implementation. Due to the rolling nature of GHDL, almost no feature is completely supported. There are some features, such as this, which are in the middle between "it was supposed to work" and "it was not, but it should". Hence, I marked it as an enhancement while I waited for Tristan to comment. Nevertheless, please do comment in other issues if you think they are unproperly labeled.