ghdl/ghdl #1383: Unable to update ghdl: regression in recent ghdl-gcc ?

csantosb, 2020-06-26:
**Description**

A managed to build and execute code one month ago. Today's version os ghdl crashes with same vhdl code.

**Expected behaviour**

I would expect to get the same behavior as in previous version.

**How to reproduce?**

Using a docker image built on 2020.05.24. All is ok. Docker image contains a base arch linux, plus [ghdl-gcc-git](https://aur.archlinux.org/packages/ghdl-gcc-git/) and a couple other packages. Dockerfile is [here](https://gitlab.com/csantosb/dockerfiles/-/blob/master/arch-vhdl/Dockerfile). Compilation report is [here](https://gitlab.com/ip-vhdl/mux_fifo/-/jobs/612706067). Executable elaborates with no issue (line 70).

I rebuild the docker image, containing an updated ghdl-gcc-git. Using same vhdl code, resulting report is [here](https://gitlab.com/ip-vhdl/mux_fifo/-/jobs/613480265).

So, all in all, the [file](https://gitlab.com/ip-vhdl/mux_fifo/-/blob/master/ip/mux_fifo_fifo/mux_fifo_fifo_sim_netlist.vhdl) and the command

    ghdl -a --std=08 --ieee=synopsys -P/opt/Xilinx/CompiledLibraries/Vivado --mb-comments -fpsl -C -fexplicit -frelaxed --no-vital-checks --warn-unused --warn-reserved --warn-parenthesis --warn-delayed-checks --warn-others --warn-pure --warn-static -O2   --work=work --workdir=work mux_fifo_fifo_sim_netlist.vhdl

fail to analyze with a segmentation error

    durant l'étape GIMPLE: ivopts
    /home/user/Projects/perso/ip-vhdl/mux_fifo/ip/mux_fifo_fifo/mux_fifo_fifo_sim_netlist.vhdl: Dans la fonction « work__mux_fifo_fifo_rd_bin_cntr__ARCH__structure__P0__PROC »:
    /home/user/Projects/perso/ip-vhdl/mux_fifo/ip/mux_fifo_fifo/mux_fifo_fifo_sim_netlist.vhdl:1215: erreur interne du compilateur: Erreur de segmentation
     1215 | entity mux_fifo_fifo_rd_bin_cntr is
          | 
    Veuillez soumettre un rapport d’anomalies complet,
    avec la sortie du préprocesseur si nécessaire.
    Voir <https://gcc.gnu.org/bugs/> pour les instructions.
    ghdl:error: exec error

For sure, the behavior reproduces in a native arch, same setup.

**Context**

- OS: arch linux
- Origin:
  - [X] Package manager: `https://aur.archlinux.org/packages/ghdl-gcc-git/`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
durant l'étape GIMPLE: ivopts
/home/user/Projects/perso/ip-vhdl/mux_fifo/ip/mux_fifo_fifo/mux_fifo_fifo_sim_netlist.vhdl: Dans la fonction « work__mux_fifo_fifo_rd_bin_cntr__ARCH__structure__P0__PROC »:
/home/user/Projects/perso/ip-vhdl/mux_fifo/ip/mux_fifo_fifo/mux_fifo_fifo_sim_netlist.vhdl:1215: erreur interne du compilateur: Erreur de segmentation
 1215 | entity mux_fifo_fifo_rd_bin_cntr is
      | 
Veuillez soumettre un rapport d’anomalies complet,
avec la sortie du préprocesseur si nécessaire.
Voir <https://gcc.gnu.org/bugs/> pour les instructions.
ghdl:error: exec error
```

**Additional context**

tgingold, 2020-06-26:
Does it work without `-O2` ?  I think I have a fix.

csantosb, 2020-06-26:
Wow, this is fast ! Indeed, yes, this fixes the problem.

tgingold, 2020-06-27:
The interface with gcc has been fixed, so I suppose this issue is also fixed.

Thank you for the issue.

closed 2020-06-27 by tgingold

csantosb, 2020-06-27:
I confirm, fixed issue. Thanks !