ghdl/ghdl #1384: Crash when assigning to a record

inkdot7, 2020-06-29:
**Description**
Crash when assigning to a record.  With gcc -O3.  Without, or -O1 and -O2 work.

**Expected behaviour**
No crash

**How to reproduce?**
```vhd :file: tb_issue.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

entity tb_issue is
end entity tb_issue;

architecture dataflow of tb_issue is

  type arec is record
    member : std_logic_vector(31 downto 0);
  end record arec;

  signal a : arec;

begin

  process
  begin
    a.member <= (0 => '1', others => '0');
  end process;

end architecture dataflow;
```

```sh :image: ghdl/ghdl:buster-mcode
mkdir -p sim
ghdl -i -O3 --workdir=sim tb_issue.vhd
ghdl -m -O3 --workdir=sim tb_issue
```

**Context**
Please, provide the following information:

- OS: Docker image, see below

```
Using Docker executor with image ghdl/vunit:gcc ...
Pulling docker image ghdl/vunit:gcc ...
Using docker image sha256:f41dd85b9ece7d3cd99eee89f0159fbb66972a55926251043afd2e1f9e59fb97 for ghdl/vunit:gcc ...
ghdl --version
GHDL 1.0-dev (v0.37.0-789-ge2d751d6@buster-gcc) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 GCC back-end code generator
```

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
62 mkdir -p sim/simulation
63 (cd sim ; ghdl -i -O3 --workdir=simulation \
64    --work=work \
65   ../tb_pexorseq.vhd)
66 (cd sim ; ghdl -m -O3 --workdir=simulation \
67    --work=work tb_pexorseq)
68 analyze ../tb_pexorseq.vhd
69 during GIMPLE pass: cunroll
70 /builds/expsubphys/pexorseq/sim/../tb_pexorseq.vhd: In function 'work__tb_pexorseq__ARCH__dataflow__P0__PROC':
71 /builds/expsubphys/pexorseq/sim/../tb_pexorseq.vhd:4: internal compiler error: in const_binop, at fold-const.c:1637
72  entity tb_pexorseq is
73  
74 0x615e77 const_binop(tree_code, tree_node*, tree_node*, tree_node*)
75 	../../gcc-srcs/gcc/fold-const.c:1636
76 0xe663c2 gimple_resimplify2(gimple**, code_helper*, tree_node*, tree_node**, tree_node* (*)(tree_node*))
77 	../../gcc-srcs/gcc/gimple-match-head.c:150
78 0xf31160 gimple_simplify(gimple*, code_helper*, tree_node**, gimple**, tree_node* (*)(tree_node*), tree_node* (*)(tree_node*))
79 	../../gcc-srcs/gcc/gimple-match-head.c:686
80 0x932d0f fold_stmt_1
81 	../../gcc-srcs/gcc/gimple-fold.c:4757
82 0xb95fcb replace_uses_by(tree_node*, tree_node*)
83 	../../gcc-srcs/gcc/tree-cfg.c:2025
84 0xc685c5 propagate_constants_for_unrolling
85 	../../gcc-srcs/gcc/tree-ssa-loop-ivcanon.c:1356
86 0xc685c5 tree_unroll_loops_completely
87 	../../gcc-srcs/gcc/tree-ssa-loop-ivcanon.c:1491
88 0xc68724 execute
89 	../../gcc-srcs/gcc/tree-ssa-loop-ivcanon.c:1610
90 0xc68724 execute
91 	../../gcc-srcs/gcc/tree-ssa-loop-ivcanon.c:1600
92 Please submit a full bug report,
93 with preprocessed source if appropriate.
94 Please include the complete backtrace with any bug report.
95 See <https://gcc.gnu.org/bugs/> for instructions.
96 ghdl: exec error
97 make: *** [Makefile:53: sim/tb_pexorseq] Error 3
98 ERROR: Job failed: exit code 1

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

** Other failure **

Before reducing the error to just one file, I had an error like this.  Perhaps they make sense together.
Otherwise I can try to recreate the error with the multiple files as well.

```
35   VHDL   sim/tb_pexorseq
36 analyze ../pexorseq_records.vhd
37 analyze ../pexorseq_sm.vhd
38 during RTL pass: expand
39 /builds/expsubphys/pexorseq/sim/../pexorseq_sm.vhd: In function 'work__pexorseq_sm__ARCH__rtl__P0__PROC':
40 /builds/expsubphys/pexorseq/sim/../pexorseq_sm.vhd:9: internal compiler error: in expand_expr_real_2, at expr.c:9767
41  entity pexorseq_sm is
42  
43 0x615784 expand_expr_real_2(separate_ops*, rtx_def*, machine_mode, expand_modifier)
44 	../../gcc-srcs/gcc/expr.c:9767
45 0x8c5bfe expand_expr_real_1(tree_node*, rtx_def*, machine_mode, expand_modifier, rtx_def**, bool)
46 	../../gcc-srcs/gcc/expr.c:9968
47 0x8cee7b store_expr_with_bounds(tree_node*, rtx_def*, int, bool, bool, tree_node*)
48 	../../gcc-srcs/gcc/expr.c:5651
49 0x8d0183 expand_assignment(tree_node*, tree_node*, bool)
50 	../../gcc-srcs/gcc/expr.c:5419
51 0x7d5440 expand_gimple_stmt_1
52 	../../gcc-srcs/gcc/cfgexpand.c:3706
53 0x7d5440 expand_gimple_stmt
54 	../../gcc-srcs/gcc/cfgexpand.c:3804
55 0x7d775f expand_gimple_basic_block
56 	../../gcc-srcs/gcc/cfgexpand.c:5833
57 0x7dcc96 execute
58 	../../gcc-srcs/gcc/cfgexpand.c:6439
59 Please submit a full bug report,
```

tgingold, 2020-06-30:
I am testing a fix for this issue.  This is due to vectorisation enabled at -O3.

inkdot7, 2020-06-30:
Curiosity: are the existing tests also tested at the various optimisation levels?

tgingold, 2020-06-30:
No, not all of them and not at all levels.

commit 8ab2ac73 (2020-07-01): ortho/gcc: build vector types in type_for_mode. Fix #1384
closed 2020-07-01 by tgingold