ghdl/ghdl #1385: Attribute expression for 'entity' must be locally static error in GHDL but not Vivado

talonmyburgh, 2020-06-30:
**Description**
Hi all, 
I have an entity in which I have declared the 'use_dsp' attribute (so that Vivado to infers it to a DSP48 chip). It is done so that a string generic, specifying "yes" or "no" instantiates the entity with "yes" or "no" (example 1).

In a separate file, I declare the component (example 2).

And finally, in the last file, I create the generic and port mappings (passing the "yes" or "no" string into the generic depending on whether I want the unit implemented in a DSP block or not (example 3). 

I use VUnit and was hoping to run tests on this setup (creating a tb to run - example 4), but when running it, I get a GHDL error:

> attribute expression for 'entity' must be locally static

**Expected behaviour**
Vivado is able to synthesise and implement this build the top module (like the one in example 3) perfectly - mapping to DSP48 when "yes" is passed, and using LUT's when "no" is passed.

I have not simulated yet with Vivado but I don't think it'd be a problem (perhaps I'm wrong?).
One difference is that in Vivado I am not building the files as though they are VHDL 2008 files, while VUnit does simulate with GHDL as though they are.

**How to reproduce?**

***Example 1***
```entity ex1 is
        generic (
        g_use_dsp          : STRING   := "YES"
        );
	attribute use_dsp : string;
	attribute use_dsp of ex1: entity is g_use_dsp;
end entity;

architecture a of ex1 is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end a;
```
***Example 2***
```Package expackage is
   component ex1
       generic
       (
       g_use_dsp          : STRING := "yes"
       );
  end component ex1;
end expackage;
```
***Example 3***
```entity ex3 is
        generic (
        g_use_dsp          : STRING   := "YES"
        );
end entity;

architecture a of ex3 is
begin
 mapping: ex1
  	generic map(
	g_use_dsp  => g_use_dsp
	)
end a;
```
***Example 4***
```Library vunit_lib;
context vunit_lib.vunit_context;
entity ex4_tb is 
generic (
    g_use_dsp : string:= "yes"
);
end ex4_tb;

architecture a of ex4_tb is
dut: entity ex3
   generic map
   (
   g_use_dsp => g_use_dsp
   );
```

Through a run.py script, I'll populate the string generic use_dsp with "yes" or "no" and when doing so I get the error mentioned earlier.

**Context**

- OS: PopOS 18.04 LTS (same as Ubuntu 18.04 LTS)
- GHDL version: 0.37-dev (v0.36-1612-g93922174-dirty) [Dunoon edition] (Compiled with GNAT Version: 7.5.0
 llvm code generator)
- Vivado version:  Vivado v2019.1 (64-bit)
- VUnit version: 4.4.0

**Full GHDL error as reported by VUnit:**
```
******************** GHDL Bug occurred ***************************
=== Command used: ===
/etc/ghdl-llvm/bin/ghdl -a --workdir=/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_multiplier/vunit_out/ghdl/libraries/casper_multiplier_lib --work=casper_multiplier_lib --std=08 -P/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_multiplier/vunit_out/ghdl/libraries/vunit_lib -P/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_multiplier/vunit_out/ghdl/libraries/casper_multiplier_lib -P/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_multiplier/vunit_out/ghdl/libraries/common_components_lib -P/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_multiplier/vunit_out/ghdl/libraries/common_pkg_lib /home/talon/Documents/CASPERWORK/casper_dspdevel/casper_multiplier/ip_cmult_rtl_4dsp.vhd

=== Command output: ===
/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_multiplier/ip_cmult_rtl_4dsp.vhd:90:18: attribute expression for 'entity' must be locally static
/home/talon/Documents/CASPERWORK/casper_dspdevel/casper_multiplier/ip_cmult_rtl_4dsp.vhd:93:21: entity 'ip_cmult_rtl_4dsp' was not analysed
/etc/ghdl-llvm/bin/ghdl: compilation error
******************************************************************
```

***PS***
The files provided are not the ones I am working on obviously. They are just examples - which I hope reproduce the error. The one's I work with are fairly large and come with 2 other libraries as requirements. I'd be happy to share them should these examples not be sufficient.

Thank you in advance.

tgingold, 2020-06-30:
That's a rule from the language (which is ignored by Vivado).

That's a candidate to be handled by -frelaxed.

talonmyburgh, 2020-07-01:
Okay so are you saying that using the flag -frelaxed will ignore this violation (meaning that I will need to alter the way in which VUnit calls GHDL)?
Or are you saying that this is something to be incorporated into -frelaxed in a future version?

Thank you.

tgingold, 2020-07-01:
I am testing the change to incorporate that info -frelaxed.

So you have to wait a little bit and prepare a change in VUnit.

closed 2020-07-01 by tgingold

eine, 2020-07-01:
@talonmyburgh, in case you are not sure about how to change you VUnit script, see https://github.com/VUnit/vunit/blob/master/examples/vhdl/array/run.py#L28-L29

talonmyburgh, 2020-07-02:
Cheers, thanks.

commit 66ed6fde (2020-07-01): vhdl: Relax check for locally static attribute specification in entities.

Fix #1385