ghdl/ghdl #1386: Support -gGENERIC=VALUE for generics of type std_logic_vector

roeekalinsky, 2020-07-01:
Setting generic values using the -gGENERIC=VALUE method when executing the binary supports generics of type integer, string, std_logic, and possibly others, but conspicuously it does not support type std_logic_vector.

Attempting it with a generic of type std_logic_vector fails, returning the following error messages:

> error: unhandled type for generic override of 'generic_name'
> error: error during elaboration

Given its ubiquity, it would be very useful if type std_logic_vector was supported as well.

Given how std_logic and strings are both already supported, std_logic_vector wouldn't be much of a stretch.  The command line argument format would naturally look something like this:

> -gNAME="01101001"

Some context for my particular needs: Currently migrating a large body of IP and work flow from using Modelsim as the primary simulator to using GHDL instead (many advantages!).  Lack of support for -gGEN=VAL for type std_logic_vector, however, is currently one of the last few obstacles.  An alternative in my situation could be to modify the IP to get rid of generics of type std_logic_vector and replace them with other types that are supported.  This however, entails other headaches, and would be detrimental to the cleanliness/flexibility/generality/quality of the IP.  Whereas full support for a type as ubiquitous as std_logic_vector, including its use with top level generics, is something one would expect and find broadly useful in GHDL, as it is generally supported with other VHDL simulators.

I'm currently using GHDL release v0.37 with LLVM backend, built from sources, on CentOS 7 and 8.

eine, 2020-07-01:
@roeekalinsky see #1117.

roeekalinsky, 2020-07-01:
Thanks, @eine.  Sorry for the duplicate.

eine, 2020-07-01:
It's absolutely ok ;). In fact, I think we can leave this open as it is about `std_logic_vector`, while #1117 covers other array types.

roeekalinsky, 2020-07-02:
I'd say #1117 could be construed a superset, as a request to support -gGENERIC=VALUE for array types in general.  And that naturally should include std_logic_vector as well as the numeric_std signed and unsigned types.  But anyhow, no problem.  Here's a MWE:

example.vhd:
```vhd :file: example.vhd
library ieee;
use ieee.std_logic_1164.all;

entity example is
  generic (
    PARAMETER : std_logic_vector(7 downto 0));
end example;

architecture behavioral of example is
begin
end behavioral;
```

example.sh:
```sh :file: example.sh
ghdl -a example.vhd
ghdl -e example
./example -gPARAMETER=00110101
```

roeekalinsky, 2020-07-02:
In fact, I would further generalize the request to include support for generics that are declared as unbound array types, their length determined by the length of the value string provided with -gGENERIC=VALUE.  This is allowed in VHDL, and supported by other simulators.

I would say the unbound case is of lesser priority than the bound case as shown in the MWE above, but both should be supported for completeness.

Here's a MWE for unbound:

example.vhd:
```vhd :file: example.vhd
library ieee;
use ieee.std_logic_1164.all;

entity example is
  generic (
    PARAMETER : std_logic_vector);
end example;

architecture behavioral of example is
begin
end behavioral;
```

example.sh:
```sh :file: example.sh
ghdl -a example.vhd
ghdl -e example
./example -gPARAMETER=00110101
```

eine, 2020-07-02:
Thanks. Much appreciated!

commit 169f89c6 (2021-10-04): grt-change_generics: handle subtype for elements.  Fix #1386
closed 2021-10-04 by tgingold