ghdl/ghdl #1388: A case in favor of preserving (i.e. not deprecating) the run -gGENERIC=VALUE capability

roeekalinsky, 2020-07-02:
Regarding the following comments in the GHDL documentation pertaining to the -gGENERIC=VALUE feature:

https://ghdl.readthedocs.io/en/latest/using/Simulation.html:
> This is currently a run option; but in the (not near) future it might be deprecated to become an elaboration option only. As a result, now you can generate a single binary and execute it multiple times with different arguments. That might not be possible in the future.
> 
> As explained in -e, performing a complete elaboration in terms of the LRM requires to get rid of the compile and link model. This is mostly because delaying certain elaboration steps to the runtime prevents elaboration-time optimisions.

While true, there are also use cases that do benefit from being able to set generic values when executing the binary rather than at the elaboration step, even if that means foregoing the benefit of elaboration-time optimizations.

One such use case is when running a very large number of very quick simulations of the same module in different configurations.  In this use case, having to run the elaboration step to generate a different binary executable for every configuration may end up being far more costly in total CPU time (and disk I/O, etc.) than any improvement in simulation run time that could be gained by elaboration-time optimizations.

This is not a contrived use case.  I'm currently exercising exactly such a use case for regression testing of low-level IP modules.  Running millions of independent simulations of a given module over a large configuration space, with run time of an individual simulation often on the order of a second or less.

This use case, by the way, also lends itself very well to running many independent simulations in parallel on a multi-core machine, which I do extensively.  And this is a huge benefit to using GHDL over commercial simulators who's licenses don't allow a "seat" to run multiple sessions concurrently.  A strong pitch for GHDL! ;-)  But I digress...

Just providing an example of the value of this runtime capability, as an argument for preserving it as an option into the future.

tgingold, 2020-07-02:
Quick question: which back-end do you use ? (if you don't know, it is in the output of `ghdl --version`)

roeekalinsky, 2020-07-02:
I'm using the LLVM back end currently, although I intend to also test out the GCC back end in the near future to see if there's a performance advantage.

Why do you ask?  I'd be happy to hear any insights or recommendations you might have.  Thanks!

### Detailed version info:

> [rkalinsky@localhost ~]$ ghdl --version
> GHDL 0.37 (tarball) [Dunoon edition]
>  Compiled with GNAT Version: Community 2020 (20200429-93)
>  llvm code generator
> Written by Tristan Gingold.
> 
> Copyright (C) 2003 - 2020 Tristan Gingold.
> GHDL is free software, covered by the GNU General Public License.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> [rkalinsky@localhost ~]$ llvm-config-64 --version
> 9.0.1

> [rkalinsky@localhost ~]$ cat /etc/centos-release
> CentOS Linux release 8.2.2004 (Core)

eine, 2020-07-02:
@roeekalinsky you might find some more details in #800 (I thought it was linked somewhere in the docs...):

> https://github.com/ghdl/ghdl/issues/800#issuecomment-486076526
> The compile & link model used by LLVM and gcc backends prevents from doing post elaboration optimizations. And this model is not used by commercial simulators. Improving the simulation speed means getting rid of that model. That's why it could be deprecated in the future.

> https://github.com/ghdl/ghdl/issues/800#issuecomment-486339952
> As a preliminary comment, getting rid of the normal compile & link flow is right now just a plan (based on optimization purpose).
> 
> I think this would mean getting rid of the gcc build, but not of llvm as llvm supports jit.

> https://github.com/ghdl/ghdl/issues/800#issuecomment-486936060
> Two features are missing from the LLVM backend: debugging and coverage.  To be added.

tgingold, 2020-07-03:
@roeekalinsky It would be interested if you could quickly evaluate the performances of the backends (llvm, gcc and mcode) for your task.

roeekalinsky, 2020-07-04:
It's not quick, and it may be a couple of weeks before I even get to it, but I'll post relative performance data between LLVM and GCC back ends when I have it.

As for mcode, I can't run these same benchmarks on it as mcode is incompatible with the flow I created.  But FYI about a year ago when I first started experimenting with GHDL (v0.36 at that time), I did do some comparative testing between mcode and LLVM with a few stand-alone sims.  I found mcode to be generally much slower than LLVM, so I looked no further at mcode.

Looping back to the original topic of this thread, FYI, I am interested in (and intend to benchmark GCC vs LLVM at) both ends of the spectrum of simulation size:

1) High number of very small simulations, as in the use case I described above.  And as I described, in this scenario, time incurred at start-up such as configuration-specific elaboration optimization could become a significant contributor to overall run time, potentially even the dominant contributor if the actual simulation run time is already only a fraction of a second.

2) Relatively small numbers of very large system-level simulations that individually can take minutes, hours, or even days.  This is a very different use case that I also exercise extensively.  And notably, in this scenario, time incurred at start-up such as configuration-specific elaboration optimization is unlikely to be significant.  Simulation run time performance will certainly dominate the overall run time, and therefore overall run time would more likely benefit from time spent on elaboration optimizations up front.

These two extremes, both of which commonly exist in typical design workflow, are why there is potentially a benefit to supporting both methods of specifying generics, at elaboration time and at run time.  And in fact, some commercial simulators (e.g. Modelsim SE / Questa) do support both methods.  Each individual generic can be either specified at elaboration time, thereby locking it down and allowing for elaboration optimizations on that, or left to be specified at run time, foregoing elaboration optimizations on it.  This gives the user the ultimate flexibility to make the optimal decision for any given use case.

eine, 2020-07-04:
> Each individual generic can be either specified at elaboration time, thereby locking it down and allowing for elaboration optimizations on that, or left to be specified at run time, foregoing [...]

Should it be technically feasible in GHDL, and should Tristan like the proposal, I just love it.

@roeekalinsky, I'd be interested on the plumbing you use to get benchmark figures, as it could be reused for benchmarking other designs.

roeekalinsky, 2020-07-05:
> @roeekalinsky, I'd be interested on the plumbing you use to get benchmark figures, as it could be reused for benchmarking other designs.

That I can't share, as the environment is proprietary.  But I'll be happy to share my findings.

roeekalinsky, 2020-07-16:
This thread ended up drifting away from the original topic...  LLVM vs GCC benchmark results are shared under a separated thread: #1394.