ghdl/ghdl #1389: Math operations with large real numbers yield large error

GlenNicholls, 2020-07-07:
During simulation when doing math that produces large real values, there appears to be a lot of error. In the below case, I'm taking `3600/1e-15` which should produce `3.6e18`, but instead I see an error of `4.096e3`. I do not see this problem in ModelSim

```vhd :file: mwe.vhd
library ieee;
context ieee.ieee_std_context;
use ieee.math_real.all;
use ieee.numeric_std_unsigned.all;

entity mwe is
end entity;

architecture a of mwe is
begin
  process
    constant hr_real  : real := 60.0**2.0;
    constant div_real : real := 1.0e-15;
    constant expect   : real := 3.6e18;
    variable got      : real;
    variable err      : real;
  begin
    got := hr_real/div_real;
    err := abs(got-expect);
    assert got = expect
      report
        "got=" & real'image(got) &LF&
        "expect=" & real'image(expect) &LF&
        "err=" & real'image(err)
      severity error;
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 mwe.vhd
ghdl --elab-run --std=08 mwe
```

yields

```
mwe.vhd:20:5:@0ms:(assertion error): got=3.599999999999996e18
expect=3.6e18
err=4.096e3
```

- OS: windows 10
- Origin:
  - [x] Built from sources: `1666a4c7`

eine, 2020-07-08:
@GlenNicholls, I believe the error is "within the expected bounds", considering you are using floating point types. The error is not "really" 4.096e3, but `3.6 ~= 3.599999999999996`; that is, in the least signifficant bit/digit of the mantisa (16 for decimal64: https://en.wikipedia.org/wiki/IEEE_754#Basic_and_interchange_formats). `3.6` cannot be accurately represented in binary. With 24 bits for the mantisa: 

- `3.60000014305114746094 <-> 11.1001100110011001100111`
- `3.59999990463256835938 <-> 11.1001100110011001100110`

Note that GHDL (VHDL) uses binary64 instead of binary32, so 53 bits are used for the mantisa; thus the difference between `3.599999999999996` and `3.59999991` above.

Anyway, the error seems to be large because you are looking at the absolute value. If you compute the relative error instead (`4.096e3 / 3.6e18 = 1.1377777777777777777777777777778e-15`) which is the expected machine epsilon for doubles.

See #1088.

GlenNicholls, 2020-07-08:
@eine I guess what I don't understand then is the fact that it properly prints 3.6e18 for the expected value, but the math does not yield this result? Is this just the limitation of FP math in VHDL as binary64? The other thing that does not make sense to me is why doesn't ModelSim have the same problem for identical code? Regarding the linked issue, this isn't a printing problem, rather an equality check problem.

eine, 2020-07-08:
> I guess what I don't understand then is the fact that it properly prints 3.6e18 for the expected value, but the math does not yield this result?

I cannot answer this. My guess is that, being a constant, the "source" is printed without processing it as a binary64. I'd try adding `0.0` or `1.0` to see if it makes any difference.

> Is this just the limitation of FP math in VHDL as binary64?

I think so. VHDL does not define it, it just uses it. I'd expect the same code in C or Matlab/Octave to produce similar results, unless they use decimal64: https://en.wikipedia.org/wiki/Double-precision_floating-point_format#Implementations. Moreover, I'd expect this not to be implemented by Tristan, but either in the standard Ada or C libs.

> The other thing that does not make sense to me is why doesn't ModelSim have the same problem for identical code? 

Maybe they don't modify the significand. They check that the divisor is `1.0`, so they just operate on the exponents and leave the dividend's significand as-is. Or maybe they use decimal64 (which would be strange...).

> Regarding the linked issue, this isn't a printing problem, rather an equality check problem.

I opened the issue because I had problems using an assertion, as you are doing. I solved it by using VUnit's `max_diff`: https://github.com/VUnit/vunit/blob/master/vunit/vhdl/check/src/check_api.vhd#L945. Then, I opened the issue here to understand which value should I put in `max_diff` (the theoretical bound).

GlenNicholls, 2020-07-08:
> I opened the issue because I had problems using an assertion, as you are doing. I solved it by using VUnit's max_diff: [...]

I ended up doing that as well. @tgingold feel free to close this out if you don't think any action needs to be taken.

closed 2020-08-08 by tgingold