ghdl/ghdl #1390: synth: TYPES.INTERNAL_ERROR : netlists-memories.adb:334 (two-dimensional array)

tmeissner, 2020-07-13:
**Description**
My AES design cannot be synthesized. I assume, that GHDL has a problem with the s-boxes implemented as two-dimensional arrays which are detected as memories.

Sadly I could not derive a MWE until now :unamused:

**Expected behaviour**
The design should synthesize fine.

**How to reproduce?**
Get these files:
https://github.com/tmeissner/cryptocores/blob/master/aes/rtl/vhdl/aes_enc.vhd
https://github.com/tmeissner/cryptocores/blob/master/aes/rtl/vhdl/aes_pkg.vhd

```sh :image: ghdl/synth:formal
$ ghdl --synth --std=08 aes_pkg.vhd aes_enc.vhd -e aes_enc
```
results in the crash.

**Context**

- OS: Debian Buster
- Origin:
  - [x] Docker (Image: ghdl/synth:formal)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
$ ghdl --synth --std=08 aes_pkg.vhd aes_enc.vhd -e aes_enc
aes_pkg.vhd:174:59:note: found ROM "n208", width: 128 bits, depth: 16
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev () [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/build/aes/rtl/vhdl/
Command line:
ghdl --synth --std=08 -gformal=true aes_pkg.vhd aes_enc.vhd -e aes_enc
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-memories.adb:334
Call stack traceback locations:
0x55f6e059a50d 0x55f6e059a91f 0x55f6e059c255 0x55f6e059c355 0x55f6e05a0ba0 0x55f6e0689e36 0x55f6e0689ec4 0x55f6e0683078 0x55f6e0705491 0x55f6e0615aab 0x55f6e070ed89 0x55f6e04bffc5 0x7f6fd578c099 0x55f6e04bedf8 0xfffffffffffffffe
******************************************************************

```

tgingold, 2020-07-24:
Now fixed.  Thank you for the issue.
Do you have a testbench ?

closed 2020-07-24 by tgingold

tmeissner, 2020-07-24:
Yes. But it works tests against openSSL using VHPIdirect. If that is useful for you, you can find it here: https://github.com/tmeissner/cryptocores/tree/master/aes/sim/vhdl

Thanks for fixing this issue 😃

commit 921398c1 (2020-07-24): synth: fix handling of multi-dim ROM. Fix #1390

Document index order for memidx.