ghdl/ghdl #1392: Bound Check error on File read of unconstrained array type.

sckoarn, 2020-07-14:
**Description**
Reading from a file of unconstrained array type into a container that is not big enough. As per LRM 2000 section 3.4.1 statement:
If the object associated with formal parameter VALUE is shorter than this length, then only that portion of the array value read by the operation that can be contained in the object is returned by the READ operation, and the rest of the value is lost.

**Expected behaviour**
I expect the read to succeed and the elements of the small array filled with the first items read until filled.

**How to reproduce?**
https://www.edaplayground.com/x/2Dqs

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2020-07-19:
Now fixed.  Thank you for the issue.

closed 2020-07-19 by tgingold

diogratia, 2020-07-20:
This isn't a -2008 Feature. 

See the applicable portion of IEEE Std 1076-1987 3.4.1 File operations:

>For a file type declaration in which the type mark denotes an unconstrained array type, the same operations are implicitly declared, except that the READ operation is declared as follows:
>  
> ```vhdl
>      procedure READ (file F: FT; VALUE: out TM; LENGTH: out Natural);
> ```
>  
>The READ operation for such a type performs the same function as the READ operation for other types, but in addition it returns a value in parameter LENGTH that specifies the actual length of the array value read by the operation. If the object associated with formal parameter VALUE is shorter than this length, then only that portion of the array value read by the operation that can be contained in the object is returned by the READ operation, and the rest of the value is lost. If the object associated with formal parameter VALUE is longer than this length, then the entire value is returned and remaining elements of the object are unaffected by the READ operation.  
>  
>An error will occur when a READ operation is performed on file F if ENDFILE(F) would return True at that point. 

The last quoted paragraph covers attempting to read something longer than stored in the file while the previous paragraph covers the case presented here. There's an implication you should test ENDFILE before reading, particularly for an unconstrained array type.

There is a work around for the particular example by adding an overload for READ, but it doesn't generalize:

```vhdl
-- Code your testbench here
library IEEE;
use IEEE.std_logic_1164.all;
--use std.textio.all;
--use work.proc_pkg.all;

entity file15 is

end entity file15;

architecture rtl of file15 is

  type char_array_t is array(natural range <>) of character;
  type char_arr_file is file of char_array_t;
  
begin
  process
    
    file cr : char_arr_file;
    
    variable mcrr : char_array_t(12 downto 0);
    variable small_mcrr : char_array_t(2 downto 0);
    variable c : character := 'a';
    variable len : natural;
    
    procedure read (
        file caf:               char_arr_file;
        variable value:     out char_array_t;
        variable length:    out natural;
        constant cheat:     in  natural
    ) is
        variable cat_len:   natural;
        variable catv:      char_array_t (1 to cheat);
    begin
        read (caf, catv, cat_len); -- implicitly defined read procedure
        report "value = " & string(catv);  -- CAN BE REMOVED
        value := catv (1 to value'length);
        length := cat_len;
    end procedure; 
  begin
    
    file_open(cr, "./char_arr_file.txt", WRITE_MODE);
    for i in 0 to 12 loop
      mcrr(i) := c;
      c := character'succ(c);
    end loop;
    write(cr, mcrr);
    file_close(cr);
    
    -- small container
    file_open(cr, "./char_arr_file.txt", READ_MODE);
    -- read(cr, small_mcrr, len); -- implicitly defined read procedure
    read(cr, small_mcrr, len, mcrr'length);
    for i in 0 to 2 loop
      report "" & small_mcrr(i);
    end loop;
    file_close(cr);
    
    
    wait;
  end process;
    
end rtl;
```

```none
ghdl -a file15.vhdl
ghdl -e file15
ghdl -r file15
file15.vhdl:37:9:@0ms:(report note): value = mlkjihgfedcba
file15.vhdl:56:7:@0ms:(report note): k
file15.vhdl:56:7:@0ms:(report note): l
file15.vhdl:56:7:@0ms:(report note): m
```
I'd imagine sckoarn was after the general fix, this is a contrived test case. The length of the unconstrained array values can vary from one datum stored in the file to another.

commit 8677e93d (2020-07-19): vhdl trans-chap7: fix result of read with length.  Fix #1392