ghdl/ghdl #1394: Simulation speed benchmark: LLVM vs GCC backend

roeekalinsky, 2020-07-16:
I ran some comparative benchmarks of simulation speed between the LLVM and GCC backends, and at different -O optimization level settings.  Thought I'd share some results.

**Version info:**
- GHDL v0.37, built from sources
- GNAT Community 2020 version
- GCC 9.3.1 (sources obtained from GNAT Community 2020 version)
- LLVM 9.0.1
- Host CPU and OS: AMD Threadripper 3970X running CentOS 8

**Methods:**

The flow and VHDL IP I used are proprietary, can't share much detail on those.  But in the most general terms, I created three different benchmarks using IP having different levels of complexity.  At one extreme was a set of ~130K small simulations of different configurations of about 100 low level building block modules.  At the opposite extreme was a single simulation of a large and complex DSP oriented core that could occupy a high-end FPGA.  And the characteristics of the third benchmark were somewhere in between, a few thousand configurations of a mid-size design.

**Results:**

I can't share the actual data, but the following is a summary of the highlights.

- Most notably, simulation with the LLVM backend was faster than the GCC backend in all cases, and by a significant margin in most cases (min 3%, typ 35%, max 40%).

- Surprisingly, in most cases LLVM even with no optimizations (i.e. -O0) produced faster simulation speeds than GCC at any of its optimization levels.  And LLVM with optimizations was much faster yet.

- With GCC backend, the fastest simulation speeds were attained at -O2 optimization level.  Interestingly, simulation speed is actually consistently slower at -O3 optimization level.

- With LLVM backend, the fastest simulation speeds were attained at -O1 and -O2 optimization levels, which yield nearly identical results.  In some cases -O2 is marginally faster than -O1, in other cases the reverse is true.  And interestingly, simulation speed is actually consistently slower at -O3 optimization level (this was found to be the case with both LLVM and GCC backends).

- Analysis and elaboration speed was also consistently faster with the LLVM backend than the GCC backend at equivalent -O optimization levels, the difference growing larger at higher -O levels.

**Conclusions and Limitations:**

- The above results indicate that the LLVM backend is uniformly superior to the GCC backend on this platform, and the fastest simulation performance is attained using LLVM -O1 or -O2.

- Note however that the relative differences between LLVM and GCC, and between different -O optimization levels, may vary in significance between different VHDL designs being simulated.  In my limited testing, this variability did show up, although the relative differences were actually quite consistent in trend.  They only varied in magnitude.

- Note also that these results are limited in scope to the software versions listed above, and to the AMD 3970X CPU.  It's entirely possible that the results would look different with different software versions and/or a different CPU, especially an Intel rather than AMD CPU.  It's been observed in other benchmarks (not specifically related to GHDL) that relative performance between LLVM and GCC can sometimes skew significantly between AMD vs Intel CPUs.

marzoul, 2020-07-16:
Thank you for this very interesting topic.
Could you precise if the measured times include the compilation time, in addition to the run itself ?
Separating the two is important, because often it is handy (and time-saving) to use one bulky precompiled executable and run it with different input data and/or parameters.

roeekalinsky, 2020-07-16:
@marzoul, you are correct that the two are distinct, and I do treat them as such.  In my flow I compile once and then execute many times with different generic parameter values set at run time via -gNAME=VALUE.  In my benchmarking effort I was primarily interested in execution time, since it dominates in my work flow.  But I did also observe and report compile times.  Compilation times and execution times are reported separately.

To clear up terminology, my references to "simulation" are equivalent to "execution", and my references to "analysis and elaboration" are equivalent to "compilation".

In my previous post, if it wasn't clear, the following statements apply to execution only:

> - Most notably, simulation with the LLVM backend was faster than the GCC backend in all cases, and by a significant margin in most cases (min 3%, typ 35%, max 40%).
>
> - Surprisingly, in most cases [...]

While the following statement applies to compilation only, i.e. analysis and elaboration:

> - Analysis and elaboration speed was also consistently faster with the LLVM backend than the GCC backend at equivalent -O optimization levels, the difference growing larger at higher -O levels.

I hope that clears it up.

GyrosGeier, 2026-09-20:
Stupid question: is anyone actually setting up the correct CPU type on GCC?

We've been seeing failures on several platforms, like assembler messages on RISC-V

```
/build/ghdl-2.0.0+dfsg/builddir/gcc/build/gcc/ghdl -a --GHDL1=/build/ghdl-2.0.0+dfsg/builddir/gcc/bu
ild/gcc/ghdl1 --std=87 --bootstrap --work=std ../../src/std/v87/textio.vhdl
textio.s: Assembler messages:
textio.s:3: Error: the architecture string of -march and elf architecture attributes cannot be empty
/build/ghdl-2.0.0+dfsg/builddir/gcc/build/gcc/ghdl: compilation error
```

because on RISC-V, the backend code assumes that it is called via the GCC frontend, which interprets the `specs` file and inserts the required `-m` options. I suspect something similar might be going on here, except that the x86 backend has defaults that work, but use a very conservative architecture specification.

That would also explain `-O3` generating code that is worse for a more recent CPU model, but better for the default.