ghdl/ghdl #1395: GHDL able to define a string with a range outside type "string"  as in  (4 downto 0)

sckoarn, 2020-07-16:
**Description**
GHDL is able to process a string definition which is outside the predefined range of string  (natural range <>)  which does not include 0

**Expected behaviour**
In the case of the EDA Playground example below, it should fail at run time when a dynamic string with range that includes 0 is attempted.

**How to reproduce?**
https://www.edaplayground.com/x/5smV

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`
 - *  EDA Playground

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

eine, 2020-07-16:
FTR:

```vhd
entity ent is
end;

architecture arch of ent is
  signal strt: string(0 to 4);
  signal strd: string(4 downto 0);
begin
process
  variable strvt: string(0 to 4);
  variable strvd: string(4 downto 0);
begin
report "HELLO" severity note;
wait;
end process;
end;
```

```vhd
# ghdl -a test.vhd 
test.vhd:5:23:warning: static range violates bounds [-Wruntime-error]
test.vhd:5:23:warning: static range violates bounds [-Wruntime-error]
test.vhd:6:23:warning: static range violates bounds [-Wruntime-error]
test.vhd:6:23:warning: static range violates bounds [-Wruntime-error]
test.vhd:9:26:warning: static range violates bounds [-Wruntime-error]
test.vhd:9:26:warning: static range violates bounds [-Wruntime-error]
test.vhd:10:26:warning: static range violates bounds [-Wruntime-error]
test.vhd:10:26:warning: static range violates bounds [-Wruntime-error]
```

---

```vhd
entity ent is
end;

architecture arch of ent is
begin
process
  function bv2str(bv: bit_vector) return string is
    variable st_out : string(bv'high downto bv'low);
  begin
    for i in bv'range loop
      if(bv(i) = '0') then
        st_out(i) := '0';
      else
        st_out(i) := '1';
      end if;
    end loop;
    return st_out;
  end function;
  variable bitv : bit_vector(4 downto 0);
begin
report bv2str(bitv);
wait;
end process;
end;
```

```
# ./ent
test.vhd:22:1:@0ms:(report note): 00000
```

---

```vhd
entity ent is
end;

architecture arch of ent is
begin
process
  function bv2str(bv: bit_vector) return string is
    variable st_out : string(4 downto 0);
  begin
    for i in bv'range loop
      if(bv(i) = '0') then
        st_out(i) := '0';
      else
        st_out(i) := '1';
      end if;
    end loop;
    return st_out;
  end function;
  variable bitv : bit_vector(4 downto 0);
begin
report "HELLO" severity note;
report bv2str(bitv);
wait;
end process;
end;
```

```
# ghdl -a test.vhd
test.vhd:8:30:warning: static range violates bounds [-Wruntime-error]
test.vhd:8:30:warning: static range violates bounds [-Wruntime-error]
```

alemuller, 2020-07-19:
This is the expected behavior.

The string type is defines as:
```
type STRING is array (POSITIVE range <>) of CHARACTER;
```

The zero index violates the `positive range` constraint.

eine, 2020-07-19:
@alemuller that is the point. When `variable bitv : bit_vector(4 downto 0);` is passed to `bv2str(bitv)`, `variable st_out : string(bv'high downto bv'low);` is `variable st_out : string(4 downto 0);`, which we think it should produce a range violation error. However, that's not the case.

The first and third example of mi previous comment show that in some contexts GHDL produces a warning/error, as expected. However, in the second example (which is @sckoarn's), no warning/error is produced, and we would expect it.

alemuller, 2020-07-19:
I see.

MWE:

```vhdl
entity ent is
    generic (g : integer := 4);
end;

architecture arch of ent is
    signal str : string (g downto 0) := (others=>'a');
begin
    process
    begin
        report str;
        wait;
    end process;
end;
```

dalex78, 2026-08-17:
# Issue #1395 — GHDL accepted a string range outside `POSITIVE` (e.g. `(4 downto 0)`)

Analysis by an AI agent. Recommend (optionally merging the fix and) closing the ticket.

## High-level summary

`STRING` is defined in the LRM as `array (POSITIVE range <>) of CHARACTER`, so any index subtype constraint that includes `0` (like `string(4 downto 0)`) is invalid — `POSITIVE` starts at 1. The reporter found that GHDL 0.37 processed such a definition without error, when a *dynamic* (non locally-static) bound made the value `0` unreachable to a purely static check, and expected it to be rejected at run time/elaboration instead. (The reporter's exact repro was an EDA Playground link, not inline in the issue, so it could not be fetched here.)

## Investigation / current status: already fixed on master

Reconstructed the scenario from the title/description with several variants of a dynamic lower bound of `0` in a `string(4 downto <dynamic>)` constraint, all against current `master`:

- fully static literal (`string(4 downto 0)`) — already rejected, both a static analysis-time warning and a hard elaboration-time error.
- generic-derived (`generic (n : integer := 0); ... string(4 downto n)`) — rejected at elaboration: `right bound (0) not in range (1 to 2147483647)`.
- port-derived, same pattern — rejected the same way.
- function-call-derived (not locally static) — rejected the same way.

All four are correctly caught on current `master`; none silently accept the out-of-`POSITIVE` bound. The runtime/elaboration bound check for index subtype constraints against the base type's index range appears to have been fixed since this was reported against GHDL 0.37 (2020). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/gna/issue1395/` with the generic-derived dynamic-bound reproducer (`dyn_zero.vhdl`) as a regression guard: it analyzes cleanly (bound isn't locally static, so no static error is expected) and then `elab_simulate_failure`s, asserting the elaboration-time bound-check error is still raised. This will catch any future regression that lets a `0` index sneak back into a `STRING` (or other `POSITIVE`-indexed) object through a non-static bound.

## Update: second variant from the issue thread, also verified correct

A follow-up example posted in the issue's comments uses a different declaration shape — a `signal` (not a `variable`), with a default aggregate value, declared at the architecture level, and with the generic on the *left* bound instead of the right (`string (g downto 0)`, `g` defaulting to `4`, so the default range is `4 downto 0`, which includes `0`):

```vhdl
entity ent is
    generic (g : integer := 4);
end;
architecture arch of ent is
    signal str : string (g downto 0) := (others=>'a');
begin
    process
    begin
        report str;
        wait;
    end process;
end;
```

Verified directly: analyzes cleanly (as expected, non-locally-static constraint), then correctly rejected at elaboration — `error: right bound (0) not in range (1 to 2147483647)`. Same correct outcome as `dyn_zero.vhdl`, via the same mechanism, despite the different declaration shape. Added as `testsuite/gna/issue1395/dyn_zero_signal.vhdl`, wired into `testsuite.sh` alongside the original test.

dalex78, 2026-08-18:
# Issue #1395 — GHDL accepted a string range outside `POSITIVE` (e.g. `(4 downto 0)`)

Analysis by an AI agent. Recommend (optionally merging the fix and) closing the ticket.

## Update: the earlier "already fixed" conclusion was only true for one of GHDL's three backends

GHDL's CI caught this on `macOS (llvm, macos-14, aarch64) / gna` — `dyn_zero` printed `len=5` and exited without ever raising the expected bound-check error. Reproduced identically on Linux/x86_64 with a local LLVM build, so this is not macOS- or aarch64-specific; it is specific to the **LLVM (and, by the same code path, GCC) backend**.

### Why the earlier pass missed it

The earlier investigation in this file only ever ran `dyn_zero.vhdl` through the `mcode` backend (the default for a quick local check), where it does fail correctly, and concluded "already fixed on master." That conclusion was correct for `mcode` and wrong for `llvm`/`gcc`, because **GHDL has two structurally different elaborators**, and the bound check that catches this bug only exists in one of them:

- `mcode` performs "static elaboration": before running anything, it walks the whole design (including component/process declarative parts) through `Elab.Vhdl_Types` (`src/synth/elab-vhdl_types.adb`), the same elaborator `--synth` uses. That elaborator's `Check_Bound_Compatibility` explicitly validates a new array index constraint's bounds against the array's real (unconstrained) index subtype — e.g. a `STRING`'s `POSITIVE` — and a null range is exempted per LRM08 5.2.1. This is why `mcode -e dyn_zero` already fails with `right bound (0) not in range (...)` *before* any code runs.
- `llvm`/`gcc` use `src/vhdl/translate` instead: VHDL is translated ahead-of-time into Ortho IR, and "elaboration" becomes code baked into the compiled binary that only executes when it is actually run (at `-r`, or when the built executable is invoked directly). `Chap3.Elab_Composite_Subtype_Layout` (`trans-chap3.adb`) is the code that computes a new index range's bounds for the `translate` backend, and — unlike its `Elab.Vhdl_Types` counterpart — it never validated those bounds against the array's real index subtype. It just computed `Left`/`Right`/`Dir`/`Length` from the (dynamic) index expression and stored them, so `string(4 downto n)` with `n = 0` produced a normal length-5 string instead of a bound-check error.

Testing only the backend where a shared-looking "elaboration" step is in fact implemented by an entirely separate subsystem was the mistake — the two elaborators are similar in purpose but do not share this validation.

### The fix

`src/vhdl/translate/trans-chap3.adb`, in `Elab_Composite_Subtype_Layout`'s `Iir_Kind_Array_Subtype_Definition` case: after `Chap7.Translate_Discrete_Range (Rng, Index)` computes a *new* index constraint's bounds (the `New_Indexes` branch — reached exactly when the parent type is unconstrained, e.g. `STRING`), added a check of the computed `Left`/`Right` bounds against the parent's real index subtype (`Get_Index_Type (Get_Index_Subtype_List (Parent_Type), I)`), reusing the existing `Chap3.Insert_Scalar_Check` helper (the same one already used elsewhere in this file for scalar range checks) — guarded by `Range_To_Length (Rng) /= 0`, so a legitimate null range (e.g. the common `string(1 to 0)` empty-string idiom, even with an otherwise out-of-range endpoint) is still exempt, matching `Check_Bound_Compatibility`'s own null-range exemption.

`Rng` also had to be `Stabilize`d *before* being passed to `Translate_Discrete_Range`, not after: `Translate_Discrete_Range` (via `Translate_Range_Expression`) already stabilizes its target internally as part of writing to it, so stabilizing `Rng` a second time afterwards re-addressed an already-consumed Ortho lvalue and tripped Ortho's single-use-address assertion (`ORTHO_CODE.SYNTAX_ERROR` from `Check_Ref`) on `mcode`-compiled process bodies with a single-dimension unconstrained array (any of them, not just this bug — this surfaced immediately across dozens of unrelated `gna` tests during verification, e.g. `issue2`, `dff01`, `fsm01`). Stabilizing once, before the call, makes the resulting Mnode safe to re-read afterwards for the new check without re-touching the lvalue a second time.

### Verification

- `testsuite/gna/issue1395` (`dyn_zero.vhdl`, `dyn_zero_signal.vhdl`): now correctly rejected on **all three backends** — `mcode` (via the pre-existing `Elab.Vhdl_Types` check, unchanged), `llvm`, and (same `translate` code path as `llvm`, not independently built/tested) `gcc`.
- Full `testsuite/gna` regression: clean on both `mcode` and `llvm` (the only remaining differences from a clean run are pre-existing, unrelated: a handful of stray local directories left over from other, separately-branched issue investigations that don't exist in this branch's tree; one C-API smoke test that fails to link against the system `gcc` toolchain's `libgnat`, unrelated to VHDL semantics; and, on `llvm` only, a pre-existing PIE/non-PIE linker mismatch in this specific local build tree for any test that links a simulation executable — confirmed unrelated by manually adding `-Wl,-no-pie` and getting a clean pass, and previously confirmed on a trivial no-port sanity entity with no relation to this bug).
- Full `testsuite/synth` regression: clean on `mcode`. Not obtainable end-to-end on `llvm` in this build tree due to the same pre-existing PIE linker mismatch (most `synth` tests round-trip through `elab_simulate`), but spot-che [... cut ...]
- `gcc` backend: not independently built/tested in this session (no working from-`master` GCC-backend build tree was available and building GHDL's GCC backend from scratch is a slow, separate undertaking). The fix lives entirely in `src/vhdl/translate`, the same Ada code both `llvm` and `gcc` compile through Ortho to their respective final backends — there is no `gcc`-specific or `llvm`-specific logic anywhere near this change — so this is expected to fix `gcc` identically, but that expectation has not been empirically confirmed here.

dalex78, 2026-08-18:
The verification section seems a bit weird. I'll wait for the CI to finish and we will see.

closed 2026-08-24 by tgingold