ghdl/ghdl #1398: Foreign attribute mapping to function pointers (Python callbacks)

umarcor, 2020-07-17:
`pycb` is an example from ghdl-cosim to execute arbitrary Python callbacks from VHDL. See [documentation](https://ghdl.github.io/ghdl-cosim/vhpidirect/examples/shared.html#pycb) and [sources](https://github.com/ghdl/ghdl-cosim/tree/master/vhpidirect/shared/pycb).

A function pointer is defined in C. Then, when the simulation is loaded (as a shared lib), but before executing `ghdl_main`, the  pointer is overriden from Python.

```c
void (*plot)(int*, int*, int);

int py_main(int argc, void** argv, void (*fptr)(int*, int*, int)) {
  if (fptr != NULL) {
    plot = fptr;
  }
  ghdl_main(argc, argv);
  return 0;
}
```

> NOTE: I'm investigating how to avoid `py_main`: https://stackoverflow.com/questions/62947076/replace-a-function-pointer-in-a-shared-library-with-ctypes

I tried to bind/map the function pointer (`*plot`) through VHPIDIRECT:

```vhdl
  procedure plot(x, y: arr_t; l: integer);
  attribute foreign of plot: procedure is "VHPIDIRECT *plot";
```

However, it fails with `OSError: ./caux.so: undefined symbol: *plot`.

The same applies to `  attribute foreign of plot: procedure is "VHPIDIRECT (*plot)";`

I tried the following too:

```vhdl
  procedure plot(x, y: arr_t; l: integer);
  attribute foreign of plot: procedure is "VHPIDIRECT plot";
```

This will load and start the simulation, but it will fail when using the function:

```
tb.vhd:17:5:@0ms:(report note): Hello shared/pycb/pyghdl!
../run.py:error: NULL access dereferenced
```

So, I need to use an intermediate function in C, just to dereference the pointer:

```c
void ghdl_plot (int* x, int* y, int l) {
  printf("ghdl_plot!\n");
  plot(x, y, l);
};
```

```vhdl
  procedure plot(x, y: arr_t; l: integer);
  attribute foreign of plot: procedure is "VHPIDIRECT ghdl_plot";
```

This is unfortunate, because it increases the verbosity of an otherwise nice and clean solution.

From VHDL 2008 LRM:

> 20.2.4.3 Standard direct binding
>
> standard_direct_binding ::=
> standard_direct_architecture_binding | standard_direct_subprogram_binding
>
> standard_direct_architecture_binding ::=
>   VHPIDIRECT object_library_specifier elaboration_specifier execution_function_name
>
> standard_direct_subprogram_binding ::=
>   VHPIDIRECT object_library_specifier execution_specifier
>
> object_library_specifier ::= object_library_path | null
>
> object_library_path ::=
> graphic_character { graphic_character }

and

> elaboration_specifier ::= elaboration_function_name | null
>
> elaboration_function_name ::= C_identifier
>
> execution_specifier ::= execution_function_name | null
>
> execution_function_name ::= C_identifier

So, my understanding is that the following cases are allowed:

```
VHPIDIRECT
VHPIDIRECT execution_function_name
VHPIDIRECT elaboration_function_name execution_function_name
VHPIDIRECT object_library_path
VHPIDIRECT object_library_path execution_function_name
VHPIDIRECT object_library_path elaboration_function_name execution_function_name
```

Where `C_identifier` might be valid for describing both `ghdl_plot` and `*plot`. If so, this might be a bug (incomplete feature) in GHDL.

Otherwise, as far as I am aware, the standard does not define the notion of accesses to functions/procedures. In fact, I'm not sure it would be required for any other use case. Moreover, in C, function pointers are dereferenced implicitly when called. Last, according to https://stackoverflow.com/questions/62947076/replace-a-function-pointer-in-a-shared-library-with-ctypes, Python/ctypes might have a similar limitation. Hence, I don't know whether it is technically feasible.

---

As a related feature, for "dynamic overriding" of functions, it might be interesting to support some format such as:

`attribute foreign of plot: procedure is "VHPIDIRECT _protoype_";`

The effect of such declaration would be for the tool (simulator) to create the symbol/prototype of the function pointer. As a result, it would be possible to generate shared libs from simulation models, without requiring "any" helper C function. It would allow them to be directy loaded dynamically from a non-compiled language, as long as an MWE such as https://stackoverflow.com/questions/62947076/replace-a-function-pointer-in-a-shared-library-with-ctypes works.

/cc @JimLewis

This "requirement" is listed in https://github.com/umarcor/ghdl-cosim/blob/vhdl202x-lcd-dpi/vhdl202x-lcs-dpi/main.md#requirements

tgingold, 2020-07-19:
According to the LRM, VHPIDIRECT accepts a function name, and not a variable name of a pointer to function.

In any case, you need to declare a variable name.  Just create a C macro that declare both the variable name and the wrapper to call it.

umarcor, 2020-07-19:
Note that all this is to write a proposal/LCS for VHDL 202X. NOT anything to be requested for GHDL to support in the following months/years. Anyway, Python seems to also have issues with this currently corner case: https://stackoverflow.com/questions/62947076/replace-a-function-pointer-in-a-shared-library-with-ctypes.

> According to the LRM, VHPIDIRECT accepts a function name, and not a variable name of a pointer to function.

I don't think that using function pointers was considered when the current standard was written; I agree on that. However, I do neither think that the current wording avoids it explicitly. That is, `execution_function_name ::= C_identifier`, and a "variable name of a pointer to function" is a C identifier, isn't it? If so, I think the only required enhancement in the standard would be to rename `execution_function_name` with `execution_function`, for clarification.

> In any case, you need to declare a variable name. Just create a C macro that declare both the variable name and the wrapper to call it.

My point is: can this be done by the tool (in this case GHDL) implicitly? There are two different syntax proposals:

- In http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/DpiProposal#Code_Examples, it is proposed to provide the full C prototype in the VHPIDIRECT string.
- Above, I propose to use `"VHPIDIRECT _protoype_"`. Since the tool does already know which function prototype is required to match a given VHDL procedure/function, I find redundant the requirement to specify it.

When the foreign language is C/C++, using a macro is trivial, and I agree with it being an acceptable solution. However, when the foreign language is other (say Python), ideally users would only need to use VHDL and the foreign language of choice, not C too. This is related to http://grouper.ieee.org/groups/1076/email/msg00978.html and http://grouper.ieee.org/groups/1076/email/msg01287.html. So, this issue is about automatically creating prototypes for foreign subprograms. The complementary feature would be to automatically expose the VHDL functions/procedures using a similar approach.