ghdl/ghdl #1400: Cannot build ghdl: Nonrepresentable section or output

distributed, 2020-07-23:
I cloned the ghdl repo, latest master commit 01c82f, and tried to build it using the instructions at https://ghdl.readthedocs.io/en/latest/getting/mcode.html.

The `make step` fails as follows:
```
x86_64-linux-gnu-gnatlink-8 ghdl_jit.ali -o ghdl_mcode -g memsegs_c.o chkstk.o jumps.o times.o grt-cstdio.o grt-cgnatrts.o grt-cvpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o -ldl -lm -Wl,--version-script=/home/memi/sw/ghdl/build/../src/grt/grt.ver -Wl,--export-dynamic
/usr/bin/ld: jumps.o: relocation R_X86_64_32S against `.text' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: grt-cstdio.o: relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: grt-cvpi.o: relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: fstapi.o: relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
x86_64-linux-gnu-gnatlink-8: error when calling /usr/bin/x86_64-linux-gnu-gcc-8
gnatmake: *** link failed.
Makefile:184: recipe for target 'ghdl_mcode' failed
make: *** [ghdl_mcode] Error 4
```

I am running the build on Ubuntu 18.08. I am happy to provide whatever version information is helpful.

eine, 2020-07-23:
Hi @distributed! In [this](https://github.com/ghdl/docker/runs/879974386?check_suite_focus=true) CI job latest GHDL with mcode backend is built in an Ubuntu 18.08 Docker container. So, we know it must work. Let's find the problem!

Can you please write the exact commands that you are using. For example:

```
git clone https://github.com/ghdl/ghdl
cd ghdl
mkdir build
cd build
../configure
make
```

Surely, you had to install some packages before. Please add `apt install gnat` or any other that you installed (including the version).

For reference:

- https://github.com/ghdl/docker/blob/master/dockerfiles/build_debian#L16-L19
- https://github.com/ghdl/docker/blob/master/run.sh#L135

Although `gnat-7` is used there, I tried it locally with `gnat 8.4.0` and `gcc 7.5.0`.

distributed, 2020-07-23:
Hi @eine, thanks for the quick response.

The sequence of commands is

```
git clone https://github.com/ghdl/ghdl.git
cd ghdl
mkdir build
cd build
../configure --prefix=~/usr
make
```

The gnat packages were already installed on my system for a long time, since I dabbled with Ada once. That unfortunatelz means that I tell you what I installed back then. I can however give you the output of `apt list --installed | grep gnat`:
```
gnat-8/bionic-updates,bionic-security,now 8.4.0-1ubuntu1~18.04 amd64 [installed]
libgnat-8/bionic-updates,bionic-security,now 8.4.0-1ubuntu1~18.04 amd64 [installed,automatic]
libgnatvsn8/bionic-updates,bionic-security,now 8.4.0-1ubuntu1~18.04 amd64 [installed,automatic]
libgnatvsn8-dev/bionic-updates,bionic-security,now 8.4.0-1ubuntu1~18.04 amd64 [installed]
```

On my system, x86_64-linux-gnu-gcc-8 (8.4.0) is picked as the compiler. Reading your message, I tried two things:
1. `CC=gcc-7 ../configure --prefix=~/usr`
2. `../configure --prefix=~/usr --default-pic`

Both result in successful compilation! :)

eine, 2020-07-23:
@distributed nice :tada:

Please, leave this issue open for @tgingold to decide if the `default-pic` detection logic needs to be fixed for GCC-8 on Debian/Ubuntu.

For context, in ghdl/docker and ghdl/cosim GHDL is always built with `--default-pic`, regardless of the default configuration of GCC on the host. This is because fPIC is a requirement to build shared libs.

distributed, 2020-07-23:
Thanks for your help :) That was quick!

Tangentially reading discussions on the internet, it seems like some defaults regarding PIC were switched around gcc 8. I will leave the issue open.

tgingold, 2020-07-31:
I have the feeling that it is due to the use of different configuration for gcc and gnat.
Not sure how it could be investigate as it doesn't reproduce in CI.

closed 2020-07-31 by tgingold