ghdl/ghdl #1402: /usr/bin/ghdl-mcode:error: bound check failure at ../../src/openieee/v93/numeric_std-body.vhdl:890

tonyre4, 2020-07-23:
Hi guys.
**Description**
I have this component (rgbtohsv.vhdl):

```--rgbtohsv.vhdl
library ieee;
--use ieee.std_logic_unsigned.all;
use ieee.std_logic_1164.all;
--use IEEE.std_logic_arith.all;
use ieee.numeric_std.all;

entity RGBtoHSV is
port(	clk,rst,de: in   std_logic;
        R,G,B:  in   unsigned(7 downto 0);
	    H:      out  unsigned(7 downto 0);
	    S,V:    out  unsigned(7 downto 0));
	--clk,rst: in std_logic );
end RGBtoHSV;

architecture zeros of RGBtoHSV is
begin
    H <= (others => '0');
    S <= (others => '0');
    V <= (others => '0');
end architecture;

architecture main of RGBtoHSV is
begin

process(clk,R,G,B)
variable max,min,diff,Ss:  unsigned(7 downto 0) := (others=>'0');
variable difft255,Sbuff : unsigned(diff'length+8-1 downto 0):= (others=>'0');
variable Hb,Hbb : unsigned(7 downto 0):= (others=>'0');
variable GmB,BmG,BmR,RmB,RmG,GmR : unsigned(R'length-1 downto 0):= (others=>'0');
variable GmBt60,BmGt60,BmRt60,RmBt60,RmGt60,GmRt60 : unsigned(R'length+6-1 downto 0):= (others=>'0');
variable S0div,S1div,S2div : unsigned(RmG'length-1+6 downto 0):= (others=>'0');
variable S0sum,S1sum,S2sum : unsigned(RmG'length-1+6 downto 0):= (others=>'0');
variable S3div,S4div,S5div : unsigned(RmG'length-1+6 downto 0):= (others=>'0');
variable S3sum,S4sum,S5sum : unsigned(RmG'length-1+6 downto 0):= (others=>'0');

begin
    
    --max
        if R>=B and R>=G then
            max := R;
        elsif B>=R and B>=G then
            max := B;
        elsif G>=R and G>=B then
            max := G;
        end if;
        
        --min
        if R<=B and R<=G then
            min := R;
        elsif B<=R and B<=G then
            min := B;
        elsif G<=R and G<=B then
            min := G;
        end if;
        
        diff := max - min;
        difft255 := diff*"11111111";
        
        --Determinar S
        if max=0 then
            Ss := (others => '0');
        else
            Sbuff := difft255/max;
            Ss := Sbuff(7 downto 0);
        end if;
        
        --restas
        GmB := G-B;
        BmG := B-G;
        BmR := B-R;
        RmB := R-B;
        RmG := R-G;
        GmR := G-R;
        
        --Multiplicaciones por 60
        GmBt60 := GmB*"111100";
        BmGt60 := BmG*"111100";
        BmRt60 := BmR*"111100";
        RmBt60 := RmB*"111100";
        RmGt60 := RmG*"111100";
        GmRt60 := GmR*"111100";
        
        if diff=0 then
            Hbb := (others => '0');
        else
            s0div := GmBt60/diff;
            s1div := GmBt60/diff;
            s2div := GmBt60/diff;
            s3div := GmBt60/diff;
            s4div := GmBt60/diff;
            s5div := GmBt60/diff;
            
            --Solving addings or substractions
            s0sum := to_unsigned(  0,s0div'length) + s0div;
            s1sum := to_unsigned(360,s1div'length) - s1div;
            s2sum := to_unsigned(120,s2div'length) + s2div;
            s3sum := to_unsigned(120,s3div'length) - s3div;
            s4sum := to_unsigned(240,s4div'length) + s4div;
            s5sum := to_unsigned(240,s5div'length) - s5div;
            
            if R=max then
                if G>B then
                    Hb := s0sum(8 downto 1);
                else --B>=G
                    Hb := s1sum(8 downto 1);
                end if;
            elsif G=max then
                if B>R then
                    Hb := s2sum(8 downto 1);
                else -- B<=R
                    Hb := s3sum(8 downto 1);
                end if;
            elsif B=max then
                if R>G then
                    Hb := s4sum(8 downto 1);
                else
                    Hb := s5sum(8 downto 1);
                end if;
            end if;
            
            if Hb="10110100" then
                Hbb := (others => '0');
            else
                Hbb := Hb;
            end if;  
        end if;
    
    if clk'event and clk='1' and de='1' then
        H <= Hbb;
        S <= Ss;
        V <= max;
    end if;
end process;

end architecture;
```
As you can see, "zeros" architecture is just a dummy architecture that only outputs zeros on component's outputs. The "main" architecture is a RGB to HSV space color. This is a component of a larger project so I have a lot of more components connected. 

The problem raises when I set the "main" architecture. I get this:
```
../../src/openieee/v93/numeric_std-body.vhdl:2397:9:@0ms:(assertion warning): NUMERIC_STD.">=": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:2397:9:@0ms:(assertion warning): NUMERIC_STD.">=": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:2397:9:@0ms:(assertion warning): NUMERIC_STD.">=": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:2685:9:@0ms:(assertion warning): NUMERIC_STD."<=": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:2685:9:@0ms:(assertion warning): NUMERIC_STD."<=": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:2685:9:@0ms:(assertion warning): NUMERIC_STD."<=": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:644:9:@0ms:(assertion warning): NUMERIC_STD."-": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:644:9:@0ms:(assertion warning): NUMERIC_STD."-": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:644:9:@0ms:(assertion warning): NUMERIC_STD."-": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:644:9:@0ms:(assertion warning): NUMERIC_STD."-": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:644:9:@0ms:(assertion warning): NUMERIC_STD."-": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:644:9:@0ms:(assertion warning): NUMERIC_STD."-": non logical value detected
/usr/bin/ghdl-mcode:error: bound check failure at ../../src/openieee/v93/numeric_std-body.vhdl:890
in process .tb(text_io).top@core_t(rgbhsvgray).hsvmod@rgbtohsv(main).P0
  from: ieee.numeric_std."*" at numeric_std-body.vhdl:890
/usr/bin/ghdl-mcode:error: simulation failed
```
When I set the "zeros" architecture there's no problem, so I know the main problem is the way I coded my component. I made this component originally on vivado and as you can imagine it compiles and work.

**How to reproduce?**
In my case (with the whole project) I run this on terminal:
```
ghdl --clean
ghdl -a --ieee=synopsys vhdls/divider.vhdl
ghdl -a --ieee=synopsys vhdls/rgbtohsv.vhdl
ghdl -a --ieee=synopsys vhdls/controlsat.vhd
ghdl -a --ieee=synopsys vhdls/rgblum.vhd
ghdl -a --ieee=synopsys vhdls/treeChain.vhdl
ghdl -a --ieee=synopsys vhdls/sgm.vhdl
ghdl -a --ieee=synopsys vhdls/morph.vhd
ghdl -a --ieee=synopsys vhdls/videoMux.vhd
ghdl -a --ieee=synopsys vhdls/rowcolcntr.vhd
ghdl -a --ieee=synopsys vhdls/drawer.vhd
ghdl -a --ieee=synopsys vhdls/pipe.vhd
ghdl -a --ieee=synopsys vhdls/clclass.vhdl
ghdl -a --ieee=synopsys vhdls/fivebinhisto.vhd
ghdl -a --ieee=synopsys vhdls/treeclass.vhd
ghdl -a --ieee=synopsys vhdls/core.vhd
ghdl -a --ieee=synopsys vhdls/tb_imagefeeder.vhdl
ghdl -e --ieee=synopsys tb
ghdl -r --ieee=synopsys tb --vcd=wave.vcd  --stop-time=342060ns
```
I made a testbench for the component and you can see [here](https://imgur.com/a/jer6a8Y) (or the attachment) it works. Here is the tb:
**tb.vhdl**
```
library IEEE;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
entity tb is
end entity;

architecture text_io of tb is
    signal clkv: std_logic := '0';
	signal rst: std_logic := '1';
	signal r_in,g_in,b_in: unsigned(7 downto 0) := (others =>'0');
    type RAM is array (0 to 2) of integer range 0 to 255;

    constant R : RAM := (255,0,128);
    constant G : RAM := (255,200,120);
    constant B : RAM := (25,10,12);
    signal e : integer := 0;

begin
    clkv <=  '1' after 2.5 ns when clkv = '0' else
        	 '0' after 2.5 ns when clkv = '1';
    rst <=  '0' after 10    ns when rst = '1' else
        	'1' after 10000 ms when rst = '0';
  
    process(clkv)
    begin
        if clkv'event and clkv='1' then
            if e = 2 then 
                e <= 0;
            else
                e <= e+1;
            end if;
            r_in <= to_unsigned(R(e),r_in'length);
            g_in <= to_unsigned(G(e),r_in'length);
            b_in <= to_unsigned(B(e),r_in'length);
        end if;
    end process;

	top: entity work.rgbtohsv (main) 
	     port map(
		clk => clkv,
		rst => rst,
		de => '1',
		R => r_in,
		G => g_in,
		B => b_in,
        H => open,
        S => open,
        V => open 
	    );	
end architecture;
```
To run:
```
ghdl --clean
ghdl -a --ieee=synopsys rgbtohsv.vhdl
ghdl -a --ieee=synopsys tb.vhdl
ghdl -e --ieee=synopsys tb
ghdl -r --ieee=synopsys tb --vcd=wave.vcd  --stop-time=100ns
```
I have no idea what I'm doing wrong. Can anyone help? 
Thanks in advance!

**Context**

- OS: Ubuntu 20.04 
- GHDL: GHDL 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]
              Compiled with GNAT Version: 9.3.0

[image]

tonyre4, 2020-07-24:
Finally I solved it...

The solution was VERY STRANGE, and maybe is something wrong with the vhdl code parser or someting related because the solution was pretty random...

In the file rgbtohsv.vhdl I changed this:
```
        GmBt60 := GmB*"111100";
        BmGt60 := BmG*"111100";
        BmRt60 := BmR*"111100";
        RmBt60 := RmB*"111100";
        RmGt60 := RmG*"111100";
        GmRt60 := GmR*"111100";
```
To this:
```
        GmBt60 := sixty * GmB;
        BmGt60 := sixty * BmG;
        BmRt60 := sixty * BmR;
        RmBt60 := sixty * RmB;
        RmGt60 := sixty * RmG;
        GmRt60 := sixty * GmR;
```
where:
`constant sixty : unsigned(5 downto 0) := to_unsigned(60,6);`

I think something was missunderstood by the parser because it was complaining about the "*" character. The compiler wanted to load new libraries instead doing a simple multiplication...

Also...
This is good: 
`GmBt60 := sixty * GmB; `
If I put this is wrong: 
`GmBt60 := GmB * sixty; `

**VERY STRANGE AND ANNOYING ERROR that it shouldn't be.**

tgingold, 2020-07-24:
I am not yet able to reproduce the issue, but it looks like a problem in then openieee library.

tgingold, 2020-07-25:
I was not able to reproduce the initial bound check failure with your testbench. 

I also do not understand what you mean with: complaining about the "*" character.  What is the error message ?  Do you have a full reproducer for this sub-issue ?

Likewise with "load new libraries".  I am not even sure of the meaning.

tonyre4, 2020-07-25:
I'm sorry for my English. Let me make a full tb to reproduce the error.

Also I got another error with other component making a simple division.

Is there any difference for GHDL compiler of compiling an component instantiation before "begin" keyword in architecture vs this?

`u2: entity work.Parity generic map(N => 8) port map(A => Data, Odd => ParityByte);`

I'm using last one to instantiate my components and I haven't tried the old way yet.

tgingold, 2020-07-25:
You can instantiate components or you can directly instantiate entities.  Both are supported by GHDL.

In both cases, the instantiation must be done after the `begin`.
And the component must be declared either in a package or before the `begin`.

My advice is to directly instantiate entities.  This is simpler as it avoids to declare components.

Do not hesitate to open issues if an error message is difficult to understand.  But it is important to join a reproducer so that comments are accurate and investigation is possible.

tgingold, 2020-08-04:
I close this issue as it cannot be reproduced.
If you reopen it, please join a reproducer.

closed 2020-08-04 by tgingold