ghdl/ghdl #1403: Makefile generated with mcode and llvm backend have differents GHDLFLAGS

alemuller, 2020-07-24:
**Description**

Makefile generated with mcode and llvm backend have differents `GHDLFLAGS`:

```
$ ghdl-mcode --gen-makefile -frelaxed ent | grep GHDLFLAGS | head -1
GHDLFLAGS= -frelaxed
$ ghdl-llvm --gen-makefile -frelaxed ent | grep GHDLFLAGS | head -1
GHDLFLAGS= 
```

I also noticed the mcode backend throws an error for unknown options and llvm one don't.

**Context**

```
$ ghdl-mcode --version
GHDL 1.0-dev (v0.37.0-849-ga1ea7f48) [Dunoon edition]
 Compiled with GNAT Version: 10.1.0
 mcode code generator
Written by Tristan Gingold.
```

```
$ ghdl-llvm --version
GHDL 1.0-dev (v0.37.0-849-ga1ea7f48) [Dunoon edition]
 Compiled with GNAT Version: 10.1.0
 llvm code generator
Written by Tristan Gingold.
```

**How to reproduce?**
Create a dummy file:

```vhdl file:ent.vhdl
entity ent is
end;

architecture arch of ent is
begin
end;
```

Analise the file:
```
$ ghdl-llvm -a ent.vhdl
```

Generate the makefile with both backends:
```
$ ghdl-llvm --gen-makefile -P$GHDL_VIVADO_LIB_DIR -frelaxed ent > llvm.mk
$ ghdl-mcode --gen-makefile -P$GHDL_VIVADO_LIB_DIR -frelaxed ent > mcode.mk
```

First lines from both files:

`llvm.mk`
```makefile file:llvm.mk
# Makefile automatically generated by ghdl
# Version: GHDL 1.0-dev (v0.37.0-849-ga1ea7f48) [Dunoon edition] - llvm code generator
# Command used to generate this makefile:
# ghdl-llvm --gen-makefile -P/home/alemuller/.local/lib/xilinx-vivado -frelaxed ent

GHDL=ghdl-llvm
GHDLFLAGS= -P/home/alemuller/.local/lib/xilinx-vivado

# Default target
all: ent
```

`mcode.mk`
```makefile file:mcode.mk

# Makefile automatically generated by ghdl
# Version: GHDL 1.0-dev (v0.37.0-849-ga1ea7f48) [Dunoon edition] - mcode code generator
# Command used to generate this makefile:
# ghdl-mcode --gen-makefile -P/home/alemuller/.local/lib/xilinx-vivado -frelaxed ent

GHDL=ghdl-mcode
GHDLFLAGS= -P/home/alemuller/.local/lib/xilinx-vivado -frelaxed
GHDLRUNFLAGS=

# Default target : elaborate
all : elab
```

eine, 2020-07-24:
> I also noticed the `--gen-makefile` with mcode backend throws an error for unknown options but with llvm backend it dosen't.

I may have overlooked your message, but I can't see which error is this. I mean, is it a crash or does it work despite of the error?

alemuller, 2020-07-24:
The issue is `ghdl-llvm` not including ` -frelaxed` options on the makefile.

---

> > I also noticed the --gen-makefile with mcode backend throws an error for unknown options but with llvm backend it dosen't.
>
> I may have overlooked your message, but I can't see which error is this. I mean, is it a crash or does it work despite of the error?

The quoted part is not an issue, just an observations:

`ghdl-llvm` accepts wrong syntax, `ghdl-mcode` doesn't.

```
$ ghdl-mcode --gen-makefile -foo ent
ghdl-mcode: error: unknown option '-foo'
$ ghdl-llvm --gen-makefile -foo ent
# Makefile automatically generated by ghdl
etc
```

alemuller, 2020-07-24:
Ops. Miss clicked.

commit 7f0f6c77 (2020-07-30): ghdldrv: factorize part of --gen-makefile

Fix #1403
closed 2020-07-30 by tgingold