ghdl/ghdl #1404: internal error: grt.waves.find_type

c-thaler, 2020-07-24:
**Description**
Running a simulation with my design and writing GHW waveforms leads to an internal error:

`internal error: grt.waves.find_type`

From the context, it seems that the type "pass_through_t" is missing. This is a record defined in "th_dnx_tfu_mipmap_sampler".

**How to reproduce?**
I was not able to produce a minimal example yet. The design is very large and has lots of (auto-generated) types.
Maybe you have some ideas how to get closer to the problem?

**Context**

- OS: Linux
- Origin:
  - [ ] Built from sources: `285ccdbd04eec0d2ddbae0296df5cae089f9739c` and also tried tag v0.37.0

**Additional context**
Backtrace:
#0  <__ghdl_fatal> () at /home/xxx/tools/ghdl_llvm/src/grt/grt-errors.adb:71
#1  0x0000555555f80c55 in grt.errors.internal_error (msg="grt.waves.find_type") at /home/xxx/tools/ghdl_llvm/src/grt/grt-errors.adb:233
#2  0x0000555555fca483 in grt.waves.find_type (rti=0x5555565070e0 <davenx__th_dnx_tfu_mipmap_sampler__ARCH__rtl__pass_through_t__RTI>, ctxt=
      (base => (system.address) 0x555586d6fad0, block => 0x5555565070a0 <davenx__th_dnx_tfu_mipmap_sampler__ARCH__rtl__RTI>), n_ctxt=
      (base => (system.address) 0x7fffffffcf70, block => 0x0), id=-12592) at /home/xxx/tools/ghdl_llvm/src/grt/grt-waves.adb:541
#3  0x0000555555fca9c2 in grt.waves.create_type (rti=0x5555565070e0 <davenx__th_dnx_tfu_mipmap_sampler__ARCH__rtl__pass_through_t__RTI>, ctxt=
      (base => (system.address) 0x555586d6fad0, block => 0x5555565070a0 <davenx__th_dnx_tfu_mipmap_sampler__ARCH__rtl__RTI>))
    at /home/xxx/tools/ghdl_llvm/src/grt/grt-waves.adb:607
#4  0x0000555555fcb274 in grt.waves.create_object_type (obj=
        (kind => vhpisigdeclk, ctxt => (base => (system.address) 0x555586d6fad0, block => 0x5555565070a0 <davenx__th_dnx_tfu_mipmap_sampler__ARCH__rtl__RTI>), obj => 0x555556506d40 <davenx__th_dnx_tfu_mipmap_sampler__ARCH__rtl__input_pass_through__RTI>))
    at /home/xxx/tools/ghdl_llvm/src/grt/grt-waves.adb:742
#5  0x0000555555fcbada in grt.waves.wave_put_hierarchy_1 (inst=
        (kind => vhpicompinststmtk, ctxt => (base => (system.address) 0x555586d12f60, block => 0x55555651b730 <davenx__th_dnx_tfu_addr_impl__ARCH__rtl__RTI>), inst => 0x55555651b920 <davenx__th_dnx_tfu_addr_impl__ARCH__rtl__mipmap_sampler__RTI>), step=step_name, match_list=0x0)
    at /home/xxx/tools/ghdl_llvm/src/grt/grt-waves.adb:988
...

Frame 1:
print *Rti
$1 = (kind => ghdl_rtik_type_record, depth => 1, mode => 1, max_depth => 0)

tgingold, 2020-07-24:
If your project is opensource, just give me the URL and I will investigate.

Otherwise, are you using vhdl2008 types ?

c-thaler, 2020-07-26:
Unfortunately it is not open source.

The project is VHDL 93, no VHDL 2008 types are used.
A lot of records, arrays of records and nested records are in the project. But I was not able to reproduce it in a simpler test case yet.

tgingold, 2020-07-27:
Can you try to run your design with `--disp-tree` and with `--dump-rti` ?
I will display a lot of text but shouldn't crash.

Does --vcd= work ?

c-thaler, 2020-07-28:
> Does --vcd= work ?
Yes. VCD does work. But it does not support some of the types. So I went to GHW.

Here are the dumps:
[dump.zip](https://github.com/ghdl/ghdl/files/4986875/dump.zip)

BTW: thanks for the quick response and help!

tgingold, 2020-07-29:
I had a look at the dumps, but I haven't spotted anything unusual...
I have to investigate the aliases that could be not fully handled.

c-thaler, 2020-07-29:
You mean the aliases that cause `ghdl_rtik_alias ?` ?

I have a minimal example that causes this:
[alias.zip](https://github.com/ghdl/ghdl/files/4993456/alias.zip)

tgingold, 2020-07-31:
Thanks!  I will add support of alias in --dump-rti, but they are correctly handled by --wave.

I am a bit of out idea on how to handle this issue.  If you could send privately the design to me that would be the easiest way, otherwise we have to wait until someone hits the same issue.

c-thaler, 2020-07-31:
Thank you.

Unfortunately I cannot share the design. But I will go on trying to make a minimal example. Maybe I'm lucky during the next weeks.

c-thaler, 2020-08-20:
Good news everyone. Finally, I reproduced the issue in a minimal example.
The example consists of two libraries (design and mylib). There is also a testbench in the default work library.

`make` should build the libraries, the testbench and run it with the `--waves` options, that triggers the issue.

Please find the sources below:
[minimal_example.zip](https://github.com/ghdl/ghdl/files/5103727/minimal_example.zip)

tgingold, 2020-08-25:
Thank you for the reproducer, that's very useful.  I am testing a fix.

closed 2020-08-26 by tgingold