ghdl/ghdl #1406: synth: simple clocked half adder

NextHendrix, 2020-07-25:
I wanted to check out the --synth feature, so built from source and knocked out the canonical clocked half adder to see what it could do. I'm using the LLVM backend and compiled with GNAT 9.3.0

```vhd :file: adder.vhd
library ieee;
use ieee.std_logic_1164.all;

entity half_adder is
  port (
    clk_in   : in  std_logic;
    n_rst_in : in  std_logic;
    a_in     : in  std_logic;
    b_in     : in  std_logic;
    c_out    : out std_logic;
    s_out    : out std_logic
    );
end half_adder;

architecture rtl of half_adder is
begin
  reg_proc : process(clk_in, n_rst_in)
  begin
    if n_rst_in = '0' then
      c_out <= '0';
      s_out <= '0';
    elsif rising_edge(clk_in) then
      c_out <= a_in and b_in;
      s_out <= a_in xor b_in;
    end if;
  end process reg_proc;
end rtl;
```

```sh
ghdl -a --std=08 --workdir=work half_adder.vhd
ghdl --synth --std=08 --workdir=work half_adder
```
which produces the following

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-859-g2f37e351) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/chris/Projects/
Command line:
ghdl --synth --std=08 --workdir=work half_adder
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Association_Choices_Chain
Call stack traceback locations:
0x7fb914f90d12 0x557b64677c71 0x557b647c2790 0x557b647c25f1 0x557b647c2945 0x557b647c46ad 0x557b647c53f0 0x557b647bb7c5 0x557b647bfef5 0x557b647c14cf 0x557b647c16f0 0x557b647c0686 0x557b647912f4 0x557b647913e0 0x557b6478b542 0x557b6480cc7a 0x557b6471db42 0x557b648166f7 0x557b645c8ff3 0x7fb914b650b1 0x557b645c7e0c 0xfffffffffffffffe
******************************************************************
```

The output of
``` sh
ghdl --version
```
is as follows
```
GHDL 1.0-dev (v0.37.0-859-g2f37e351) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

eine, 2020-07-25:
@NextHendrix it seems that you are using the latest commit (pushed a few hours ago). Unfortunately, it broke several things. Please try an earlier commit which has all successful CI. For example, https://github.com/ghdl/ghdl/commit/285ccdbd04eec0d2ddbae0296df5cae089f9739c.

NextHendrix, 2020-07-25:
> @NextHendrix it seems that you are using the latest commit (pushed a few hours ago). Unfortunately, it broke several things. Please try an earlier commit which has all successful CI. For example, [285ccdb](https://github.com/ghdl/ghdl/commit/285ccdbd04eec0d2ddbae0296df5cae089f9739c).

Compiling 285ccdbd04eec0d2ddbae0296df5cae089f9739c with the following options
``` sh
./configure --prefix=~/.local/ --with-llvm-config --enable-openieee --enable-libghdl --enable-synth
```
and running the same commands on the same file produces a similar error

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-851-g285ccdbd) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/chris/Projects/
Command line:
ghdl --synth --workdir=work --std=08 half_adder
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Association_Choices_Chain
Call stack traceback locations:
0x7f42232bbd12 0x561cf65e1c71 0x561cf672c760 0x561cf672c5c1 0x561cf672c915 0x561cf672e67d 0x561cf672f3c0 0x561cf6725795 0x561cf6729ec5 0x561cf672b49f 0x561cf672b6c0 0x561cf672a656 0x561cf66fb2c4 0x561cf66fb3b0 0x561cf66f5512 0x561cf6776c4a 0x561cf6687b3c 0x561cf67806c7 0x561cf6532ff3 0x7f4222e900b1 0x561cf6531e0c 0xfffffffffffffffe
```

I should mention that this is running on Kubuntu 20.04 and, if relevant, I don't have the version of ghdl from the ubuntu repos globally installed.

eine, 2020-07-25:
Unfortunately, I cannot reproduce... I tried:

```sh
ghdl -a --std=08 --workdir=work half_adder.vhd
ghdl --synth --std=08 --workdir=work half_adder
```

on MINGW64:

```
GHDL 1.0-dev (v0.37.0-614-g90c92fe9) [Dunoon edition]
 Compiled with GNAT Version: 10.1.0
 llvm code generator
```

and also in a ghdl/vunit:llvm container:

```
GHDL 1.0-dev (v0.37.0-851-g285ccdbd@buster-llvm-7) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 llvm code generator
```

tgingold, 2020-07-25:
Apparently the issue happens only with openieee.  I am investigating.

NextHendrix, 2020-07-25:
I've been trying compiles with earlier and earlier commits. The earliest I've tried is 077dae8a (Apr 11th), with the error still appears.

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-271-g077dae8a) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/chris/Projects/
Command line:
ghdl --synth --std=08 --workdir=work half_adder
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Association_Choices_Chain
Call stack traceback locations:
0x7f61c85a4d12 0x557999d61abf 0x557999e96c7e 0x557999e96b20 0x557999e96e44 0x557999e98838 0x557999e9927e 0x557999e91aec 0x557999e94c86 0x557999e95a3e 0x557999e95d06 0x557999e95088 0x557999e66ffa 0x557999e670e6 0x557999e61046 0x557999ee01ea 0x557999df6dec 0x557999ee9c4b 0x557999caef9f 0x7f61c81790b1 0x557999caddfc 0xfffffffffffffffe
******************************************************************
```

> Apparently the issue happens only with openieee. I am investigating.

Compiling the same version without --with-openieee works without error, so you could well be right

tgingold, 2020-07-25:
Should now be fixed.  This was due to a specific construct not widely used!

Thank you for the issue.

commit 9f2f749c (2020-07-25): synth-aggr: handle strings in aggregates.  Fix #1406
closed 2020-07-25 by tgingold