ghdl/ghdl #1412: GHDL allows non-static signal names for 'event

sckoarn, 2020-07-27:
**Description**
S'event
Prefix: Any signal denoted by the **static** signal name S.

**Expected behaviour**
The example should fail to compile,  but instead runs and outputs messages.

**How to reproduce?**
https://www.edaplayground.com/x/kaY

It is a bit long,  I did pare it down some from original

**Context**
Please, provide the following information:

- OS:
- Origin:
  -  EDA Playground

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2020-07-27:
Looks like a missing check.  All prefixes of signal attributes must be static.
Will fix.

tgingold, 2020-07-31:
On the other hand, it looks a 'bug' in the LRM.  You are allowed to pass a non-static signal name to a function and take the event attribute in the function and return it.

So, you able able to write:
function is_event (signal v : std_logic) return boolean is
begin
  return v'event;
end;

... is_event (name (i)) ..

but not:
...  name (i)'event ...

sckoarn, 2020-07-31:
Yes,  That is was my implementer stated as well.   But, passing a signal to a function is different from using a dynamic slice to detect an event on.  I see it that by the time you get to the 'event in the function v  is constant.

I see that my example got over written :( on  EDA PG.  I will be sure to not let that happen again.  Want me to reproduce some more example and testing code?

tgingold, 2020-08-01:
No that's ok, I have a reproducer.

commit 42e8cd6e (2020-08-08): vhdl-sem_names: check name staticness of signal attributes.  Fix #1412
closed 2020-08-08 by tgingold