ghdl/ghdl #1414: PSL assert statement sometimes misparsed as cover statement

rlee287, 2020-07-28:
**Description**
PSL assert statements sometimes misparse as a cover statement.

**Expected behaviour**
No cover statements end up in the synthesis output.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: cover_nonoverlap_misparse.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity cover_nonoverlap_misparse is
    Port (
        clk: in STD_LOGIC;
        counter_val: out STD_LOGIC_VECTOR(3 downto 0)
    );
end entity;

architecture Behavioral of cover_nonoverlap_misparse is
    signal ctr_internal: UNSIGNED(3 downto 0) := x"0";
begin
    process(clk) is
    begin
        ctr_internal <= ctr_internal + 1;
    end process;
    counter_val <= STD_LOGIC_VECTOR(ctr_internal);

    -- psl default clock is rising_edge(clk);
    --psl assert always ctr_internal = 0 |=> ctr_internal = 1;
end Behavioral;
```
```vhd :file: cover_overlap_misparse.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity cover_overlap_misparse is
    Port (
        clk: in STD_LOGIC;
        counter_val: out STD_LOGIC_VECTOR(3 downto 0)
    );
end entity;

architecture Behavioral of cover_overlap_misparse is
    signal ctr_internal: UNSIGNED(3 downto 0) := x"0";
begin
    process(clk) is
    begin
        ctr_internal <= ctr_internal + 1;
    end process;
    counter_val <= STD_LOGIC_VECTOR(ctr_internal);

    -- psl default clock is rising_edge(clk);
    --psl assert always ctr_internal = 0 |-> ctr_internal = 0;
end Behavioral;
```
```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth -fpsl cover_nonoverlap_misparse.vhd -e
ghdl --synth -fpsl cover_overlap_misparse.vhd -e
```

Synthesis results for `cover_nonoverlap_misparse.vhd` (has an `assert_cover` comment):

```vhdl
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
entity cover_nonoverlap_misparse is
  port (
    clk: in STD_LOGIC;
    counter_val: out STD_LOGIC_VECTOR (3 downto 0)
  );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of cover_nonoverlap_misparse is
  signal wrap_clk: std_logic;
  signal wrap_counter_val: std_logic_vector (3 downto 0);
  signal ctr_internal : std_logic_vector (3 downto 0);
  signal n5_o : std_logic_vector (3 downto 0);
  signal n8_o : std_logic;
  signal n9_q : std_logic_vector (2 downto 0) := "001";
  signal n10_o : std_logic;
  signal n12_o : std_logic;
  signal n14_o : std_logic;
  signal n15_o : std_logic;
  signal n16_o : std_logic;
  signal n18_o : std_logic;
  signal n19_o : std_logic;
  signal n20_o : std_logic;
  signal n22_o : std_logic_vector (2 downto 0);
  signal n23_o : std_logic;
  signal n24_o : std_logic;
  signal n26_o : std_logic;
begin
  wrap_clk <= clk;
  counter_val <= wrap_counter_val;
  wrap_counter_val <= ctr_internal;
  -- cover_nonoverlap_misparse.vhd:13:12
  ctr_internal <= n5_o; -- (isignal)
  -- cover_nonoverlap_misparse.vhd:17:38
  n5_o <= std_logic_vector (unsigned (ctr_internal) + unsigned'("0001"));
  -- cover_nonoverlap_misparse.vhd:21:29
  n8_o <= '1' when rising_edge (wrap_clk) else '0';
  -- cover_nonoverlap_misparse.vhd:22:11
  process (wrap_clk)
  begin
    if rising_edge (wrap_clk) then
      n9_q <= n22_o;
    end if;
  end process;
  -- cover_nonoverlap_misparse.vhd:22:11
  n10_o <= n9_q (0);
  -- cover_nonoverlap_misparse.vhd:22:11
  n12_o <= n10_o and '1';
  -- cover_nonoverlap_misparse.vhd:22:38
  n14_o <= '1' when ctr_internal = "0000" else '0';
  -- cover_nonoverlap_misparse.vhd:22:11
  n15_o <= n10_o and n14_o;
  -- cover_nonoverlap_misparse.vhd:22:11
  n16_o <= n9_q (1);
  -- cover_nonoverlap_misparse.vhd:22:59
  n18_o <= '1' when ctr_internal = "0001" else '0';
  -- cover_nonoverlap_misparse.vhd:22:59
  n19_o <= not n18_o;
  -- cover_nonoverlap_misparse.vhd:22:11
  n20_o <= n16_o and n19_o;
  n22_o <= n20_o & n15_o & n12_o;
  -- cover_nonoverlap_misparse.vhd:22:11
  n23_o <= n22_o (2);
  -- cover_nonoverlap_misparse.vhd:22:11
  n24_o <= not n23_o;
  -- cover_nonoverlap_misparse.vhd:22:11
  n25: postponed assert n24_o = '1' severity error;
  n26_o <= n22_o (1);
  -- cover_nonoverlap_misparse.vhd:22:11
  n27: assert n26_o = '1' severity note; --  assert_cover
end rtl;
```

Synthesis results for `cover_overlap_misparse.vhd` (has an `assert_cover` comment):

```vhdl
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
entity cover_overlap_misparse is
  port (
    clk: in STD_LOGIC;
    counter_val: out STD_LOGIC_VECTOR (3 downto 0)
  );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of cover_overlap_misparse is
  signal wrap_clk: std_logic;
  signal wrap_counter_val: std_logic_vector (3 downto 0);
  signal ctr_internal : std_logic_vector (3 downto 0);
  signal n5_o : std_logic_vector (3 downto 0);
  signal n8_o : std_logic;
  signal n9_q : std_logic_vector (2 downto 0) := "001";
  signal n10_o : std_logic;
  signal n12_o : std_logic;
  signal n14_o : std_logic;
  signal n15_o : std_logic;
  signal n17_o : std_logic;
  signal n19_o : std_logic;
  signal n20_o : std_logic;
  signal n21_o : std_logic;
  signal n22_o : std_logic;
  signal n25_o : std_logic_vector (2 downto 0);
  signal n26_o : std_logic;
  signal n27_o : std_logic;
  signal n29_o : std_logic;
begin
  wrap_clk <= clk;
  counter_val <= wrap_counter_val;
  wrap_counter_val <= ctr_internal;
  -- cover_overlap_misparse.vhd:13:12
  ctr_internal <= n5_o; -- (isignal)
  -- cover_overlap_misparse.vhd:17:38
  n5_o <= std_logic_vector (unsigned (ctr_internal) + unsigned'("0001"));
  -- cover_overlap_misparse.vhd:21:29
  n8_o <= '1' when rising_edge (wrap_clk) else '0';
  -- cover_overlap_misparse.vhd:22:11
  process (wrap_clk)
  begin
    if rising_edge (wrap_clk) then
      n9_q <= n25_o;
    end if;
  end process;
  -- cover_overlap_misparse.vhd:22:11
  n10_o <= n9_q (0);
  -- cover_overlap_misparse.vhd:22:11
  n12_o <= n10_o and '1';
  -- cover_overlap_misparse.vhd:22:38
  n14_o <= '1' when ctr_internal = "0000" else '0';
  -- cover_overlap_misparse.vhd:22:11
  n15_o <= n10_o and n14_o;
  -- cover_overlap_misparse.vhd:22:38
  n17_o <= '1' when ctr_internal = "0000" else '0';
  -- cover_overlap_misparse.vhd:22:59
  n19_o <= '1' when ctr_internal = "0000" else '0';
  -- cover_overlap_misparse.vhd:22:59
  n20_o <= not n19_o;
  -- cover_overlap_misparse.vhd:22:59
  n21_o <= n20_o and n17_o;
  -- cover_overlap_misparse.vhd:22:11
  n22_o <= n10_o and n21_o;
  n25_o <= n22_o & n15_o & n12_o;
  -- cover_overlap_misparse.vhd:22:11
  n26_o <= n25_o (2);
  -- cover_overlap_misparse.vhd:22:11
  n27_o <= not n26_o;
  -- cover_overlap_misparse.vhd:22:11
  n28: postponed assert n27_o = '1' severity error;
  n29_o <= n25_o (1);
  -- cover_overlap_misparse.vhd:22:11
  n30: assert n29_o = '1' severity note; --  assert_cover
end rtl;
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin: Built from sources: `4bfaa10014d923f1e27aca563362f9f389e9e4a2`

**Additional context**
The same behavior occurs with uncommented PSL assertions in `--std=08` mode.

tmeissner, 2020-07-28:
This is intentional and no misparsing. For each PSL assert directive, an additional cover directive is synthesized. Thats a very useful feature in formal verification, you don't have to write all these cover directives to check, if yout assertions are reachable. Simply use `--no-formal` option during synthesis to get a clean netlist without any PSL-directives.

rlee287, 2020-07-28:
I've only observed this behavior with sequential assertions that span multiple timesteps and not combinational assertions, and I would like to confirm that this is the intended behavior before trying to update the documentation related to this.

In addition, is there a way to disable these extra cover directives? `--no-formal` removes all PSL annotations and not just the `cover` statements. For example, if I use SymbiYosys to do a `bmc`/`prove` check before running a `cover` check, these inserted statements result in duplicated computation. I think it could be helpful to have these as an option but disabled by default, but it would still be fine as inserted by default as long as it is documented.

tgingold, 2020-07-28:
You cannot cover a combinational assertions, as they either trigger or don't.  PSL assertions can be started but later discarded.

But it could be also possible to add a cover for a sequential assertion.

I agree it would be a good idea to add an option to not cover PSL assertions.  Any idea about the option name ?

tmeissner, 2020-07-28:
Maybe `-no-assertcovers`?

rlee287, 2020-07-29:
I think that name makes sense (although it would be better to use two dashes as in `--no-assertcovers` for better UX).

commit 9efb2706 (2020-07-31): synth: add --no-assert-cover option.  Fix #1414
closed 2020-07-31 by tgingold