ghdl/ghdl #1415: TYPES.INTERNAL_ERROR : trans.adb:649 when analyzing a test bench

tmeissner, 2020-07-28:
**Description**
My Wishbone test bench cannot be analyzed anymore. It could until some change in GHDL of the last days broke it. Current head (4bfaa10) and an version which I built on july 25 don't work with the test bench.

Sadly I could not derive a MWE until now :unamused:

**Expected behaviour**
The test bench should analyse fine.

**How to reproduce?**
Clone this repo: https://github.com/tmeissner/libvhdl.git

In the repo directory, do following steps:
```sh :image: ghdl/synth:formal
$ cd test
$ make wishbonet
```
which results in the crash.

**Context**

- OS: Debian Buster
- Origin:
  - [x] Docker (Image: ghdl/synth:formal)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev () [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/build/test/
Command line:
/opt/ghdl/bin/ghdl1-llvm --std=08 -fpsl -P/opt/ghdl/lib/ghdl/ieee/v08/ -P/opt/ghdl/lib/ghdl/ -c -fpic -o WishBoneT.o WishBoneT.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:649
Call stack traceback locations:
0x4a9101 0x5cce01 0x5cd01b 0x5cd46b 0x5cd981 0x5ce706 0x5c1866 0x5c1d38 0x63f22f 0x59310d 0x645ae7 0x6473be 0x648a1c 0x40c3cd 0x7f8426cfb099 0x40b578 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-07-29:
I can reproduce the crash.

closed 2020-07-29 by tgingold

tmeissner, 2020-07-29:
Thanks for the quick fix :smiley:

commit bf5cc4bb (2020-07-29): translate: add a function to get allocation kind of layout var.

Fix #1415