ghdl/ghdl #1416: Support of force/release (VHDL-08)

tmeissner, 2020-07-29:
**Is your feature request related to a problem? Please describe.**
It would be very useful to have support of the VHDL-08 feature force/release in GHDL. Use cases are error injection in test benches for example. At the moment, you have to use workarounds like mulktiplexers or even usage of VHPIdirect or VPI, which is cumbersome.

**Describe the solution you'd like**
Support of force/release as defined in VHDL standard 1076-2008

**Describe alternatives you've considered**
At the moment, I use a multiplexer in a [UART test bench](https://github.com/tmeissner/libvhdl/blob/master/test/UartT.vhd#L142) which switches between good & bad value, controlled by a error injection procedure. Something like this:

```vhdl
-- Error injection based on random
injectError(s_error_inject);
s_rx_uart <= s_tx_uart when not s_error_inject else not(s_tx_uart);
```

**Additional context**
Full power of force/release is available with hierarchical names, but also without that it would be very useful. For example manipulating buses in a testbench, as with my UART test bench.

tgingold, 2020-07-30:
Make sense, but to be honest I haven't seen vhdl-2008 code that uses force/release.

tmeissner, 2020-07-30:
I use it at work (with Mentor tools) when verifying units which contain some kind of error detection/correction. I don't know if it is widely used. However, you won't find many open-source projects here on Github as there is no free simulator which supports it :wink:

go2sh, 2020-07-30:
I used it during my master thesis to inect errors into components and it works quite well in combination with external signals.

Paebbels, 2020-07-30:
force/release is a needed feature for OSVVM. The OSVVM library itself doesn't require it, but stub models will need this feature.

Let's amuse you have an SoC design and you want to check the interrupt handling. In such a case you need a force/release to trigger an artificial interrupt from testbench code deep inside the SoC.

tgingold, 2020-07-31:
It looks the description of release/force needs some corrections.  According to the LRM, a driver would be creating for the process that has a force or release signal assignment, which is certainly not expected by users.

tgingold, 2020-08-04:
It is now implemented, but only slightly tested.

closed 2020-08-04 by tgingold