ghdl/ghdl #1421: ERROR: wire not found for $posedge while building leon3mp

20Mhz, 2020-08-02:
I'm trying to build leon3mp from gaisler, but I'm getting the error below. I'll package a test case.

```
Importing module leon3mp.
Importing module clkpad_53_0_3_0_0_0.
Importing module inpad_53_0_3_0_0.
Importing module clkgen_53_2_2_1_0_0_0_0_50000_0_0_1_0_0.
Importing module outpad_53_0_1_3_24.
Importing module rstgen.
Importing module ahbctrl_0_0_1_0_4095_4095_4080_4080_7_8_0_0_0_2_0_1_0_0_0_0_0_0_0_0_1_1_0_0_0_0_0_0_0.
ERROR: wire not found for $posedge

```
**Expected behaviour**
What you expected to happen, and what is happening instead.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
make leon3mp
```

20Mhz, 2020-08-02:
Please use this:
https://github.com/20Mhz/issues/blob/master/leon3mp_testcase.tar.gz

Will need to deposit grlib-gpl-2020.2-b4254.tar.gz, please link lib folder into test case folder. File here https://www.gaisler.com/index.php/downloads/leongrlib

marph91, 2020-08-02:
I got the same error message `ERROR: wire not found for $posedge`. A workaround was to remove a variable, like [here](https://github.com/marph91/yaaes/blob/42f82f26d5518d3f34e9134c37ab832bfff99113/src/cipher.vhd#L87). Unfortunately I couldn't boil it down to a MWE yet.

eine, 2020-08-02:
Is any error produced with `ghdl --synth`, instead of using yosys?

tgingold, 2020-08-03:
I can reproduce the error with yaaes.  I will first investigate it.

20Mhz, 2020-08-03:
No errors when doing `ghdl --synth leon3mp`

20Mhz, 2020-08-04:
Just tested, but still seeing the problem, let me know if/how I can provide more info.
```
Importing module ahbctrl_0_0_1_0_4095_4095_4080_4080_7_8_0_0_0_2_0_1_0_0_0_0_0_0_0_0_1_1_0_0_0_0_0_0_0.
ERROR: wire not found for $posedge
[Mon Aug 03 22:37:00 ronaldv@Ronalds-MBP16:~/Projects/leon3mp ] $ which ghdl
/usr/local/bin/ghdl
[Mon Aug 03 22:37:32 ronaldv@Ronalds-MBP16:~/Projects/leon3mp ] $ ls -lrst /usr/local/bin/ghdl
10696 -rwxr-xr-x  1 root  admin  5474068 Aug  3 22:34 /usr/local/bin/ghdl
[Mon Aug 03 22:37:38 ronaldv@Ronalds-MBP16:~/Projects/leon3mp ] $ ghdl --version
GHDL 1.0-dev (v0.37.0-913-g1d5221aa) [Dunoon edition]
```

tgingold, 2020-08-04:
I have fixed the issue for yaaes, but if it doesn't fix the leon3mp case I will investigate it!

tgingold, 2020-08-06:
Should be fixed.  Thank you for the issue.

closed 2020-08-06 by tgingold

20Mhz, 2020-08-06:
Works great, thanks!