ghdl/ghdl #1422: Add rich rule checking

cuijialang, 2020-08-03:
**Is your feature request related to a problem? Please describe.**
Hello, do you have plans to add richer rules to check for syntax errors, such as Leda / spyglass, etc

**Describe the solution you'd like**
Can you direct us on how to add this feature to your version or modify those files / source code.

**Describe alternatives you've considered**
I want to add these rules, but I am still a novice and need your guidance and help

tgingold, 2020-08-03:
It depends on the kind of rules you want to check.

If you want to check for vhdl style rules (like number of spaces, case, form of identifiers), you can use the python API of ghdl.  slightly outdated) example would be on [ohwr](https://ohwr.org/project/vhdl-style/tree/master)

If you want to check synthesis rules (like CDC) you'd better to start from the netlist and that could be integrated within yosys.

eine, 2020-08-03:
Refs jeremiah-c-leary/vhdl-style-guide#328 jeremiah-c-leary/vhdl-style-guide#312

/cc @jeremiah-c-leary @m-kru

cuijialang, 2020-08-03:
thanks very much
Is there a similar set of rules，such as
RMM，DO-254，ESSENTIAL..

jeremiah-c-leary, 2020-08-04:
We use spyglass where I work.  It is focused more on finding issues like floating inputs, undriven outputs, bits that are stuck, etc...

vhdl-style-guide (VSG) is focused purely on the format of the code.

I found this link for RMM:  https://filebox.ece.vt.edu/~athanas/4514/ledadoc/html/pol_rmm.html#1066843
I think I read that book way back when.

VSG could be configured to address some of those rules.  Other RMM rules would require new rules added for VSG.

The last DO-254 project I was on was over 15 years ago.  I am guessing those rule checks are not for formatting.

I am not aware of ESSENTIAL.  Could you point me to a resource?

cuijialang, 2020-08-05:
yes,you are right
Can you tell me how to add these rule sets,As a plug-in
Checklist_ruleset、DO-254_ruleset、RMM_rulesetSafety-Critical_ruleset、Verification_ruleset、Essentials_ruleset、RTLC_ruleset
RMM_ruleset.zip
this is the rulesets
[RMM_ruleset.zip](https://github.com/ghdl/ghdl/files/5025536/RMM_ruleset.zip)

cuijialang, 2020-08-05:
@jeremiah-c-leary

closed 2020-08-06 by cuijialang