ghdl/ghdl #1423: Problems installing on Centos 7

Tismar, 2020-08-04:
Hi,
I am trying to install the latest version of GHDL in Centos 7 but keep getting an error when running the Makefile (after running the configuration script). This is the error reported:
```
gcc -c -I./ -gnat12 -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/mcode -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -gnatw.A -I- /home/me/software/ghdl/ghdl/src/synth/netlists.adb
netlists.ads:355:04: warning: no warning suppressed by this pragma
gnatmake: "/home/me/software/ghdl/ghdl/src/synth/netlists.adb" compilation error
make: *** [ghdl_mcode] Error 4
```
My system is up to date so I don't know if maybe Centos 7 packages are too old to use newer version of GHDL.
Thanks in advance.

Fatsie, 2020-08-04:
I do indeed think no recent enough version of gcc-gnat is packaged for CentOS7, I compiled myself version 8.3.0 of gcc with GNAT support to get ghdl to compile on CentOS7.

Tismar, 2020-08-04:
Thanks for the prompt response.
I've used [SCL](https://www.softwarecollections.org/en/) to be able to use GCC 8 so I can see that the GCC version is:
```
gcc (GCC) 8.3.1 20190311 (Red Hat 8.3.1-3)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
The gnat version is:
```
GNATMAKE 4.8.5 20150623 (Red Hat 4.8.5-39)
Copyright (C) 1995-2013, Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Which doesn't seem to be as up to date as the GCC.
@Fatsie , could you check with which version of GNATMAKE you were able to compile GHDL?
Thanks.

Fatsie, 2020-08-04:
gnatmake version 8.3.0; I don't think that the scl gcc versions are compiled with ADA e.g. GNAT support; there is no devtoolset-x-gcc-gnat package.

eine, 2020-08-05:
Ref #888

@sharkcz are you familiar with CentOS or you use Fedora only?

sharkcz, 2020-08-05:
I'm involved with Fedora primarily, but as has been already mentioned, the problem with CentOS/RHEL (7 and 8) will be the lack of suitable Ada compiler in EPEL or the devtoolset SCL.

Tismar, 2020-08-06:
Thanks all. I ended up compiling a newer version of GCC/GNAT (v9.2.0) and it compiled. I guess that this thread can be closed now.

closed 2020-08-06 by Tismar