ghdl/ghdl #1425: Warning on incomplete process sensitivity lists

c-thaler, 2020-08-05:
**Is your feature request related to a problem? Please describe.**
I'm usually checking my RTL simulations against the behavior of an actual FPGA implementation. It's always frustrating to see different behavior in the simulation and on the actual hardware.
In many cases I've seen, this had been caused by an incomplete sensitivity list of a process.
Checking this manually can be very time consuming in a big project. For the time being, the fastest way for me to find these issues is to compile with ModelSim, which shows warnings for signals that are used in a process and are not part of the sensitivity lists.

**Describe the solution you'd like**
It would be nice to have a warning in GHDL (`--warn-sensitivity`) that shows all signals used in a process body, which are not part of the process' sensitivity list. 

I know that some designs are making use of this to accelerate simulation. Thus, it may be appropriate to disable this warning by default.

tgingold, 2020-08-05:
Unfortunately, we cannot blindly report signals used in a process that aren't in the sensitivity list.
For processes that describe dff, many signals aren't in the sensitivity list.

Is there any reason you cannot use vhdl2008 process (all) ?

I agree that this is a useful feature I always planned to implement!

c-thaler, 2020-08-06:
> Unfortunately, we cannot blindly report signals used in a process that aren't in the sensitivity list.
> For processes that describe dff, many signals aren't in the sensitivity list.

Ah, you're right. I didn't think of clocked processes. So, a way to distinguish between clocked and combinational processes is needed.
Maybe it is possible to omit all signals whose assignment depends on an `xxx'event`? Do the internal data structures of GHDL allow this?

ModelSim somehow omits clocked processes. However, I'm not sure how this works. If you want to, I can test a few scenarios and try to figure out how it works.

> Is there any reason you cannot use VHDL2008 process (all) ?
The problem is, that I am bound to VHDL 93 since I'm working with quite old designs. Furthermore, there are tools used by me and by ASIC customers that only support '93. Sad but true :)

Blebowski, 2021-08-29:
Hi, I started to work on "dummy" implementation of this, see: [branch](https://github.com/Blebowski/ghdl/tree/warning-on-incomplete-process-sensitivity-lists)

The idea is to use sensitivity list extraction which forms the sensitivity list for ALL sensitized processes as starter when
forming "Golden" sensitivity list. This is done already and missing/extra signals are reported.

There is additional command line option defined (--check-synthesis) which can be issued at analysis time per-file
(similar to Modelsim). This option turns on sensitivity list checks.

The approach runs into the same trouble (with clocked processes) as you discussed above. There could be two ways
around this:
1. Start from all-sensitized list, walk through the process, and stop upon " 'event" expression, then throw away
    all signals within the list which are within "if" or "elsif" where " 'event' is in condition expression.
3. Walk the process tree in before, start building sensitivity list. If we encounter " 'event ", then do not nest into the
    condition which contains " 'event" in expression (or nest when walking, but stop adding signals to sensitivity list).
    Further, a warning could be thrown if there are 'events for multiple signals within one process.

There is still one issue though. When "pragma synthesis on" is set, then I think the correct behavior should be to
ignore these checks (OTOH, if TB/check code is part of RTL under pragma, it is usually non-sensitized process).
I am not sure how to resolve this. Any tips?

tgingold, 2021-08-29:
I wouldn't be worried about 'pragma synthesis off'.  At worst, we can add a flag to the processes to know if they are within the pragma or not.
But even then, there could be corner cases, like on/off within a part of a process.  But as you said, that should be OK: those processes are usually non-sensitized processes.

Don't forget about handling rising_edge/falling_edge like 'event.

I would follow the second strategy: detect for 'event (direct or indirect) in the expression of 'if' statements, and skip them when present in a conjunction.

Note that at least XST warns for conditions like 'rising_edge(clk) and enable' if 'enable' is not in the sensitivity list.  But IMHO this is not what we want.

Once you have a prototype, it could be tested on real design to check how it behaves and then adjust.

Thanks for working on that!

Blebowski, 2021-08-31:
@tgingold thanks for pointers/advices.

So far I will go with following:
1. Detect if process is "clocked process". Clocked process will be defined as process which contain one of following:

Blebowski, 2021-08-31:
Current state (which matches the description above) is in the branch.
CI fails on changed Python bindings. How do I update it to match current ADA sources?

Paebbels, 2021-08-31:
Please go into `src/vhdl` and call `make`.

umarcor, 2021-08-31:
Or, execute [scripts/update_py_bindings.sh](https://github.com/ghdl/ghdl/blob/master/scripts/update_py_bindings.sh).