ghdl/ghdl #1426: various errors like "unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS"

IIupor, 2020-08-05:
**Description**
Functions from ieee.std_logic_arith library does not work

**List errors:**

- **dcml_adjust**
```
dcml_adjust_rtl.vhd:103:41:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_GT_UNS_UNS
dcml_adjust_rtl.vhd:103:41:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_GT_UNS_UNS
dcml_adjust_rtl.vhd:88:41:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_GT_UNS_UNS
```
- **comb_divider**
```
comb_divider_rtl.vhd:88:72:warning: comparing non-numeric vector is unexpected
comb_divider_rtl.vhd:92:21:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
comb_divider_rtl.vhd:103:21:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
```
- **comb_mltplr**
```
comb_mltplr_rtl.vhd:80:18:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
```
- **addsub_cy**
```
addsub_cy_rtl.vhd:84:19:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
addsub_cy_rtl.vhd:88:19:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
```
- **addsub_ovcy**
```
addsub_ovcy_rtl.vhd:118:21:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
addsub_ovcy_rtl.vhd:121:22:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
addsub_ovcy_rtl.vhd:125:21:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
addsub_ovcy_rtl.vhd:128:22:error: unhandled function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_UNSIGNED_UNS
```
- **control_fsm**
```
control_fsm_rtl.vhd:727:38:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_NE_UNS_INT
control_fsm_rtl.vhd:754:25:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_NE_UNS_INT
control_fsm_rtl.vhd:782:25:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_NE_UNS_INT
control_fsm_rtl.vhd:810:25:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_NE_UNS_INT
control_fsm_rtl.vhd:989:38:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_NE_UNS_INT
control_fsm_rtl.vhd:1016:38:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_NE_UNS_INT
control_fsm_rtl.vhd:1219:32:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_NE_UNS_UNS
```
- **control_mem**
```
control_mem_rtl.vhd:278:42:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_LT_UNS_INT
control_mem_rtl.vhd:279:15:error: unhandled (static) function: IIR_PREDEFINED_IEEE_STD_LOGIC_ARITH_CONV_INTEGER_INT
```
- **bar**
```
bar.vhd:35:42:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_STD_LOGIC_UNSIGNED_NE_SLV_INT
```

**How to reproduce?**
**bar:**
```
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;	 
use IEEE.STD_LOGIC_ARITH.ALL;
use IEEE.STD_LOGIC_UNSIGNED.ALL;   

entity bar is
	generic(
		Clk_freq 	: integer := 50;
		Bit_rate 	: integer := 460800
		);
    port (
		rst_i 		: in std_logic;
		clk_i 		: in std_logic;	 
		start_i     : in std_logic;
    	output      : out std_logic_vector(5 downto 0)
  	);
end bar;

architecture bar of bar is

constant Freq_ratio 	: integer := Clk_freq*(10**6)/Bit_rate;
signal period_counter	: std_logic_vector(5 downto 0);
signal res			    : std_logic_vector(5 downto 0);

begin
counter: process(rst_i, clk_i)
begin		
	if rst_i = '1' then
		period_counter <= (others => '0');
	elsif rising_edge(clk_i) then
		if start_i = '1' then 
			if period_counter /= freq_ratio - 1 then
				period_counter <= period_counter + 1;
			else
				period_counter <= (others => '0');
			end if;	
		else
			period_counter <= (others => '0');
		end if;	  
	end if;
end process;	
result: process(rst_i, clk_i)
begin		
	if rst_i = '1' then
		res <= conv_std_logic_vector(0, 6);
	elsif rising_edge(clk_i) then  
		if start_i = '1' then
			res <= period_counter; 							
		else	
			res <= conv_std_logic_vector(0, 6);				
		end if;	
	end if;
end process;	
output <= res;
end bar;
```
Another VHDL modules from [https://www.oreganosystems.at/application/files/9215/3313/6321/mc8051_design_v1.6.zip](url)

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work \
dcml_adjust_.vhd dcml_adjust_rtl.vhd dcml_adjust_rtl_cfg.vhd -e dcml_adjust  

ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work \
comb_divider_.vhd comb_divider_rtl.vhd comb_divider_rtl_cfg.vhd -e comb_divider

ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work \
comb_mltplr_.vhd comb_mltplr_rtl.vhd comb_mltplr_rtl_cfg.vhd -e comb_mltplr

ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work \
addsub_cy_.vhd addsub_cy_rtl.vhd addsub_cy_rtl_cfg.vhd -e addsub_cy

ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work \
addsub_ovcy_.vhd addsub_ovcy_rtl.vhd addsub_ovcy_rtl_cfg.vhd -e addsub_ovcy

ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work \
 mc8051_p.vhd control_fsm_.vhd control_fsm_rtl.vhd control_fsm_rtl_cfg.vhd -e control_fsm

ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work \
 mc8051_p.vhd control_mem_.vhd control_mem_rtl.vhd control_mem_rtl_cfg.vhd -e control_mem

ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work bar.vhd -e bar
```

- OS:
- Origin:
  - [ ] Built from sources: #https://github.com/ghdl/ghdl/commit/83a03454a3b213d214c11f1bafa364d2dfe1d5c5

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-924-g36f25166@buster-mcode) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/mnt/test/8051/for_issuses/
Command line:
ghdl --synth --std=93 --ieee=standard -fsynopsys --work=work mc8051_p.vhd control_mem_.vhd control_mem_rtl.vhd control_mem_rtl_cfg.vhd -e control_mem
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-context.adb:516 access check failed
Call stack traceback locations:
0x55d76ab5be6d 0x55d76ab5c41c 0x55d76ab689c0 0x55d76ab75dad 0x55d76ab75fe5 0x55d76ab64acf 0x55d76ab64c36 0x55d76ab5ddea 0x55d76abc6f27 0x55d76aaf0305 0x55d76ac752f7 0x55d76a8aea15 0x7f7248a9b099 0x55d76a8ad1a8 0xfffffffffffffffe
******************************************************************
```

tmeissner, 2020-08-05:
Very sad that the hints everywhere these proprietary libraries are bad are without avail :cry:  Nevertheless, we should fix that - users of vendor tools expect them to work.

tgingold, 2020-08-06:
The test is now passing.  Thank you for the issue.

closed 2020-08-06 by tgingold