ghdl/ghdl #1427: Support for Unclocked PSL Assertions

rlee287, 2020-08-08:
**Is your feature request related to a problem? Please describe.**
Unclocked assertions are required to create combinational assertions, but GHDL does not support them. The specific use case I am considering is in the VHDL backend for Yosys that I am developing (ghdl/ghdl-yosys-plugin#122). Yosys `$assert`, `$assume` and `$cover` cells are combinational and of the form `(assert|assume|cover) EN -> A`, so I am dumping these cells as combinational asserts with the implicit clock `@(True)`. Without unclocked PSL assertions, I cannot feed the output of `write_vhdl` back into the `ghdl` synthesis frontend for testing purposes.

There may be other uses as well for unclocked assertions, such as a combinational PSL assertion in which the inputs are in different clock domains.

**Describe the solution you'd like**
Implementation of unclocked implicit `@(True)` assertions in PSL, or ideally, full support of PSL clock expressions.

**Describe alternatives you've considered**
For the exact use case described above, I could inject a `default clock is rising_edge(clk)` statement or something equivalent, which works fine for synchronous designs with a single clock. Such a workaround breaks down for designs that use multiple clocks or have asynchronous cells (like a flip-flop with asynchronous reset).

**Additional context**
I believe this can be implemented in a similar way to the handling of `process(all)`, i.e. that the assertion only needs to be checked initially and then rechecked when its input changes.

tgingold, 2020-08-09:
For assertions, I think you can use standard (concurrent) VHDL assertions.

For assume and cover, I think you simply need the support of boolean expressions.  That should be simpler to support than unclocked PSL expressions.

rlee287, 2020-08-09:
I agree with your suggestions, but I request that this issue be left open for the future so that you (or another contributor to GHDL) can potentially return to implement support for the default `@(True)` PSL clocking statement.

tmeissner, 2020-08-09:
> For assume and cover, I think you simply need the support of boolean expressions. That should be simpler to support than unclocked PSL expressions.

@tgingold What exactly do you mean with boolean expressions? As cover & assume directives only work with sequences which could contain boolean expressions. This is already implemented, AFAIK.

rlee287, 2020-08-09:
I'm interpreting that statement to mean boolean expressions in clock expressions (section 5.3 of the IEEE-1850-2005 PSL specification). My aim here is to dump Yosys `$cover` and `$assume` cells like `cover en -> a @(en'event or a'event)`. (Incidentally, using `'event` like this causes a "GHDL Bug Occured" report, which I can file as a separate issue if it remains after this issue is resolved).

tmeissner, 2020-08-10:
Ah, finally some discussions about PSL, I'm not the only one who use it with GHDL 😃 

> I'm interpreting that statement to mean boolean expressions in clock expressions (section 5.3 of the IEEE-1850-2005 PSL specification).

Ah, okay. I see 👍 

> My aim here is to dump Yosys `$cover` and `$assume` cells like `cover en -> a @(en'event or a'event)`.

You can't use PSL cover or assume directives with properties containing implications like that. They only work on sequences, clocked or not. Implications make no sense with cover or assume directives, as you want to cover are allow signal behaviour over time, so the nearest example to what you expressed would be `cover {en; a}@(en'event or a'event);` (Expect the case boolean expressions inside clock expressions would be implemented).

rlee287, 2020-08-10:
The PSL spec includes the following example:

> The directive
> assume always (ack -> next !ack);
> instructs the verification tool to constrain the verification (e.g., the behavior of the input signals) so that the property
> always (ack -> next !ack)
> holds in the design.

where `->` is a "Boolean implication operator" (and not a sequence operator). Semantically the example can be interpreted as `!ack || next(!ack)` as a boolean expression forming a sequence of length 1.

tmeissner, 2020-08-10:
Ah, okay, you're right. Next time I have to read the LRM more carefully. Most of the time I look into the book of Cindy Eisner, but also there they mention that:

> assume: This directive instructs a verification tool to constrain the verification so that a property holds.

However, `cover` & `restrict` only work on SERE.

rlee287, 2020-10-04:
Cross-ref #1479 so that both of these issues can be closed once the required functionality is implemented.