ghdl/ghdl #1428: ERROR: vhdl import failed. while trying new version of leon3mp

20Mhz, 2020-08-09:
Hi, I've updated the leon3mp test case so now is passing simulations, however nos is failing synthesis, but there is no indication of what went wrong, not sure how to investigate further.
Test case here:
https://github.com/20Mhz/leon3mp_yosys.git
https://www.gaisler.com/index.php/downloads/leongrlib

To reproduce

l```
n -s ../grlib-gpl-2020.2-b4254/lib/ .
make setup
make leon3mp
make yosys
```

Then...

yosys> ghdl leon3mp

```
1. Executing GHDL.
...
../lib/techmap/inferred/memory_inferred.vhd:48:10:note: found RAM "memarr", width: 32 bits, depth: 1024
../lib/gaisler/srmmu/mmutlb.vhd:234:14:warning: no assignment for offset 35 of variable "wb_transdata"
../lib/techmap/inferred/memory_inferred.vhd:48:10:note: found RAM "memarr", width: 8 bits, depth: 1024
../lib/techmap/inferred/memory_inferred.vhd:48:10:note: found RAM "memarr", width: 30 bits, depth: 8
ERROR: vhdl import failed.
```

tgingold, 2020-08-09:
You missed the error message:
```
../lib/gaisler/memctrl/srctrl.vhd:63:5:error: multiple assignments for offsets 297:312
```

I will try to improve the error message (to know which fields are concerned).

20Mhz, 2020-08-09:
ah! I thought I was passing the errors through tee, but it seems I didn't, below didn't help

`	cd VHDL && yosys -m ghdl -c ../scripts/yosys.tcl 2>&1 | tee ../logs/yosys.log`

20Mhz, 2020-08-09:
I've been looking at ../lib/gaisler/memctrl/srctrl.vhd but I'm still not sure what tool is trying to tell me

tgingold, 2020-08-26:
I will soon commit a patch to improve the error message, but the issue is in srctrl.vhd: `sro.scb` is assigned twice (with the same value).

commit d7f0f4d5 (2020-08-27): synth: improve diagnostic for multiple assignment. Fix #1428
closed 2020-08-27 by tgingold