ghdl/ghdl #1432: Building GHDL with mcode  backend and gives "undefined reference to `backtrace_create_state'" build error

zerowe-seven, 2020-08-12:
I configure the GHDL HEAD (at commit 68b879eb404e12a6414ef56e697b6dc99618d457) using the following command on Ubuntu 20.04 with GNAT 9.3.0 for build the GHDL mcode 

And met below error while  ghdl-llvm and ghdl-gcc work fine.

/usr/bin/ld: ./grt-backtraces-gcc.o: in function `grt__backtraces__gcc__symbolizer':
/tools/ghdl/ghdl_src/src/grt/grt-backtraces-gcc.adb:117: undefined reference to `backtrace_create_state'
/usr/bin/ld: /tools/ghdl/ghdl_src/src/grt/grt-backtraces-gcc.adb:126: undefined reference to `backtrace_pcinfo'
collect2: error: ld returned 1 exit status
x86_64-linux-gnu-gnatlink-9: error when calling /usr/bin/x86_64-linux-gnu-gcc-9
gnatmake: *** link failed.

eine, 2020-08-12:
@zerowe-seven, unfortunately, I cannot reproduce in an Ubuntu 20.04 Docker container. Can you please provide the exact sequence of commands that you are using?

See https://github.com/ghdl/ghdl/issues/931#issuecomment-535672409.

zerowe-seven, 2020-08-12:
It works now after I used a clean build dir.

1. git clone https://github.com/ghdl/ghdl.git
2. cd ghdl; mkdir build; cd build
4. ../configure --prefix=<path to>/ghdl
5. make ; make install

Before this, I was built ghdl-gcc first then ghdl-llvm, both was successfully installed.
Then when I built ghdl mcode, met this issue.

ghdl-gcc was built with GCC-6.3, but ghdl mcode using default GCC(x86_64-linux-gnu-gcc-9), maybe this is the reason caused issue.

I am closing this issue.

closed 2020-08-12 by zerowe-seven

eine, 2020-08-12:
Yes, that makes sense. The build dir should not be reused or, at least, it should be cleaned. Anyway, thanks for explaining the solution.