ghdl/ghdl #1435: Constants & Variables

ZirconfleX, 2020-08-14:
Issue: In other simulators as Modelsim, Xsim, ... it is possible to show in the waveform window the in the design and simulation used constants and variables. I did not find a way to do this in GHDL. Maybe it's possible maybe not. If not it might be an idea to add this option

Solution: a command control option like --vhd93

Alternative: Did the simulation in Xilinx Xsim.

If it's possible to display constants and variables can somebody tell me how?
I'm using GHDL compiled with LLVM on Linux-Mint-20.4 (cinnamon).

Thanks,

eine, 2020-08-15:
@ZirconfleX, you are doing nothing wrong. Unfortunately, dumping constants and/or variables is not supported in GHDL. See VUnit/vunit#449.

I'm leaving this issue open for Tristan to confirm whether this is a technical limitation or just a yet not implemented feature. Then, I believe we should update the docs accordingly.

tgingold, 2020-08-23:
How do you want variables to be dumped ?  At the end of each cycle ?

ZirconfleX, 2020-08-24:
Hi,

Xilinx ISIM and Modelsim (Questa) show constants as signals in the
waveform (see picture).
Modelsim shows also variables in a pane in its main window. They are
showed at the end of a simulation run.

Regards,

aspdigital, 2020-08-25:
It would be nice if variables were dumped at the end of the delta cycle.

A variable used as a loop iterator isn't all that interesting, as what we'll end up seeing is just the final value of the variable.

But where dumping variables is useful is when you use them as an intermediate part of a larger assignment.

travle1, 2025-08-14:
Sorry, replying to an old thread but I couldn't find any comments or documentation elsewhere about this issue.

Was this ever addressed or added as an option for the .ghw file output?
Is there a technical limitation blocking this or is it just not a priority?

Thanks for all the hard work on this great tool. (maybe I need to learn ada :) )