ghdl/ghdl #1436: GHDL -r sort of hanging, waiting till CTRL-C to fall through

ZirconfleX, 2020-08-14:
**Description**
Design with several VHDL files and one top level file.
A tester file to provide stimuli and capture reactions from the top level design. 
The reactions to the stimuli are converted to other stimuli and so on.
A testbench connection the top level design and the tester.

Analyse of all files happens perfect, no warnings at all.
Elaboration idem ditto, passes perfectly.
Run option starts the run, a message is printed to the terminal and then nothing happens any more (PC goes in overdrive).

I could not figure out what went wrong and why it went wrong.

To the design and simulation into Vivado and used the XSIM simulator.
Viavdo also crashed but it also left a log file telling what went wrong.

The tester read a text file with data and teh path to that text file was wrong. 
The text file could not be read and simulation ended closing Vivado.

After setting the path to the file in the tester correctly both XSIM and GHDL just worked.

Is there a way to let GHDL print a log file of what it's doing?
I found debug options in teh manual but seem not functional (LLVM version is the issue I guess).

**Context**
- OS: Linux-Mint-20.4
- Origin: ghdl --version displays this:
    GHDL 1.0-dev () [Dunoon edition]
    Compiled with GNAT Version: Community 2020 (20200429-93)
    llvm code generator
    Written by Tristan Gingold.

LLVM = version 10

If a example is required, please let me know then I can create a small version of the design.

Regards,

tmeissner, 2020-08-14:
Do you use the `STATUS` parameter of the `file_open` procedure to check if it was successful?

From section 5.5.2 of the VHDL-08 standard:

```vhdl
procedure FILE_OPEN (
  Status        : out FILE_OPEN_STATUS;
  file F        : FT;
  External_Name : in STRING;
  Open_Kind     : in FILE_OPEN_KIND := READ_MODE
);
```

> the value returned through the Status parameter indicates the results of the procedure call:
>
> — A value of OPEN_OK indicates that the call to FILE_OPEN was successful. If the call to
> FILE_OPEN specifies an external file that does not exist at the beginning of the call, and if the
> access mode of the file object passed to the call is write-only, then the external file is created.
>
> — A value of STATUS_ERROR indicates that the file object already has an external file associated
> with it.
>
> — A value of NAME_ERROR indicates that the external file does not exist (in the case of an attempt to
> read from the external file) or the external file cannot be created (in the case of an attempt to write or
> append to an external file that does not exist). This value is also returned if the external file cannot be
> associated with the file object for any reason.
>
> — A value of MODE_ERROR indicates that the external file cannot be opened with the requested
> Open_Kind.

With checking the `STATUS` parameter your test bench could end the simulation gracefully in case of an file not found error.

ZirconfleX, 2020-08-14:
Hi,

File_open is not used to read the file.
This is what's used (classical way to use files):

MainProc : process
fileReadFile : textopenREAD_MODE isReadDataFile;
variableReadLineLength : line;
variableSnsrRsltn : std_logic_vector(11downto0);

........

readline(ReadFile, ReadLineLength); -- Read a line.
read(ReadLineLength, SnsrRsltn); -- Put the line in a workable variable.

..........

I'm debugging an existing design I got from a customer, so cannot change
things on the fly.

tgingold, 2020-08-24:
Without a reproducer, it is difficult to have a full understanding of the problem.

If the file cannot be open, there should be an error message.

So, yes an example would be nice!

closed 2021-09-14 by tgingold