ghdl/ghdl #1438: Process "drives" entries in a record type that it can never set when array is indexed by a non-constant expression

forrestv, 2020-08-15:
**Description**
Disclaimer: This might be more of a bug in the VHDL specification than in GHDL; I don't know enough to differentiate the two.

In a process, setting an entry in a record type (`MyRecord`) that is within an array type (`MyRecordArray`) where the array index is computed (rather than constant) seems to make the process "drive" the entire array instead of just the record entries that it could possibly assign to. **This problem does not occur if the array index is constant.**

It is extremely unintuitive that changing a constant to an expression can cause a (potentially nearly invisible and hard to debug) conflict in the evaluation of separate value.

**Expected behaviour**
I would expect an entry in a record that can only be set from one place to behave normally.

Specifically, I expect the below program to give the output:

```
ghdl_bug_repro.vhd:22:9:@10ns:(report note): my_records(0).b is '1'
```

Instead, it gives the output:
```
ghdl_bug_repro.vhd:22:9:@10ns:(report note): my_records(0).b is 'U'
```

Changing `if true generate` to `if false generate` or changing the array index `i mod 2` to a constant (either `0` or `1`) results in the expected output.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.numeric_std.all;
use ieee.std_logic_1164.all;

entity ghdl_bug_repro is
end entity;

architecture arc of ghdl_bug_repro is
    type MyRecord is record
        a : std_logic;
        b : std_logic;
    end record;
    type MyRecordArray is array (0 to 1) of MyRecord;
    
    signal my_records : MyRecordArray := (others => (a => 'U', b => 'U'));
begin
    my_records(0).b <= '1';
    
    process is
    begin
        wait for 10 ns;
        report "my_records(0).b is " & std_logic'image(my_records(0).b);
        wait;
    end process;
    
    GEN : if true generate
        process is
        begin
            for i in 0 to 100 loop
                wait for 1 ns;
                my_records(i mod 2).a <= '0';
            end loop;
            wait;
        end process;
    end generate;
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ghdl_bug_repro.vhd 
ghdl -e ghdl_bug_repro
ghdl -r ghdl_bug_repro
```

**Context**
Please, provide the following information:

- OS: Ubuntu 16.04 amd64
- Origin:
  - [x] Built from sources: `51cc787952ffe352ba9caa07a9ce60cc33a02996`

eine, 2020-08-15:
@forrestv according to previous discussions, I think that the driver might be handling the full signal, which is an array, containing a record, including the field you are concerned about. IIRC, the latest related use case discussed here involved two generate blocks driving separate fields/elements from the same array/record. I will try to find the issue. Meanwhile, maybe the behaviour is different with `-frelaxed`. Did you try that?

In any case, please, don't use v0.36. That's very outdated. GHDL was improved a lot in the last 18 months. If you need help to build a more recent version (you might need to disable synthesis on Ubuntu 16.04), just let us know.

/cc @JimLewis @Paebbels

eine, 2020-08-15:
@forrestv see #1137 for an interesting discussion.

forrestv, 2020-08-15:
I just tested with 0.37.0 and master (`8789de969`); both have the same behavior as 0.36. Using `-frelaxed` results in the same behavior too.

diogratia, 2020-08-15:
Not a ghdl bug (nor an issue with the LRM).

See IEEE Std 1076-1993

7.4.2 Globally static expressions:

>An expression is said to be globally static if and only if every operator in the expression denotes a pure function and every primary in the expression is a *globally static primary*, where a globally static primary is a primary that, if it denotes an object or a function, **does not denote a dynamically elaborated named entity (see 12.5**) and is one of the following: ...

**emphasis added**

Where i in the for loop:

12.5 Dynamic elaboration

>The execution of certain constructs that involve sequential statements rather than concurrent statements also involves elaboration. Such elaboration occurs during the execution of the model.  
>  
>There are three particular instances in which elaboration occurs dynamically during simulation. These are as follows:   
>>a. Execution of a loop statement with a for iteration scheme involves the elaboration of the loop parameter specification prior to the execution of the statements enclosed by the loop (see 8.9 ). This elaboration creates the loop parameter and evaluates the discrete range.   
>> ...

The longest static prefix of the signal assignment target is the array itself. You'll have drivers for both array elements and both record elements.

You could note the standard refers to the loop parameter being static within the sequence of statements, not execution of the for loop itself.

With only two array elements and two record elements there are easy work arounds using static names.

eine, 2020-08-15:
@diogratia, would using `std_ulogic` would solve it? That was the proposed solution in #1137.

JimLewis, 2020-08-15:
Look at longest static prefix in the VHDL FAQ.

diogratia, 2020-08-16:
std_ulogic isn't a resolved type it'll make the problem visible and require fixing. Understanding why there's a driver conflict requires understanding drivers and how they are elaborated.

4.3.1.2 Signal declarations

>A signal may have one or more sources. For a signal of a scalar type,each source is either a driver (see 12.6.1 ) or an **out**, **inout**, **buffer,** or **linkage** port of a component instance or of a block statement with which the signal is associated. For a signal of a composite type, each composite source is a collection of scalar sources, one for each scalar subelement of the signal. It is an error if, after the elaboration of a description, a signal has multiple sources and it is not a resolved signal. It is also an error if, after the elaboration of a description, a resolved signal has more sources than the number of elements in the index range of the type of the formal parameter of the resolution function associated with the resolved signal. 

12.6.1 Drivers

>Every signal assignment statement in a process statement defines a set of *drivers* for certain scalar signals. There is a single driver for a given scalar signal S in a process statement, provided that there is at least one signal assignment statement in that process statement and that the longest static prefix of the target signal of that signal assignment statement denotes S or denotes a composite signal of which S is a subelement. Each such signal assignment statement is said to be associated with that driver. Execution of a signal assignment statement affects only the associated driver(s). 

12.4 Elaboration of a statement part

 12.4.4 Other concurrent statements

>All other concurrent statements are either process statements or are statements for which there is an equivalent process statement.  
>  
>Elaboration of a process statement proceeds as follows:  

 >> a. The process declarative part is elaborated.  
 >> b. The drivers required by the process statement are created.  
 >> c. The initial transaction defined by the default value associated with each scalar signal driven by the process statement is inserted into the corresponding driver.   
>  
>Elaboration of all concurrent signal assignment statements and concurrent assertion statements consists of the construction of the equivalent process statement followed by the elaboration of the equivalent process statement.

closed 2020-08-26 by tgingold