ghdl/ghdl #1444: Std i/o problem with ghdl-0.37-mingw64-llvm build on Windows 10

edcerny, 2020-08-24:
I downloaded the above mentioned binary for Windows. It runs ok under Windows 10 PowerShell, except the libgrt.a library does not seem to contain functions related to std i/o.  During elaboration I get
C:\Users\Eduard\Documents\Analog\ghdl-0.37-mingw64-llvm\GHDL\0.37-mingw64-llvm\lib/ghdl/libgrt.a(win32.o): In function `ghdl_SEH_handler':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/config/win32.c:150: undefined reference to `__imp___acrt_iob_func'
C:\Users\Eduard\Documents\Analog\ghdl-0.37-mingw64-llvm\GHDL\0.37-mingw64-llvm\lib/ghdl/libgrt.a(grt-cstdio.o): In function `__ghdl_get_stdout':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cstdio.c:30: undefined reference to `__imp___acrt_iob_func'
C:\Users\Eduard\Documents\Analog\ghdl-0.37-mingw64-llvm\GHDL\0.37-mingw64-llvm\lib/ghdl/libgrt.a(grt-cstdio.o): In function `__ghdl_get_stdin':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cstdio.c:36: undefined reference to `__imp___acrt_iob_func'
C:\Users\Eduard\Documents\Analog\ghdl-0.37-mingw64-llvm\GHDL\0.37-mingw64-llvm\lib/ghdl/libgrt.a(grt-cstdio.o): In function `__ghdl_get_stderr':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cstdio.c:42: undefined reference to `__imp___acrt_iob_func'
C:\Users\Eduard\Documents\Analog\ghdl-0.37-mingw64-llvm\GHDL\0.37-mingw64-llvm\lib/ghdl/libgrt.a(grt-cvpi.o): In function `loadVpiModule':
C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:97: undefined reference to `__imp___acrt_iob_func'
C:\Users\Eduard\Documents\Analog\ghdl-0.37-mingw64-llvm\GHDL\0.37-mingw64-llvm\lib/ghdl/libgrt.a(grt-cvpi.o):C:/projects/ghdl-psgys/build/mingw64-llvm/../../src/grt/grt-cvpi.c:110: more undefined references to `__imp___acrt_iob_func' follow
collect2.exe: error: ld returned 1 exit status
C:\Users\Eduard\Documents\Analog\ghdl-0.37-mingw64-llvm\GHDL\0.37-mingw64-llvm\bin\ghdl.exe: compilation error

Can someone help me top resolve the problem? I do not have much knowledge of compiling programs on Windows.
Thank you,

eine, 2020-08-24:
Do you have MSYS2 installed on the host? If not, please, download https://repo.msys2.org/distrib/x86_64/msys2-base-x86_64-20200720.tar.xz, extract it, execute `ming64.exe` and proceed as explained in steps 6-9 from https://www.msys2.org/#installation.

Then, try to use the same GHDL from that terminal: `/c/users/Eduard/Documents/Analog/ghdl-0.37-mingw64-llvm/GHDL/0.37-mingw64-llvm/bin/ghdl.exe`.

The reasoning is that GHDL for Windows is built in MSYS2. Then, the resulting artifacts are zipped and published. Hence, there is no guarantee that it will work without a working MSYS2 environment. It might be possible to add the missing DLLs to the zipfile (should that be your problem), however, there is not automatic way to achieve it yet. On the other hand, GHDL might be compiled statically. I don't know whether anyone has attempted it before.

edcerny, 2020-08-24:
Thank you! I will try that and let you know.
Ed

edcerny, 2020-08-25:
It is running now! Thank you,
ed

tgingold, 2020-08-25:
Thank you for the update.  I close this issue.

closed 2020-08-25 by tgingold