ghdl/ghdl #1450: libghdl mingw32 binary missing dependencies

tumagonx, 2020-08-28:
**Description**
in version 0.37 for mingw32, libghdl-0_37.dll is shipped without gnat runtime LIBGNAT-9.DLL and gcc runtime (in this case dwarf exception LIBGCC_S_DW2-1.DLL)

**Expected behaviour**
libghdl released with complete dependencies

**How to reproduce?**
put libghdl dll into dependency walker, and it will complain missing hard dependencies

- OS: windows xp sp3
- Origin:
  - [ ] Package manager: `0.37`
  - [ ] Released binaries: `https://github.com/ghdl/ghdl/releases/download/v0.37/ghdl-0.37-mingw32-mcode.zip`
  - [ ] Built from sources: `https://github.com/ghdl/ghdl/archive/v0.37.zip`

**Additional context**
OT, is libghdl only needed for binding? should I worried if I don't care about binding? could it be build statically (single dll)?

eine, 2020-08-28:
Hi @tumagonx! Related to #1444, can you please try executing it in MSYS2?

eine, 2020-08-28:
> is libghdl only needed for binding?

It is used by the Python interface, the Yosys plugin, the language server, etc. Nevertheless, I think it is optional if you want to use GHDL for simulation only.

tumagonx, 2020-08-28:
ah I see, obviously it will works that way. the thing is I have my own built gcc (a static-only one unfortunately). So I built ghdl with dll statically linked to gnat/gcc runtime. So i wonder if the resulting libghdl will be ok, but looks like I don't need it.

eine, 2020-08-30:
@tumagonx, as commented in #1444, there are two possible solutions:

- Add the missing DLLs to the package. This cannot be done automatically yet. See gtkwave/gtkwave#20:

> [achadwick/styrene](https://github.com/achadwick/styrene) (or some other reference from [msys2.org/wiki/Distributing](https://www.msys2.org/wiki/Distributing/)) and/or [msys2.org/wiki/Using-packages/#finding-dependencies-of-a-package](https://www.msys2.org/wiki/Using-packages/#finding-dependencies-of-a-package).

- As you suggest, build GHDL statically. I think that no one tried that on Windows before. Anyway, see #1336, which is about building a static version of the yosys plugin, and it needs `libghdl.a`.

For testing if the libghdl you built is functional, you can try the following:

```sh
# in the root of the repo
export PYTHONPATH=$(pwd)/python
cd testsuite/python/units01
GHDL=ghdl ./testsuite.sh
```

tumagonx, 2020-08-31:
tested on pypy3.5 (windows XP):

`raised CONSTRAINT_ERROR : errorout.adb:284 access check failed`