ghdl/ghdl #1453: Unsigned slicing operation crashes in 2008 mode

targeted, 2020-09-01:
**Description**
Compilation crashes with assertion and stack trace in 2008 mode. Reports error w/o 2008.

**How to reproduce?**
Compile the following file.

```vhd :file: bug.vhd
library ieee;
use ieee.numeric_std.all;
package bug is
    subtype byte is unsigned (7 downto 0);
    subtype word is unsigned (15 downto 0);
end bug;
package body bug is
    function high_byte(w: word) return byte is
    begin
        return (7 downto 0 => w(15 downto 8));
    end;
end bug;
```
```sh
# ghdl -a --std=08 bug.vhd
```
**Context**
- OS: Windows 10
- Origin: 0.37-mingw64-llvm

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (v0.37) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-w64-mingw32
Command line:
C:\ghdl\0.37-mingw64-llvm\bin\ghdl1-llvm.exe --std=08 -PC:\ghdl\0.37-mingw64-llvm\lib/ghdl\ieee\v08\ -PC:\ghdl\0.37-mingw64-llvm\lib/ghdl\ -c -o bug.o bug.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap7.adb:3074
Call stack traceback locations:
0x1582786 0x5dc1f8 0x5dbc99 0x5de38d 0x5de449 0x5e1fe1 0x5e1943 0x589181 0x5a247b 0x5a263d 0x625a2c 0x5c82a8 0x626ba6 0x5782cf 0x6356b6 0x636fcf 0x6389e5 0x402685 0x40138a 0x4014b9 0x7ffea9fd7bd2 0x7ffeaaf6ce4f
******************************************************************
```

eine, 2020-09-01:
Can you please try https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-nightly-mingw64-llvm.zip or https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-x86_64-ghdl-llvm-ci-1-any.pkg.tar.zst to confirm that the bug still exists in `master`?

diogratia, 2020-09-01:
The error message without `--std=08` is valid. The default is `--std=93c` wherein the associations are of the element type. See IEEE Std 1076-1993 7.3.2 Aggregates. It's non sequitur to the bug.

The -2008 failure appears to be a lack of complete implementation and shows up in a more recent
commit (d85a1a9, from 25 June 2020):

ghdl --version
GHDL 1.0-dev (v0.37.0-773-gd85a1a9) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm code generator

See IEEE Std 1076-2008 9.3.3.3 Array aggregates:

>For an aggregate of a one-dimensional array type, each choice shall specify values of the index type, and the expression of each element association shall be of either the element type or the type of the aggregate. If the type of the expression of an element association is the type of the aggregate, then either the element association shall be positional or the choice shall be a discrete range.  
>...  
>For an element association with a choice that is a discrete range and an expression of the type of the aggregate, each element of the value of the expression is the value of the element of the aggregate at the matching index value in the range. The *matching index value* for an element of the value of the expression is determined as follows: the leftmost element of the value matches the left bound of the range; if an element matches an index value, the element immediately to its right matches the index value immediately to the right in the range. It is an error if the length of the discrete range differs from the length of the value of the expression.

The difference from earlier revisions of the standard would be to require the type of the association actual be ascertained and to invoke the above quoted semantic rule.

The function return value expression appears valid in -2008, the type of the expression is taken from the function declaration.

A work around might be:
```vhdl
    function high_byte(w: word) return byte is
        variable highbyte:  byte := w(15 downto 8);  -- ADDED
    begin
        -- return (7 downto 0 => w(15 downto 8));    -- REPLACED
        return highbyte;                             -- ADDED
    end;
```
where the variable is required for subtype conversion and it's initial value is the result of dynamic elaboration.

targeted, 2020-09-01:
Thank you for your time and advices. 

Unfortunately, nightly build still crashes:
```

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-996-g498ee9b6) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-w64-mingw32
Command line:
C:\ghdl\1.0-dev-mingw64-llvm\bin\ghdl1-llvm.exe --std=08 -PC:\ghdl\1.0-dev-mingw64-llvm\lib/ghdl\ieee\v08\ -PC:\ghdl\1.0-dev-mingw64-llvm\lib/ghdl\ -c -o bug.o bug.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap7.adb:3099
Call stack traceback locations:
0x157fba6 0x5dbe4e 0x5db8a3 0x5de15e 0x5de21b 0x5e1cbe 0x5e203a 0x5e1daf 0x58b899 0x5a3145 0x5a3321 0x6246da 0x5c787f 0x625830 0x57a421 0x633356 0x634cac 0x636531 0x402691 0x40138a 0x4014b9 0x7ffea9fd7bd2 0x7ffeaaf6ce4f
******************************************************************
C:\ghdl\1.0-dev-mingw64-llvm\bin\ghdl.exe: compilation error
```

commit 9ce2b99b (2021-09-12): trans-chap7: improve handling of vector in aggregates.  Fix #1453
closed 2021-09-12 by tgingold