ghdl/ghdl #1454: ghdl --synth instantiations removal causes assertions not to be run

LudvigVidlid, 2020-09-03:
Hi,

Maybe this issue isn't really a bug, but it seemed most fitting to post it as such :)

**Description**
When synthesizing an architecture, any instantiations in that architecture will be pruned if the signals are not used. This is nice for synthesis, but it also removes any assertions in that module causing formal checks to pass when they should fail.

**Expected behaviour**
Not removing the instantiation if it contains any kind of assertions or psl.

**How to reproduce?**
Synthesize  this code:

```vhd :file: dummy_top.vhd
library ieee;
use ieee.std_logic_1164.all;

entity dummy_sub is
port (
  clk : in std_logic;
  dummy : out std_logic
);
end entity;

architecture a of dummy_sub is
  signal first_cycle : std_logic := '1';
begin
  support : process
  begin
    wait until rising_edge(clk);
    dummy <= '0';
    assert clk = '0';
  end process;
  
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity dummy_top is
  port(
    clk : in std_logic;
    dummy : out std_logic
  );
end entity;

architecture a of dummy_top is
begin
  ------------------------------------------------------------------------------
  dummy_sub_inst : entity work.dummy_sub
    port map(
      clk => clk,
      dummy => open -- Connecting dummy here triggers instantiation of dummy_sub
    );

end architecture;
```

```sh :image: ghdl/synth:formal
ghdl --synth --std=08 -fpsl dummy_top.vhd -e dummy_top
```

What I'm trying to do is to bundle psl assertions in a separate entity, to be instantiated in multiple places.

Maybe there are better ways to do this, e.g using verification units. Any hints would be appreciated!

tmeissner, 2020-09-03:
You could use PSL vunits, but they are only supported with synthesis and VHDL-08. What you try is an use case I also use at work with Questa. The only difference is that I have an almost empty VHDL-component which I bind a PSL vunit to. So I can instantiate the VHDL component in various test benches.

LudvigVidlid, 2020-09-03:
@tmeissner, thank you. I found your example with vunits in your psl_with_ghdl example repo, but didn't realize that it could be used to achieve what I want, as I thought it might be removed as well. I'll try it!

I do have some problems understanding the purpose vunits. As they are always bound to a specific architecture they mostly seem like an extra step. But maybe this is their use case :)

tgingold, 2020-09-03:
I think the clean phase removes all cells without connected outputs.  Maybe it should be less aggressive with user cells.

tmeissner, 2020-09-03:
@LudvigVidlid The PSL standard allows to make vunits much more flexible than binding them explicitely to only one unit. As IEEE 1850-2010 says in section 7.2:

> If a Hierarchical HDL Name is given, then the verification unit is explicitly bound to the design module or
> module instance indicated by the HDL module name(s) and HDL instance name(s) of the Hierarchical HDL
> Name. If only one HDL module name is given, then the name indicates a VHDL entity, a Verilog module, a
> SystemVerilog module or interface, or a SystemC (C++) class that inherits from sc_module. If two HDL
> module names are given, then the pair of module names indicates a VHDL entity and architecture,
> respectively. If no Hierarchical HDL Name is given, then the verification unit is not explicitly bound.

But I don't know exactly which of these possibilities are supported by GHDL.

LudvigVidlid, 2020-09-04:
> I think the clean phase removes all cells without connected outputs. Maybe it should be less aggressive with user cells.

Sounds like a good suggestion, although I don't know as much as you about the side effects it may have. 

Ideally an assertion (or similar) could be considered an "output" of a cell, so that the cell is not removed. But I realize that has some complexity to it.

@tmeissner I connected a vunit to ``dummy_sub``. The connection works in the sense that my assertion is visible in the synthesis output of ``dummy_sub``. But the problem remains that ``dummy_sub`` is not instantiated within ``dummy_top``

closed 2020-09-05 by tgingold

tgingold, 2020-09-05:
This is now fixed, thank you for the issue.

The assertion doesn't fail when simulating the synthesis result because it is too sensitive to the timing.

commit 1aa515cc (2020-09-05): netlists-cleanup: do not remove user sub modules.  Fix #1454