ghdl/ghdl #1455: Constrained and unconstrained arrays of real_vector

umarcor, 2020-09-05:
```vhdl
package pkg0 is
  type my_arr2D_t is array (natural range <>) of real_vector;

  constant my_arr2D: my_arr2D_t(0 to 1)(0 to 1) := (
    (0.0, 0.1),
    (1.0, 1.1)
  );
end;

-- Correct
```

```vhdl
package pkg1 is
  type my_arr2D_t is array (natural range <>) of real_vector;

  constant my_arr2D: my_arr2D_t := (
    (0.0, 0.1),
    (1.0, 1.1)
  );
end;

-- ******************** GHDL Bug occurred ***************************
-- Please report this bug on https://github.com/ghdl/ghdl/issues
-- GHDL release: 1.0-dev (v0.37.0-949-gc689f6ad) [Dunoon edition]
-- Compiled with GNAT Version: 10.2.0
-- Target: x86_64-w64-mingw32
-- D:\data-dev\umarcor\ghdl\ghdl-cosim\
-- Command line:
-- C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o pkg1.o errors/pkg1.vhd
-- Exception TYPES.INTERNAL_ERROR raised
-- Exception information:
-- raised TYPES.INTERNAL_ERROR : trans.adb:1674
-- Call stack traceback locations:
-- 0x49ca12 0x5870bd 0x587d54 0x58a4dd 0x588a24 0x588d17 0x5c0ae8 0x5c0f17 0x5c0e83 0x5c373f 0x5c37fb 0x5f6271 0x5f6645 0x5f673b 0x5ff2da 0x5eb092 0x5ea908 0x5ea9a3 0x5d3650 0x633248 0x634ba5 0x636475 0x4026ea 0x4013bf 0x4014f4 0x7ffbc6817bd2 0x7ffbc7b2ce4f
-- ******************************************************************
-- C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

```vhdl
package pkg2 is
  type my_arr2D_t is array (natural range <>) of real_vector;

  constant my_arr2D: my_arr2D_t(0 to 1) := (
    (0.0, 0.1),
    (1.0, 1.1)
  );
end;

-- ******************** GHDL Bug occurred ***************************
-- Please report this bug on https://github.com/ghdl/ghdl/issues
-- GHDL release: 1.0-dev (v0.37.0-949-gc689f6ad) [Dunoon edition]
-- Compiled with GNAT Version: 10.2.0
-- Target: x86_64-w64-mingw32
-- D:\data-dev\umarcor\ghdl\ghdl-cosim\
-- Command line:
-- C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o pkg.o errors/pkg2.vhd
-- Exception TYPES.INTERNAL_ERROR raised
-- Exception information:
-- raised TYPES.INTERNAL_ERROR : trans.adb:1674
-- Call stack traceback locations:
-- 0x49ca12 0x5870bd 0x587d54 0x58a4dd 0x588a24 0x588d17 0x5c0ae8 0x5c0f17 0x5c0e83 0x5c373f 0x5c37fb 0x5f6271 0x5f6645 0x5f673b 0x5ff2da 0x5eb092 0x5ea908 0x5ea9a3 0x5d3650 0x633248 0x634ba5 0x636475 0x4026ea 0x4013bf 0x4014f4 0x7ffbc6817bd2 0x7ffbc7b2ce4f
-- ******************************************************************
-- C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

```vhdl
package pkg3 is
  type my_arr2D_t is array (0 to 1) of real_vector;

  constant my_arr2D: my_arr2D_t := (
    (0.0, 0.1),
    (1.0, 1.1)
  );
end;

-- ******************** GHDL Bug occurred ***************************
-- Please report this bug on https://github.com/ghdl/ghdl/issues
-- GHDL release: 1.0-dev (v0.37.0-949-gc689f6ad) [Dunoon edition]
-- Compiled with GNAT Version: 10.2.0
-- Target: x86_64-w64-mingw32
-- D:\data-dev\umarcor\ghdl\ghdl-cosim\
-- Command line:
-- C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o pkg3.o errors/pkg3.vhd
-- Exception TYPES.INTERNAL_ERROR raised
-- Exception information:
-- raised TYPES.INTERNAL_ERROR : trans.adb:1674
-- Call stack traceback locations:
-- 0x49ca12 0x5870bd 0x587d54 0x58a4dd 0x588a24 0x588d17 0x5c0ae8 0x5c0f17 0x5c0e83 0x5c373f 0x5c37fb 0x5f6271 0x5f6645 0x5f673b 0x5ff2da 0x5eb092 0x5ea908 0x5ea9a3 0x5d3650 0x633248 0x634ba5 0x636475 0x4026ea 0x4013bf 0x4014f4 0x7ffbc6817bd2 0x7ffbc7b2ce4f
-- ******************************************************************
-- C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

```vhdl
package pkg4 is
  type my_arr2D_t is array (0 to 1) of real_vector;

  constant my_arr2D: my_arr2D_t(0 to 1) := (
    (0.0, 0.1),
    (1.0, 1.1)
  );
end;

-- errors/pkg4.vhd:4:32: constrained array cannot be re-constrained
-- C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

```vhdl
package pkg5 is
  type my_arr2D_t is array (0 to 1) of real_vector;

  constant my_arr2D: my_arr2D_t(open)(0 to 1) := (
    (0.0, 0.1),
    (1.0, 1.1)
  );
end;

-- Correct
```

```vhdl
package pkg6 is
  type my_arr2D_t is array (0 to 1) of real_vector;

  constant my_arr2D: my_arr2D_t(0 to 1)(0 to 1) := (
    (0.0, 0.1),
    (1.0, 1.1)
  );
end;

-- errors/pkg6.vhd:4:32: constrained array cannot be re-constrained
-- C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

tgingold, 2020-09-05:
Yes, there has been some issues already reported but you are much more comprehensive!

umarcor, 2020-09-05:
I opened this issue because of gitlab.com/IEEE-P1076/VHDL-Issues#15](https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/15). This is indeed a summary of the issues for building 2D types as arrays of unconstrained arrays (`real_vector`, `integer_vector`, `std_logic_vector`...). I think having the MWEs in this issue "fixed" would provide a foundation for understanding proper multidimensional arrays. Note that `pkg4` and `pkg6` might be invalid VHDL, and I'm unsure about `pkg2`. So, I would prioritize `pkg1` and `pkg3`.

closed 2021-08-03 by tgingold

umarcor, 2021-08-15:
I can confirm that none of the examples crashes now. Thanks so much @tgingold! 😃

commit db7e1aa8 (2021-08-03): vhdl-sem_expr.adb: build element subtype for aggregate when possible.

In case of array aggregate whose element subtype is not bounded, extract it from
the aggregate elements.

Fix #1055
Fix #1455