ghdl/ghdl #1457: Co-simulation CMAKE project main function conflict

Astiolo, 2020-09-07:
I'm trying to build a co-simulation with cmake and my own main. I've managed to get it to work if I don't provide a main, by giving the output of "ghdl --list-link xxx" as LINK_FLAGS.

```
execute_process(COMMAND ghdl --list-link co_sim_tb
    OUTPUT_VARIABLE GHDL
    OUTPUT_STRIP_TRAILING_WHITESPACE
)
string(REPLACE "\n" " " GHDL "${GHDL}")
set_target_properties(${PROJECT_NAME} PROPERTIES LINK_FLAGS ${GHDL})
```

However, when I've got a main function defined, I get a "multiple defrinition of 'main'" error. It shows there is a main function defined inside the grt library archive in the main.adb file (https://github.com/ghdl/ghdl/blob/master/src/grt/main.adb). 

I'm struggling to understand why this isn't a conflict for standard projects wrapping the simulation when built directly with gcc, but is when it's in CMAKE.

Are there any examples of a co-simulation CMAKE project? Or, any ideas how to avoid the conflict and use my main function?

Astiolo, 2020-09-07:
Actually I've managed to get it working. I found the link.txt file that CMAKE was making with the actual call to the compiler. I managed to reorder the options in a way that make would work. Then I found that I could provide -Wl,${GHDL} in target_link_libraries in the CMakeLists and it compiled.

So the relevant part of the CMAKE now looks like this:
```
execute_process(COMMAND ghdl --list-link <vhdl_tb_module>
    OUTPUT_VARIABLE GHDL
    OUTPUT_STRIP_TRAILING_WHITESPACE
)
string(REPLACE "\n" " " GHDL "${GHDL}")

target_link_libraries(
  ${PROJECT_NAME}
  PUBLIC
  <other_link_libraries>
  -Wl,${GHDL}
)
```

I still don't really understand why the change in order fixes the multiple main issue, but it's working.

closed 2020-09-07 by Astiolo

tgingold, 2020-09-07:
There is indeed a default main() in the library (libgrt.a).  But when you link with it, you can provide you own main().  In that case, it should appear before libgrt.a so that it would be picked by the linker and the default one won't.

So you can have issues (like multiple definition of main()) if the order of files or options is not correct.

umarcor, 2020-09-07:
I'm not familiar with CMAKE, but I did find the same issue before. There are multiple ways to build VHDL along with C sources:

- Compile C sources first.
  - If you use `gcc main.c`, the output will be an standalone binary, with `main` as the entrypoint. Then, when you try to link it to GHDL, you get the error.
  - If you use `gcc -c main.c`, the output will be an object file, which includes `main` as a function, but not as the entrypoint. Then, GHdl will find it, and it will use it instead of the internal one.
- Provide C sources as arguments to `ghdl -e`. In this case, GHDL will call the compiler internally, and it will handle the existence of `main`. In fact, this is the suggested approach, unless you need to customize the output of `--list-link`.

See https://ghdl.github.io/ghdl-cosim/vhpidirect/linking.html#linking-foreign-object-files-to-ghdl and https://ghdl.github.io/ghdl-cosim/vhpidirect/linking.html#linking-ghdl-object-files-to-ada-c.