ghdl/ghdl #1460: Synthesis of float_pkg

maelic-louart, 2020-09-09:
Can float_pkg be synthesized with ghdl?

tgingold, 2020-09-10:
I have never tried.

If you have any issue with float_pkg, please report it.  Unless it is too complex, I will try to fix it.

maelic-louart, 2020-09-11:
Thank you for your quick reply.

### Context

**Description**
Using the _float_pkg_ package to program division in VHDL can not be synthesized. It leads to GHDL Bug.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.float_pkg.all;

entity division_float is 
port(
i0 : in  float (7 downto -6);
i1 : in  float (7 downto -6);
p0 : out float (7 downto -6));
end division_float ;

architecture arch1 of division_float is 

begin
 p0 <= i0/i1 ;
end arch1;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 division_float.vhd 
ghdl --synth --std=08 division_float
```

**Context**

- OS: Linux Mint 19.2 Cinnamon
- Origin:
  - [ ] Package manager: v0.37

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1001-g776b85c6) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/home/louart/vhdl/
Command line:
ghdl --synth --std=08 division_float
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Subtype_Indication
Call stack traceback locations:
0x7f35788b57b1 0x55c5b4601599 0x55c5b47708b1 0x55c5b477099b 0x55c5b477139b 0x55c5b47715e6 0x55c5b4740300 0x55c5b47403e8 0x55c5b473a383 0x55c5b47a3902 0x55c5b46ca5f6 0x55c5b4852f6f 0x55c5b448147b 0x7f3577e6ab95 0x55c5b447fd58 0xfffffffffffffffe

******************************************************************
```

**Additional context**
File is analyzed without error and cause a crash when is synthesized.

eine, 2020-09-11:
@maelic-louart, how do other tools support this use case? Typically, division for synthesis is not supported, unless it is a power of two (which translates to shiftting). Should the target architecture have hard blocks for division, I would expect a black box or simulation model to be provided in VHDL.

Anyway, GHDL should not crash.

maelic-louart, 2020-09-11:
Hi,

 It does not depend on division because I have the same bug with addition. 

Thank you.

commit 25c8d5a6 (2020-09-20): testsuite/synth: Add a test.  Close #1460
closed 2020-09-20 by tgingold